OpenIntelWireless/itlwm #746: Wi-Fi has to be toggled off then on after waking from sleep

techandrew, 2022-02-09:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: AirportItlwm 2.1.0 (also tried with itlwm: exact same issue)
- WiFi Card Model: Intel AC 8260NGW
- PCI Product ID: 8086 24f3
- macOS Version: 12.2

**Description**
With both the AirportItlwm.kext and itlwm.kext, Wi-Fi works as expected upon starting up the computer. However, after going to sleep, then waking to sleep, Wi-Fi does not work at all (no networks shown, no connection) until Wi-Fi is turned off then turned back on.

**Bug Report Archive**
Attached.
[bugreport_65416.zip](https://github.com/OpenIntelWireless/itlwm/files/8028690/bugreport_65416.zip)

**Kext Download Source**
This repository.

williambj1, 2022-02-09:
@techandrew Logs after sleep and before toggling are missing, could you recollect it? Thanks.

williambj1, 2022-02-11:
@techandrew Hmmm, logs are still incomplete (only 6 lines were collected). Could you collect the logs again with this version of DebugEnhancer? https://github.com/acidanthera/DebugEnhancer/suites/5219776724/artifacts/159934798

techandrew, 2022-02-12:
[bugreport_18281.zip](https://github.com/OpenIntelWireless/itlwm/files/8054491/bugreport_18281.zip)

ancan, 2022-02-13:
I can collect logs as well, but don't know how to use DebugEnhancer. I've installed the kext, but can't find out where the logs are or if debugging needs to be enabled in some way.

For me, toggling WiFi doesn't help. Once the computer have been sleeping, I only see my iPhone hotspot. My "proper" WiFi is Ubiquity with UAP-AC-LR and UAP-AC-Mini AP's.

techandrew, 2022-02-13:
> I can collect logs as well, but don't know how to use DebugEnhancer. I've installed the kext, but can't find out where the logs are or if debugging needs to be enabled in some way.
>
> For me, toggling [...]

To get the logs: provided you have DebugEnhancer.kext from the link from @williambj1, use HeliPort (regardless of whether you are using AirportItlwm.kext or itlwm.kext), hold Option while clicking on the menu, then click Create Diagnostics Report.

ancan, 2022-02-14:
[bugreport_5219.zip](https://github.com/OpenIntelWireless/itlwm/files/8059463/bugreport_5219.zip)

Bugreport post-sleep state attached. I've also tried with a couple of other WiFi-cards, both Intel (8260NGW and 7265NGW) but they unsurprisingly have the same issue.

singhalrishi27, 2022-03-02:
Same issue here

williambj1, 2022-03-03:
@singhalrishi27 Please provide logs and see if you have the same output below.

```
[  282.515122]: itlwm: itlwm: acquiring device failed
[  282.515129]: itlwm: itlwm: unable to init nic
[  282.515133]: itlwm: itlwm: failed to load init firmware
[  282.515136]: itlwm: iwm_stop_device
```

singhalrishi27, 2022-03-03:
> @singhalrishi27 Please provide logs and see if you have the same output below.
> 
> ```
> [  282.515122]: itlwm: itlwm: acquiring device failed
> [  282.515129]: itlwm: itlwm: unable to init nic
> [  282.515133]: itlwm: itlwm: failed to load init firmware
> [  282.515136]: itlwm: iwm_stop_device
> ```

I have logs in my privaterelay post will see

williambj1, 2022-03-03:
@singhalrishi27 Please don't spam posts if you're having another issue.

zxystd, 2022-03-17:
> I can collect logs as well, but don't know how to use DebugEnhancer. I've installed the kext, but can't find out where the logs are or if debugging needs to be enabled in some way.
>
> For me, toggling [...]

@ancan 
before sleep, works normal
```
itlwm: L1 Disabled - LTR Enabled
```
after sleep
```
itlwm: L1 Enabled - LTR Enabled
```
The driver should be work on L1 disabled state(ASPM disabled), your machine enabled it after sleep, so this issue is not about the driver, fix your EFI, thanks.

zxystd, 2022-03-17:
@techandrew 
can you try if this one would help?
[AirportItlwm_Monterey.kext.zip](https://github.com/OpenIntelWireless/itlwm/files/8284169/AirportItlwm_Monterey.kext.zip)

techandrew, 2022-03-17:
> Collaborator
> ### [[image]](/zxystd) ** [zxystd](/zxystd) ** commented [9 hours ago](#issuecomment-1070802561)

Ok, I replaced the kext with the one you provided... so far when I wake from sleep it reconnects by itself as usual, but I'll let you know if it continues to work in the next few days (with the previous kext, it occasionally reconnected after sleep and I want to rule this out).

zxystd, 2022-03-18:
Ok, waiting for your confirmation.

techandrew, 2022-03-19:
> Ok, waiting for your confirmation.

I've been using this continuously for the past day, waking from sleep many times. Seems like it's connecting automatically now.

closed 2022-03-19 by techandrew