OpenIntelWireless/itlwm #747: AC 7265 connection drop and then a system crash

jozews321, 2022-02-10:
**Have You Read Our Docs**

Yes
**Are You Reporting A Bug**

Yes
**Environment**

- Kext Version: 2.1.0 stable and 2.2.0 alpha
- WiFi Card Model: AC 7265
- PCI Product ID: 0x8086, 0x095A - ASPM L1
- macOS Version: 12.0, 12.1, 12.2

**Description**

Everything works fine at first but seemingly random the wifi connection drops and then the system becomes unusable apps crashing and then a complete crash, I've tried Airportitlwm 2.0.0, 2.1.0, 2.2.0 alpha and normal itlwm but the same issue 
**Bug Report Archive**

At the moment of the network drop
[Log_2022-02-09_06-34-31.log](https://github.com/OpenIntelWireless/itlwm/files/8037202/Log_2022-02-09_06-34-31.log)

After that
[Log_2022-02-09_06-56-56.log](https://github.com/OpenIntelWireless/itlwm/files/8037203/Log_2022-02-09_06-56-56.log)
After this log the next time that I try to get more logs I get an empty file and shortly macOS crashes

Another log of a prior crash (may be useful)
[image]

**Kext Download Source**

Official Airportitlwm stable and alpha

zxystd, 2022-02-11:
The log shows everything works fine except the picture you uploaded, but it is hard to say the crash is caused by the driver, will see if there have more people who are getting the same crash.

Overc1ocker, 2022-02-21:
You have AC7265 revision A. These cards seem to cause more problems then other versions.
I have a revision B AC7265 (0x8086, 0x095B) and it's worked perfectly since kext 2.0.0. 
I'm curious what the differences between the 2 cards are.

Make sure to specify in the title that you have revision A of the card. It will help isolate it and it's problems.

zxystd, 2022-02-24:
@jozews321  Also please attach the panic log.

closed 2024-03-16 by zxystd