OpenIntelWireless/itlwm #748: AC-9560 160Mhz Crashes

chr1srbx, 2022-02-18:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: 2.1.0
- WiFi Card Model: Intel Wireless AC-9650 160Mhz
- PCI Product ID: 4126 
- macOS Version: 11.0 Big Sur

**Description**
Crashes when making a connection with a wifi

**Kext Download Source**
Here

TingyiChen, 2022-02-20:
Same random crash on AC9560, but happens on Monterey 12.0.1 with AirportItlwm v2.1.0.
Sometimes it shows itlwm in backtrace, sometimes not.
[image]

chr1srbx, 2022-02-20:
Same thing on big sur

chr1srbx, 2022-02-20:
but it freezes the recovery

zxystd, 2022-02-21:
@TingyiChen add `keepsyms=1` to boot-args to display the full stack trace when panic happened.

zxystd, 2022-02-21:
@chr1srbx Attach the panic log too.

al-jabirr, 2023-02-21:
[@zxystd](https://github.com/zxystd)
_Please check the panic log_ with (keepsyms=1) **boot-args** ....

panic(cpu 8 caller 0xffffff8003f485f3): Kernel trap at 0xffffff8003b01057, type 14=page fault, registers:
CR0: 0x0000000080010033, CR2: 0x0000000000000000, CR3: 0x000000043a03b064, CR4: 0x00000000003626e0
RAX: 0x0000000000000000, RBX: 0xffffff96f2987800, RCX: 0x0000000000000103, RDX: 0x0000000000000103
RSP: 0xffffffd058f9f5e8, RBP: 0xffffffd058f9f620, RSI: 0xffffff96f5f2b588, RDI: 0x0000000000000000
R8:  0x0000000000000103, R9:  0x0000000000000001, R10: 0xffffff96f299c000, R11: 0xffffff9bbf520430
R12: 0xffffff96f299c000, R13: 0x0000000000000000, R14: 0xffffffd058f9fb38, R15: 0x0000000000000000
RFL: 0x0000000000010286, RIP: 0xffffff8003b01057, CS:  0x0000000000000008, SS:  0x0000000000000010
Fault CR2: 0x0000000000000000, Error code: 0x0000000000000002, Fault CPU: 0x8, PL: 0, VF: 0

Panicked task 0xffffffa5587d0208: 22 threads: pid 239: airportd
Backtrace (CPU 8), panicked thread: 0xffffff9bc003b0c8, Frame : Return Address
0xffffffd058f9efc0 : 0xffffff8003deb38d mach_kernel : _handle_debugger_trap + 0x4ad
0xffffffd058f9f010 : 0xffffff8003f58ed6 mach_kernel : _kdp_i386_trap + 0x116
0xffffffd058f9f050 : 0xffffff8003f48120 mach_kernel : _kernel_trap + 0x3e0
0xffffffd058f9f0a0 : 0xffffff8003d85951 mach_kernel : _return_from_trap + 0xc1
0xffffffd058f9f0c0 : 0xffffff8003deb66d mach_kernel : _DebuggerTrapWithState + 0x5d
0xffffffd058f9f1b0 : 0xffffff8003dead19 mach_kernel : _panic_trap_to_debugger + 0x1a9
0xffffffd058f9f210 : 0xffffff80045e072b mach_kernel : _panic + 0x84
0xffffffd058f9f300 : 0xffffff8003f485f3 mach_kernel : _sync_iss_to_iks + 0x2c3
0xffffffd058f9f480 : 0xffffff8003f482e2 mach_kernel : _kernel_trap + 0x5a2
0xffffffd058f9f4d0 : 0xffffff8003d85951 mach_kernel : _return_from_trap + 0xc1
0xffffffd058f9f4f0 : 0xffffff8003b01057 
0xffffffd058f9f620 : 0xffffff80080cb765 com.zxystd.AirportItlwm : __ZN12AirportItlwm17apple80211RequestEjiP16IO80211InterfacePv + 0x835
0xffffffd058f9f6a0 : 0xffffff8005b13fb0 com.apple.iokit.IO80211FamilyLegacy : __ZL13getAP_IE_ListP17IO80211ControllerP16IO80211InterfaceP23IO80211VirtualInterfaceP21IO80211InfraInterfaceP13apple80211reqb + 0xdb
0xffffffd058f9faf0 : 0xffffff8005b2b8f2 com.apple.iokit.IO80211FamilyLegacy : __ZN17IO80211Controller20apple80211_ioctl_getEP16IO80211InterfaceP23IO80211VirtualInterfaceP23IO80211SkywalkInterfacePv + 0x46
0xffffffd058f9fb20 : 0xffffff8005b2f40b com.apple.iokit.IO80211FamilyLegacy : __ZN16IO80211Interface19performGatedCommandEPvS0_S0_S0_S0_ + 0xf5
0xffffffd058f9fb80 : 0xffffff8006602ef2 com.apple.iokit.IONetworkingFamily : __ZN19IONetworkController20executeCommandActionEP8OSObjectPvS2_S2_S2_ + 0x38
0xffffffd058f9fbb0 : 0xffffff8004515aa4 mach_kernel : __ZN13IOCommandGate9runActionEPFiP8OSObjectPvS2_S2_S2_ES2_S2_S2_S2_ + 0xb4
0xffffffd058f9fc10 : 0xffffff8006602f6a com.apple.iokit.IONetworkingFamily : __ZN19IONetworkController14executeCommandEP8OSObjectPFiPvS2_S2_S2_S2_ES2_S2_S2_S2_S2_ + 0x60
0xffffffd058f9fc70 : 0xffffff8005b2f2c9 com.apple.iokit.IO80211FamilyLegacy : __ZN16IO80211Interface14performCommandEP19IONetworkControllermPvS2_ + 0x73
0xffffffd058f9fca0 : 0xffffff8004051927 mach_kernel : _ifnet_ioctl + 0x287
0xffffffd058f9fd00 : 0xffffff80040487ee mach_kernel : _ifioctl + 0xbde
0xffffffd058f9fdb0 : 0xffffff800404ed4f mach_kernel : _ifioctllocked + 0x2f
0xffffffd058f9fde0 : 0xffffff80043510c5 mach_kernel : _soo_select + 0x795
0xffffffd058f9fe40 : 0xffffff800434bdae mach_kernel : _ioctl + 0x4ce
0xffffffd058f9ff40 : 0xffffff800445545b mach_kernel : _unix_syscall64 + 0x1db
0xffffffd058f9ffa0 : 0xffffff8003d85d96 mach_kernel : _hndl_unix_scall64 + 0x16
      Kernel Extensions in backtrace:
         com.apple.iokit.IONetworkingFamily(3.4)[ACD4449A-4F90-31DB-82AD-2F2B83D8F6E3]@0xffffff80065f8000->0xffffff800660efff
         com.apple.iokit.IO80211FamilyLegacy(1200.12.2b1)[8CB6AB86-0DC8-324C-A9F7-B46C452522DD]@0xffffff8005a98000->0xffffff8005bdefff
            dependency: com.apple.driver.AppleMobileFileIntegrity(1.0.5)[DB38BB35-D38C-3110-9B53-0D2927F702CF]@0xffffff80053c5000->0xffffff80053f3fff
            dependency: com.apple.driver.corecapture(1.0.4)[1EFC00FC-8337-39AF-B061-EDA1CB53C6E7]@0xffffff800704a000->0xffffff800706cfff
            dependency: com.apple.iokit.CoreAnalyticsFamily(1)[BF4BE411-8D61-3123-AEA0-88671FB17B81]@0xffffff8005726000->0xffffff800572ffff
            dependency: com.apple.iokit.IONetworkingFamily(3.4)[ACD4449A-4F90-31DB-82AD-2F2B83D8F6E3]@0xffffff80065f8000->0xffffff800660efff
            dependency: com.apple.iokit.IOSkywalkFamily(1.0)[337AFA35-6288-3EA9-BC6E-AA8E075EECF6]@0xffffff8006942000->0xffffff800696ffff
            dependency: com.apple.kec.corecrypto(12.0)[CADEE2DB-DF62-3630-A9F9-BC8B75D43579]@0xffffff800707b000->0xffffff80070f3fff
         com.zxystd.AirportItlwm(2.1)[FA02D059-D593-3D8E-A031-1F99EA4D67EB]@0xffffff8008014000->0xffffff8008fadfff
            dependency: com.apple.iokit.IO80211FamilyLegacy(1200.12.2b1)[8CB6AB86-0DC8-324C-A9F7-B46C452522DD]@0xffffff8005a98000->0xffffff8005bdefff
            dependency: com.apple.iokit.IONetworkingFamily(3.4)[ACD4449A-4F90-31DB-82AD-2F2B83D8F6E3]@0xffffff80065f8000->0xffffff800660efff
            dependency: com.apple.iokit.IOPCIFamily(2.9)[802B9CA4-B851-396E-8DA4-B30E751EC48F]@0xffffff8006859000->0xffffff8006888fff

Process name corresponding to current thread (0xffffff9bc003b0c8): airportd
Boot args: -novht -noht40 itlwm_cc=BD keepsyms=1

[... cut ...]
  . PGZ   : 0xffffff908a5db000 - 0xffffff908c5dc000
  . VM    : 0xffffff908c5dc000 - 0xffffff9558ddb000
  . RO    : 0xffffff9558ddb000 - 0xffffff96f25db000
  . GEN0  : 0xffffff96f25db000 - 0xffffff9bbeddb000
  . GEN1  : 0xffffff9bbeddb000 - 0xffffffa08b5db000
  . GEN2  : 0xffffffa08b5db000 - 0xffffffa557ddb000
  . GEN3  : 0xffffffa557ddb000 - 0xffffffaa245db000
  . DATA  : 0xffffffaa245db000 - 0xffffffb08a5db000
  Metadata: 0xffffffd05d39f000 - 0xffffffd07d39f000
  Bitmaps : 0xffffffd07d39f000 - 0xffffffd08339f000

closed 2024-03-16 by zxystd