OpenIntelWireless/itlwm #749: How to use iPhone Hotspot with ITLWM

BigEmperor26, 2022-02-19:
If you are using ITLWM you cannot enable iPhone hotspot by pressing the Wi-Fi icon. However if you have the iPhone on the hotspot page, you can go to settings/network/join another network and manually enter the SSID and password of your hotspot. It will not fail.

usr-sse2, 2022-03-01:
You can just (disable Bluetooth on the notebook OR disable Bluetooth on iPhone) AND turn on hotspot in iPhone Settings or Control Center. Either of these actions will make iPhone appear not as Instant Hotspot (that shows the charge level) but as a regular Wi-Fi network, so that Mac will not try to turn on the hotspot on the iPhone but just connect to it.

williambj1, 2022-03-03:
I've updated the docs to include this guide. Thanks for the contribution.

closed 2022-03-03 by williambj1

brucewayne229, 2022-08-11:
What about using AirportItlwm ? Its the same ? I cant connect with my personal hotspot. I click on it but after few minutes it said " personal hotspot could not be enable " im using intel AC 7265