OpenIntelWireless/itlwm #750: 英特尔9560HUM

Zeroyellow, 2022-02-21:
Have You Read Our Docs
Yes

Are You Reporting A Bug
Yes

Environment
Kext Version: AirportItlwm [v2.2.0-alpha]
WiFi Card Model: intel 9560HUM 
Product ID: 
8086 A370 8086 0034  Intel Corporation              
Cannon Lake PCH CNVi WiFi                          
Network controller   Network controller   
CI0@0/CNVW@14,3 pci8086,a370    
PciRoot(0x0)/Pci(0x14,0x3) 

macOS Version: Monterey 12.2.1 (21D62)
Description
大佬，您好，dell 7070mff 我的Wi-Fi无论是AirportItlwm （各个版本都尝试过）还是 itlwm 搭配HeliPort使用，都无法驱动，尝试了很多次都没有用，为啥呢？
[image]

蓝牙倒是没问题很完美！这个是我的日志，望大佬赐教，万分感谢！

[Log_2022-02-21_23-08-49.log](https://github.com/OpenIntelWireless/itlwm/files/8110364/Log_2022-02-21_23-08-49.log)

zxystd, 2022-02-21:
Ubuntu下是否正常使用？看起来像是又是被Linux忘记支持的设备类型，这种类型的卡里面缺少了国家监管信息。

Zeroyellow, 2022-02-22:
非常感谢大佬的回复，win下正常使用，Ubuntu没测试，boot-args 添加 itlwm_cc=CN也没用，还有啥办法吗？

zxystd, 2022-02-24:
有时间的话下载个Ubuntu试一下，可以直接写到U盘直接启动就可以测试，看看Wi-Fi正常不正常。

zxystd, 2022-03-07:
@Zeroyellow  这个卡在Linux下都是没有适配的，而Windows下的驱动未开源，你如果有时间可以配合我测试的话，我可以尝试给你这个卡单独适配。

zxystd, 2022-03-22:
还在吗？

Zeroyellow, 2022-03-22:
在，大佬，请指教

M69W, 2022-03-23:
加群说话🙃

zxystd, 2022-03-25:
> 在，大佬，请指教

需要给这张卡加适配的话，建议进QQ群直接沟通。不然只能关issue了，没办法测试。