OpenIntelWireless/itlwm #751: No WPA3 for itlwm?

jdaien4783, 2022-02-22:
Hi there I wasn't sure where to post this so sorry if this is the wrong place to post but I was wondering if WPA3 support is being worked on for itlwm and if it will be added in a future release. I just purchased the AX210 card and it is working great in Mac OS Catalina with WPA2 but doesn't connect at all with WPA3. 

Before purchasing this card, I was using the TP-Link Archer T9UH USB Adapter.  I use the following driver for the usb adapter https://github.com/chris1111/WirelessAdapterCloverBigSur. This also shows up as ethernet rather than native wifi.  However, this kext supports WPA3 no problem.

zxystd, 2022-02-24:
Not support yet.

williambj1, 2022-03-03:
WPA3 requires 802.11w support which is still a work in progress in #676

marcelloinfoweb, 2023-03-25:
Not support yet?

Lorys89, 2024-01-03:
Is there any news about wpa3 support?