OpenIntelWireless/itlwm #752: Wifi only works on cold boot

murilopereirame, 2022-02-24:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: 2.1.0
- WiFi Card Model: AX200
- PCI Product ID: 2723
- macOS Version: 12.2.1

**Description**
Wifi only works on cold boot, if I reboot the system it doesn't work anymore and when the wifi works the bluethooth doesn't connect to any device, when wifi doesn't work, the bluethooth works fine. I'm using the latest stable version of  Wifi and Bluethooth kext.

**Kext Download Source**
GitHub Release

lim417dev, 2022-02-26:
I also encountered the situation that WiFi does not work. Bluetooth works normally. The configuration is the same as yours. There are the same problems with version 2.2

williambj1, 2022-03-03:
Warm reboots from others OSes leaves the card with unsupported firmware and an unsupported states. Cold boots are always recommended when switching between OSes.

murilopereirame, 2022-03-03:
I'm not switching OSes, the Wifi stop working after rebooting the OSX

williambj1, 2022-03-03:
@murilopereirame Please provide logs after the reboot.

closed 2022-03-25 by zxystd