OpenIntelWireless/itlwm #753: Takes too long to auto connect when Cloudflare WARP is enabled

naveenkrdy, 2022-03-01:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: https://github.com/OpenIntelWireless/itlwm/tree/v2.2.0-alpha
- WiFi Card Model: AX200
- PCI Product ID: 0x2723
- macOS Version: Monterey 12.2.1

**Description**
Wi-Fi takes too long to auto-connect to the network after wake from sleep, when Cloudflare WARP is enabled. 
I have to manually disable Cloudflare WARP and then switch off-on my Wi-Fi in macOS to make it reconnect fast.

[Log_2022-03-02_00-28-42.log](https://github.com/OpenIntelWireless/itlwm/files/8164454/Log_2022-03-02_00-28-42.log)

zxystd, 2022-03-02:
From the log I can see the driver reconnect instantly after wakeup.

williambj1, 2022-03-03:
@naveenkrdy Does the issue still appear if Cloudflare WARP is not enabled before sleep? User space software usually does not affect kernel extensions' behavior.

naveenkrdy, 2022-03-03:
No ! 
network is back instantly after wake from sleep. when cloudflare warp is disabled prior to sleep.

williambj1, 2022-03-03:
Sounds like Cloudflare WARP may have issues finding the active network interface after sleep, which is a common issue with macOS VPN software. Unfortunately, based on the respond this does not appear to be an issue directly related to this project. Thanks for the feedback.

closed 2022-03-03 by williambj1