OpenIntelWireless/itlwm #754: AirportItlwm would not load

shural, 2022-03-04:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: 2.2.0
- WiFi Card Model: AX1650i
- PCI Product ID: 8086:34f0
- macOS Version: Monterey 12.2

**Description**
AirportItlwn does not load automatically (after listed in OC0.77 config.plist), nor load by kextload in Terminal App. The latter shows "Authenticating extension failed: Bad code signature" .....".

**Bug Report Archive**

**Kext Download Source**
(https://github.com/OpenIntelWireless/itlwm/releases/download/v2.2.0-alpha/AirportItlwm-Monterey-v2.2.0-DEBUG-alpha-eb76e46.zip)

williambj1, 2022-03-04:
Use other sources for personal config issues.

closed 2022-03-04 by williambj1

shural, 2022-03-04:
I do not believe this is a configuration issue - actually it comes plain and simple - just drag the kext file into OCAuxiliaryTools->Kernel page. Itlwm works just fine w/ Helio, but AirportItlwm does not.

940842546, 2022-03-10:
it's not a configuration issue
I have the same issue
load on terminal shows "Authenticating extension failed: Bad code signature"
I use macOS Monterey