OpenIntelWireless/itlwm #758: 英特尔 Wireless-AC 9560

jlynet, 2022-03-11:
**Have You Read Our Docs**

**Are You Reporting A Bug**

**Environment**

- Kext Version: 
- WiFi Card Model: 
- PCI Product ID: 
- macOS Version: 

**Description**

In the process of use, there will be the problem of sudden drive failure and sudden failure to drive.

**Bug Report Archive**

**Kext Download Source**

zxystd, 2022-03-14:
log please?

jlynet, 2022-03-15:
How to export this log?

jlynet, 2022-03-15:
这个日志怎么导出啊？是在出问题情况下导出日志是吧！

zxystd, 2022-03-17:
@jlynet 对，出问题了再导出日志，尽快操作，否则会被系统的其他日志刷掉。
导出方法：https://docs.oiw.workers.dev/itlwm/Troubleshooting.html#runtime-logs

jlynet, 2022-03-18:
[image]
执行这些命令就可以了吗？

zxystd, 2022-03-18:
[DebugEnhancer-1.0.6-DEBUG.zip](https://github.com/OpenIntelWireless/itlwm/files/8300401/DebugEnhancer-1.0.6-DEBUG.zip)
要先加载这个。
另外你可以换成2.1.0稳定版或者2.2.0-alpha版再试么？

jlynet, 2022-03-21:
[Log_2022-03-21_11-06-17.log](https://github.com/OpenIntelWireless/itlwm/files/8312944/Log_2022-03-21_11-06-17.log)

日志导出了。使用的是2.1.0稳定版

zxystd, 2022-03-21:
日志不对，debugenhance要用上

jlynet, 2022-03-24:
执行：
```
sudo /System/Library/PrivateFrameworks/Apple80211.framework/Versions/A/Resources/airport debug +AllUserland +AllDriver +AllVendor
```
报错：
```
Unable to retrieve vendor debug flags...ignoring error.
Unable to set vendor debug flags...ignoring error.
```

zxystd, 2022-05-08:
Can be reopened once the necessary information for this issue is provided.

closed 2022-05-08 by zxystd