OpenIntelWireless/itlwm #760: Any plan to support Marvell Technology Group 88W8897 [AVASTAR] 802.11ac Wireless?

TomHeaven, 2022-03-22:
It's great the project wakes up tens of thousands of Intel Wireless cards for macOS. I wonder if it is possible to include support for more brands? A number of Surface Pro notebooks uses Marvell Technology Group's wireless card. It would be great that drivers for those wireless cards can be created.

Drivers for Linux are [here](https://linux-hardware.org/?id=pci:11ab-2b38&dev_class=02-80&dev_type=net%2Fwireless&dev_vendor=Marvell+Technology+Group&dev_name=88W8897+%5BAVASTAR%5D+802.11ac+Wireless&dev_ident=53c58).

Drivers for OpenBSD are [here](https://man.openbsd.org/iwm.4).

Overc1ocker, 2022-03-22:
Hi! As that card is not an Intel one, it would require a completely new codebase. That isn’t to say getting the adapter working would be impossible, and zxystd’s work porting Openbsd drivers couldn’t be helpful, but you’d need a new group of people to help you do this if you wanted to have support.

zxystd, 2022-03-22:
Sorry, No :)

closed 2022-03-22 by zxystd