OpenIntelWireless/itlwm #763: intel 8265 开启wifi使用一段时间后  无网死机

778280151, 2022-03-27:
**Have You Read Our Docs**

yes
**Are You Reporting A Bug**

yes
**Environment**

- Kext Version: 2.1/2.2
- WiFi Card Model:  intel 8265
- PCI Product ID: 
- macOS Version:  12.3

**Description**

panic(cpu 2 caller 0xffffff80174ab55a): mbuf_watchdog: 1 waiters stuck for 18 secs
54572/54604 mbufs in use:
	54568 mbufs allocated to data
	4 mbufs allocated to packet headers
	32 mbufs allocated to caches
732/1364 mbuf 2KB clusters in use
26922/26922 mbuf 4KB clusters in use
0/0 mbuf 16KB clusters in use
124508 KB allocated to network (approx. 98% in use)
0 KB returned to the system
VM allocation failures: contiguous 0, normal 0, one page 0
worker thread runs: 97, expansions: 176, cl 0/0, bigcl 96/42900, 16k 0/0
worker thread last run time: 562 (17 seconds ago)

mbuf leak detection table:
	total captured: 892 (one per 500)
	total allocs outstanding: 170
	new hash recorded: 886 allocs, 567 traces
	hash collisions: 76 allocs, 1 traces
	overwrites: 417 allocs, 0 traces
	lock conflicts: 0

top 5 outstanding traces:
[1] 49 outstanding alloc(s), 51 hit(s), 0 collision(s)
[2] 29 outstanding alloc(s), 50 hit(s), 0 collision(s)
[3] 27 outstanding alloc(s), 103 hit(s), 0 collision(s)
[4] 19 outstanding alloc(s), 72 hit(s), 0 collision(s)
[5] 19 outstanding alloc(s), 48 hit(s), 0 collision(s)

    trace [1]           trace [2]           trace [3]       	trace [4]           trace [5]      
    ------------------  ------------------  ------------------      ------------------  ------------------ 
 1: 0xffffff80008a9cfa  0xffffff80008a9cfa  0xffffff80008a9cfa  0xffffff80008a9cfa  0xffffff80008a9cfa  
 2: 0xffffff800088372f  0xffffff8000883699  0xffffff8000883699  0xffffff8000883699  0xffffff800088372f  
 3: 0xffffff80008a8c7d  0xffffff80008a8c2a  0xffffff80008adf88  0xffffff80008b0625  0xffffff80008b0625  
 4: 0xffffff800088360e  0xffffff800088360e  0xffffff80008b21db  0xffffff800087d95a  0xffffff800087d95a  
 5: 0xffffff80008b0625  0xffffff80008b0625  0xffffff800087cd23  0xffffff8002b750bd  0xffffff8002b750bd  
 6: 0xffffff800087d95a  0xffffff800087d95a  0x0  0x0  0x0  
 7: 0xffffff8002b750bd  0xffffff8002b750bd  0x0  0x0  0x0  
 8: 0x0  0x0  0x0  0x0  0x0  
 9: 0x0  0x0  0x0  0x0  0x0  
10: 0x0  0x0  0x0  0xffffff8000a47444  0xffffff8000a47444  
11: 0x0  0x0  0x0  0xffffff8000a45cfe  0xffffff8000a45cfe  
12: 0xffffff8000a47444  0xffffff8000a47444  0x0  0xffffff8000a45327  0xffffff8000a45327  
13: 0xffffff8000a45cfe  0xffffff8000a45cfe  0x0                                          
14: 0xffffff8000a45327  0xffffff8000a45327  0x0                                          
15:                                         0x0                                          
16:                                         0xffffff8000a47444                                          

Panicked task 0xffffff9045d11a20: 10 threads: pid 448: com.apple.WebKit
Backtrace (CPU 2), panicked thread: 0xffffff99dfb3faa8, Frame : Return Address
0xffffffd08513b590 : 0xffffff8016e83e1d 
0xffffffd08513b5e0 : 0xffffff8016fe3ca6 
0xffffffd08513b620 : 0xffffff8016fd34fd 
0xffffffd08513b670 : 0xffffff8016e23a60 
0xffffffd08513b690 : 0xffffff8016e841ed 
0xffffffd08513b7b0 : 0xffffff8016e839a6 
0xffffffd08513b810 : 0xffffff80177164af 
0xffffffd08513b900 : 0xffffff80174ab55a 
0xffffffd08513b990 : 0xffffff80174a9277 
0xffffffd08513ba00 : 0xffffff801748360e 
0xffffffd08513ba80 : 0xffffff80174a8c7d 
0xffffffd08513bb20 : 0xffffff801748360e 
0xffffffd08513bba0 : 0xffffff80174b0625 
0xffffffd08513bc30 : 0xffffff80171e40a2 
0xffffffd08513bcb0 : 0xffffff80171eacc2 
0xffffffd08513bd00 : 0xffffff801722fabb 
0xffffffd08513bd40 : 0xffffff801723f242 
0xffffffd08513bef0 : 0xffffff801723c91c 
0xffffffd08513bf40 : 0xffffff801758ad1b 
0xffffffd08513bfa0 : 0xffffff8016e24226 

Process name corresponding to current thread (0xffffff99dfb3faa8): com.apple.WebKit.Networking
Boot args: igfxonln=1 

Mac OS version:
21E230

Kernel version:
Darwin Kernel Version 21.4.0: Mon Feb 21 20:34:37 PST 2022; root:xnu-8020.101.4~2/RELEASE_X86_64
Kernel UUID: 9619904E-EA6D-3D08-89C4-B1C0451D39B8
KernelCache slide: 0x0000000016c00000
KernelCache base:  0xffffff8016e00000
Kernel slide:      0x0000000016c10000
Kernel text base:  0xffffff8016e10000
__HIB  text base: 0xffffff8016d00000
System model name: iMac18,1 (Mac-4B682C642B45593E)
System shutdown begun: NO
Panic diags file available: YES (0x0)
Hibernation exit count: 0

System uptime in nanoseconds: 579219904053
Last Sleep:           absolute           base_tsc          base_nano
  Uptime  : 0x00000086dc325892
  Sleep   : 0x0000000000000000 0x0000000000000000 0x0000000000000000
  Wake    : 0x0000000000000000 0x0000000d55dc9fbe 0x0000000000000000
Compressor Info: 0% of compressed pages limit (OK) and 0% of segments limit (OK) with 0 swapfiles and OK swap space
Zone info:
  Foreign : 0xffffff802f236000 - 0xffffff802f244000
  Native  : 0xffffff8047310000 - 0xffffffa047310000
  Readonly: 0xffffff8513fdc000 - 0xffffff86ad975000
  Metadata: 0xfffffff849447000 - 0xfffffff8695c8000
  Bitmaps : 0xfffffff8695c8000 - 0xfffffff86f5c8000

**Bug Report Archive**

**Kext Download Source**

zxystd, 2022-03-27:
尝试一下连接不加密的Wi-Fi会不会有这样的问题？

778280151, 2022-03-28:
https://github.com/OpenIntelWireless/itlwm/issues/763#issuecomment-1080046146

> 尝试一下连接不加密的Wi-Fi会不会有这样的问题？
好的，晚上回家尝试下，看下来我和下面这个老哥的情况差不多，safari联网用一段时间后，先是没网，然后设置里面看网络 设备 wifi 和有限网卡全部丢失，再过一会儿电脑就黑屏重启了
 https://github.com/OpenIntelWireless/itlwm/issues/735#issue-1100197460

778280151, 2022-03-28:
> 尝试一下连接不加密的Wi-Fi会不会有这样的问题？
[console.log](https://github.com/OpenIntelWireless/itlwm/files/8364466/console.log)
[itlwm.log](https://github.com/OpenIntelWireless/itlwm/files/8364467/itlwm.log)

z大我已经试过了，即使是wpa2的无密码热点也会出现问题，上面一个是系统崩溃后的日志，下面的是驱动日志。
详细情况就是用safari看b站视频，看着看着就没有网络了，然后看系统设置-网络里面Wi-Fi选项直接消失，紧接着一会儿就黑屏重启

778280151, 2022-03-29:
> * 用Chrome看视频呢？
> * 用itlwm.kext + Heliport.app呢？
> * `wpa2无密码`到底是wpa2还是无密码？
> * 以前有遇到过吗，或者在低版本的系统有遇到过吗，什么时候开始发生的
> * 可以上传一份/System/Library/Extensions/O80211FamilyLegacy.kext上来吗

1. 还没有试过用chrome，今晚回去测试
2. 没试过itlwm.kext + Heliport.app，今晚测试
3. wpa2有密码和无密码都试过了，都会一段时间后无网络
4. 发生问题的机器不是很经常用，最近当作主力机才发现问题
5. 同前两条，今晚测试后反馈

778280151, 2022-03-29:
补充：  
用有线网络没有发现这种问题

778280151, 2022-03-29:
[IO80211FamilyLegacy.kext.zip](https://github.com/OpenIntelWireless/itlwm/files/8372045/IO80211FamilyLegacy.kext.zip)

778280151, 2022-03-29:
itlwm.kext + Heliport.app  虽然不会导致电脑黑屏重启，但是还是会断网
[itlwm_heilport.log](https://github.com/OpenIntelWireless/itlwm/files/8373060/itlwm_heilport.log)

liaozh0898, 2022-06-01:
我的 ax200 也会有时候没有网络 ,然后过一会又好了

liaozh0898, 2022-06-01:
我的是ax200网卡 ,版本是2.1 然后wifi有时候会断网 ,过个大概一两分钟会连上
连接新的wifi的时候需要去设置里面关闭wifi重新打开才有网

zxystd, 2024-03-16:
Inactive.

closed 2024-03-16 by zxystd