OpenIntelWireless/itlwm #767: Panic

jief666, 2022-04-14:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: 2.1
- WiFi Card Model: How to get that ?
- PCI Product ID: How to get that ?
- macOS Version: 10.14

**Description**
Crash, panic.
I was connected to my iphone in hotspot mode. I turned the hotspot off (on the iphone side). Immediate panic.

**Bug Report Archive**
Anonymous UUID:       9AAF6F34-057B-386C-FBC4-96F6BB5D28F9

Thu Apr 14 17:07:10 2022

*** Panic Report ***
panic(cpu 2 caller 0xffffff7f8cd36ae6): itlwm: nic locks counter 0
Backtrace (CPU 2), Frame : Return Address
0xffffff8206fbb830 : 0xffffff80097af58d 
0xffffff8206fbb880 : 0xffffff80098eb145 
0xffffff8206fbb8c0 : 0xffffff80098dc87a 
0xffffff8206fbb930 : 0xffffff800975c9d0 
0xffffff8206fbb950 : 0xffffff80097aefa7 
0xffffff8206fbba70 : 0xffffff80097aedf3 
0xffffff8206fbbae0 : 0xffffff7f8cd36ae6 
0xffffff8206fbbb00 : 0xffffff7f8cd36d36 
0xffffff8206fbbb40 : 0xffffff7f8ccd2442 
0xffffff8206fbbc40 : 0xffffff7f8ccd2986 
0xffffff8206fbbce0 : 0xffffff7f8ccd2b8d 
0xffffff8206fbbd50 : 0xffffff7f8cd282cf 
0xffffff8206fbbdc0 : 0xffffff7f8cd2adcf 
0xffffff8206fbbe40 : 0xffffff7f8cd1472f 
0xffffff8206fbbfa0 : 0xffffff800975c0ce 
      Kernel Extensions in backtrace:
         com.zxystd.itlwm(2.1)[9EA21A95-A8E4-3ED3-976D-EB380BDD7073]@0xffffff7f8cc84000->0xffffff7f8dba3fff
            dependency: com.apple.iokit.IONetworkingFamily(3.4)[0271ECAA-A519-3C98-9881-F9F5F29581C2]@0xffffff7f8a931000
            dependency: com.apple.iokit.IOPCIFamily(2.9)[650862F9-9814-3BC5-8079-03D21CF4CD89]@0xffffff7f8a052000

BSD process name corresponding to current thread: kernel_task
Boot args: -v -no_compat_check igfxonln=1 -igfxlspcon -igfxtypec -cdfon alc-verbs=1 -alcdbg 

Mac OS version:
18G5033

Kernel version:
Darwin Kernel Version 18.7.0: Mon Apr 27 20:09:39 PDT 2020; root:xnu-4903.278.35~1/RELEASE_X86_64
Kernel UUID: 2C6A35B5-EEFD-317A-9A3D-901B9CC558CF
Kernel slide:     0x0000000009400000
Kernel text base: 0xffffff8009600000
__HIB  text base: 0xffffff8009500000
System model name: MacBookPro15,2 (Mac-827FB448E656EC26)

**Kext Download Source**
Downloaded from your releases.

zxystd, 2022-04-15:
Is it reproducible? Have you tried v2.2.0-alpha version? seems that something racing there, but I didn't met that.

> WiFi Card Model: How to get that ?
> PCI Product ID: How to get that ?

For these information, you can get it by `Hackintool` on `PCIe` category.

jief666, 2022-04-15:
Vendor 0x8086 (of course)
Device 3165
SubDevice 4410
Name Wireless 3165

I didn't try to reproduce because it's on my main computer. Don't like too many crashes on my main partition with all my data !

Was just FYI in case you think of something obvious. It's possible to keep on standby (close for now). I'm suppose to move to another computer. This one will be free for crash tests.

I have another issue. Let me open a new one.

zxystd, 2022-04-16:
Please try to reproduce it, I have two `ac3165` and didn't trigger this knid of panic, can you find a way to trigger it?

zxystd, 2022-04-27:
Have you check the latest alpha version?

jief666, 2022-04-27:
I haven't. Because my main problem is the other one.

jief666, 2022-04-27:
I know. That's why I opened 2 separated issues.
But I don't much time for this issue right now.

zxystd, 2024-03-16:
Resolved in v2.2.0/v2.3.0.

closed 2024-03-16 by zxystd