OpenIntelWireless/itlwm #768: Network disconnection. Slow ping time.

jief666, 2022-04-15:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: Airportitlwm 2.1
- WiFi Card Model: Wireless 3165
- PCI Product ID: 0x3165
- macOS Version: 10.14.6

**Description**
Network disconnection. Slow ping time.

**Bug Report Archive**
[Log_2022-04-15_23-28-57.log.zip](https://github.com/OpenIntelWireless/itlwm/files/8498338/Log_2022-04-15_23-28-57.log.zip)

**Kext Download Source**
Got from your releases

jief666, 2022-04-15:
Also : kext was loaded from terminal. Can't have it to load at boot when I put in Library/Extensions.
I'm gonna try itlwm.kext, loaded by clover.

jief666, 2022-04-15:
I've rebooted with itlwm.kext. This time, it connected at boot and stayed connected. Ping time are ok. Forgot to check BSSID.
I've turned wifi off and on. It re-connected easily, but ping time are bad. Maybe one router is bad and one is ok. On the macbook, I'm always connected to the good one. Here, when I select the other network with the same name, I got a disconnect and then it reconnect to the previous one.

jief666, 2022-04-15:
Ok. Did a bit more tests. I've noticed that there is more than 2 networks with the same name by using "airport -s". I have channel 1, 6 and 11.
The mac sometime change. Here, I am always connected to channel 1.

zxystd, 2022-04-16:
Work as expected.

closed 2022-04-16 by zxystd

jief666, 2022-04-16:
I wouldn't say that it works as expected. It works fine in Windows, which means the card can do it. It works fine on the macbook.
Today, maybe because it's more busy, I can't have it working more than 10 seconds.
The hackintosh only connect to channel 1. The mac never connect to channel 1. Only 6 and 11.
So yes, at 1am, I got it working fine, but I guess it was by chance and by low usage of wifi.

jief666, 2022-04-16:
It's been 9 times turning wifi on and off. Each time it worked for approx 10 seconds. Last try was 11.2 seconds. Sometime, the hackintosh tris to reconnect by itself. Again 10 seconds. Most of the time it never tries to re-connect.
Without another computer, I couldn't have written this message.
NOTE : I don't know if the channel matters or not.
That's why I wouldn't say it works as expected.

jief666, 2022-04-18:
Hi. Could we re-open this ?

zxystd, 2022-04-27:
@jief666 The best solution is not using 2.4Ghz band WiFi because the 2.4Ghz band wifi performance is bad, and the channel selection is depends on the router, not the driver. I have two ac3165 and they all work fine.

jief666, 2022-05-05:
Well, I have one and it doesn't work as expected.
I guess it's just easier to close issue that way than trying to fix the problem...

jief666, 2024-03-16:
I can't believe you never had a look on that issue ! You kept it close, stating it's working as expected when it's obviously false.
Very dismissive indeed.