OpenIntelWireless/itlwm #769: Unable to connect from the latest prerelease version

19521679, 2022-04-18:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: 2.2.0 - alpha
- WiFi Card Model: 9461
- PCI Product ID: 0x8086
- macOS Version: 12.3.0

**Description**
Cannot connect any wifi since i was updated commit number 570cc2d0c7903e9dac9f075bb02c9b97391a1cba

**Bug Report Archive**
Airportitlwm

**Kext Download Source**
from pre release

zxystd, 2022-04-18:
Please upload the running log.

19521679, 2022-04-18:
it shows wifi name but cant connect. there aren't any anomalies in the log.

zxystd, 2022-04-18:
https://docs.oiw.workers.dev/itlwm/Troubleshooting.html#runtime-logs

Here is the guide to get the running log.

zxystd, 2022-04-26:
@19521679  This issue is fixed, please use the latest alpha version to check.

closed 2022-05-15 by zxystd

19521679, 2022-05-20:
> @19521679 Hi?

sorry because I haven't used hackintosh for a long time, thanks for the fix.