OpenIntelWireless/itlwm #770: Download/Upload speed drop to 0, websites load very slowly

axemre, 2022-04-19:
**Have You Read Our Docs**
 Yes

**Are You Reporting A Bug**
 Yes

**Environment**

- Kext Version: AirportItlwm 2.1.0 stable
- WiFi Card Model: Wireless-AC 9260 160MHz (iwx)
- PCI Product ID: 0x2526
- macOS Version: Monterey 12.3.1

**Description**
 

Somehow, the download and upload speed drops to 0 and then increases again, this makes downloads and uploads extremely slow. And websites load very slowly as well. Don't know what could be wrong. 

**Bug Report Archive**

**Kext Download Source**
 Downloaded from GitHub under Releases

zxystd, 2022-04-19:
Test v2.2.0-alpha instead, and use 5GHz wifi as well. Also please attach log.

axemre, 2022-04-19:
> Test v2.2.0-alpha instead, and use 5GHz wifi as well. Also please attach log.

2.2.0 seems to do great here now, there are no drops, downloading a 5GB file now.

axemre, 2022-04-19:
It seems to be working just fine now. Thanks for the tip with using 2.2.0, it really worked. There's no drop and it doesn't panic anymore.

closed 2022-04-19 by axemre