OpenIntelWireless/itlwm #774: AirportItlwm connection problem to hidden networks

Helakres, 2022-04-26:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: AirportItlwm_v2.1.0
- WiFi Card Model: Intel AC 9560
- PCI Product ID: 
- macOS Version: Monterey 12.1

**Description**
Network hidden ssid is available. Can't connect to them. When I open the hidden ssid, there is no problem. It would be great if there is a method that we can solve the hidden ssid connection problem.

**Bug Report Archive**

**Kext Download Source**
https://github.com/OpenIntelWireless/itlwm/releases/download/v2.1.0/AirportItlwm_v2.1.0_stable_Monterey.kext.zip

zxystd, 2022-04-27:
See the doc.

closed 2022-04-27 by zxystd

Helakres, 2022-04-27:
What exactly are you talking about doctor?

rboldini, 2022-04-27:
[Doc](https://openintelwireless.github.io/itlwm/FAQ.html#features) means documentation.
Look for 'hidden'.