OpenIntelWireless/itlwm #776: [Request Test] Hidden Wi-Fi connection support for AirportItlwm

zxystd, 2022-05-15:
Hi, Today I paid some time to resolve this kind of issue #294 #771 and finally got it working, Now we can connect to the hidden Wi-Fi by click the wifi icon and then choose 'Join other network'.

Test ok with Catalina-ax201 by myself. Can anyone help to test also? Thanks!
Note: Please pay attention to whether there is a disconnection during scanning after installing this version of kext.

[High Sierra.zip](https://github.com/OpenIntelWireless/itlwm/files/8694761/High.Sierra.zip)
[Catalina.zip](https://github.com/OpenIntelWireless/itlwm/files/8694762/Catalina.zip)
[Big Sur.zip](https://github.com/OpenIntelWireless/itlwm/files/8694763/Big.Sur.zip)
[Mojave.zip](https://github.com/OpenIntelWireless/itlwm/files/8694764/Mojave.zip)
[Monterey.zip](https://github.com/OpenIntelWireless/itlwm/files/8694765/Monterey.zip)

qqsir-dev, 2022-05-15:
太牛了！！！大赞！！Awsome!!!!

yasapurnama, 2022-06-05:
tried, and it's working
but every reboot need to join manual again, seems the session is not persistent

alangsecond, 2022-06-27:
tried in Monterey,AX210,both 2.4G and 5G working,very good.
trid in Ventura beta2,non-hidden Wi-Fi connection working ,but hidden Wi-Fi connection not working
Can you resolve it ?Thanks!

trihutama, 2022-08-24:
tested on 3168NGW monterey 12.5.1, connecting to hidden Wi-Fi working normal

jinkunyyy, 2022-09-05:
AX210 AX working on Monterey 12.5(21G72),connecting to hidden Wi-Fi

javasoul95, 2022-09-10:
Hello, Could you fix the hidden network for ventura?

closed 2023-02-15 by zxystd

jinkunyyy, 2023-08-14:
> Hi, Today I paid some time to resolve this kind of issue #294 #771 and finally got it working, Now we can connect to the hidden Wi-Fi by click the wifi icon and then choose 'Join other network'. [...]

大佬什么时候可以修复一下Ventura的隐藏无线不能连接的问题，万分感谢！

Abdullah-Albanna, 2023-10-31:
works like a charm on Big Sur with Intel Centrino Advanced-N 6205, connects to a hidden networks and remembers it.

thank you so much

haol666, 2024-12-07:
working on Monterey 12.5.1 , connecting to hidden Wi-Fi