OpenIntelWireless/itlwm #777: Intel Bluetooth macOS Monterey 12.4

Torstenp122, 2022-05-19:
With Mac OS 12.3.1 Bluetooth worked and now with update to 12.4 Bluetooth does not work anymore.

Core-i99, 2022-05-19:
Try this maybe.
https://github.com/acidanthera/BrcmPatchRAM/pull/20

Torstenp122, 2022-05-19:
What must I do? Sorry

Core-i99, 2022-05-19:
Use bluetoolfixup from here:
https://github.com/acidanthera/BrcmPatchRAM/suites/6569025865/artifacts/245840661

I'm unsure if it will fix the problem, but always worth trying.

williambj1, 2022-05-19:
User space bluetooth has nothing to do with a wifi driver.

I'm using 9560ac Bluetooth with 12.4 right now, works as expected.

Issue opened with zero details, closing as invalid.

closed 2022-05-19 by williambj1

fanlasila, 2022-05-19:
> 从这里使用 bluetoolfixup： [https ://github.com/acidanthera/BrcmPatchRAM/suites/6569025865/artifacts/245840661](https://github.com/acidanthera/BrcmPatchRAM/suites/6569025865/artifacts/245840661)
> 
> 我不确定它是否能解决问题，但总是值得一试。

well done

rafaelbrunner, 2022-06-07:
Still not working, even with latest BlueToolFixup... :( with Intel AC 3165

djsnakezinho, 2022-06-07:
Hi
I was in your situation too since 12.4 update (X1Carbon 7th with Intel AC 9560).
I replaced BlueToolFixup.kext ([https://github.com/acidanthera/BrcmPatchRAM/releases](https://github.com/acidanthera/BrcmPatchRAM/releases)) with the latest release from yesterday and now it came back to life :)