OpenIntelWireless/itlwm #779: AX210 there is no wifi after waking up in battery mode

taruyato, 2022-05-28:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: 2.2.0 alpha commit [73026f7](https://github.com/OpenIntelWireless/itlwm/commit/73026f736c3a9544f78f1c5472ad56e895c368a0)
- WiFi Card Model: Intel 6E AX210
- PCI Product ID: 0x2725
- macOS Version: Big Sur 11.6.6

**Description**
My wifi is lost and doesn't find any networks after waking up in battery mode, I have to restart to make it work again. If I plug the power adapter, it works correctly after waking up and no issue.
I also tried to test with my CNVI 9560 and everything is fine.
Laptop spec: Dell G7 7588, i7-8750H

**Bug Report Archive**
Here are 2 logs with 9 hrs sleeping, one is normal with coldboot, one is after waking up.
[Log_2022-05-28_09-35-24.log](https://github.com/OpenIntelWireless/itlwm/files/8791201/Log_2022-05-28_09-35-24.log)

[Log_2022-05-28_18-45-28.log](https://github.com/OpenIntelWireless/itlwm/files/8791203/Log_2022-05-28_18-45-28.log)

zxystd, 2022-05-30:
@taruyato  Is it reproducible and happened every time? Is it works fine with v2.1.0? Seems like it is an hackintosh related issue, I have ax210 and everything works fine. But you can try this one kext and see if it is helpful.
[Big Sur.zip](https://github.com/OpenIntelWireless/itlwm/files/8797316/Big.Sur.zip)

taruyato, 2022-05-30:
> @taruyato  Is it reproducible and happened every time? Is it works fine with v2.1.0? Seems like it is an hackintosh related issue, I have ax210 and everything works fine. But you can try this one [...]

Thank you for your reply. It happens everytimes in battery mode.
Sry I haven't tried 2.1.0 yet 😅. Will try it and your version^ asap and tell you the result.

taruyato, 2022-05-30:
@zxystd I have just tested and seems 2.1.0 is working fine with both battery and power adapter mode. The version which you sent me doesn't, same result with 2.2.0 alpha commit 73026f7.

zxystd, 2022-05-31:
@taruyato Please also attach the two logs, I can't reproduce the problem. Have you tried toggling the WiFi switch to see if the wifi will work again?

taruyato, 2022-05-31:
> @taruyato Please also attach the two logs, I can't reproduce the problem. Have you tried toggling the WiFi switch to see if the wifi will work again?

Oh ya, I have tried toggling the switch. 2.1.0 works fine, your version doesn't.

Here are 2 logs I have just dumped:
[zxystd's version Coldboot.log](https://github.com/OpenIntelWireless/itlwm/files/8801322/zxystd.s.version.Coldboot.log)
[zxystd's version After wake.log](https://github.com/OpenIntelWireless/itlwm/files/8801323/zxystd.s.version.After.wake.log)

Also 2.1.0 logs if you need:
[2.1.0 Coldboot.log](https://github.com/OpenIntelWireless/itlwm/files/8801325/2.1.0.Coldboot.log)
[2.1.0 After wake.log](https://github.com/OpenIntelWireless/itlwm/files/8801326/2.1.0.After.wake.log)

taruyato, 2022-05-31:
> Oh ya, I have tried toggling the switch. 2.1.0 works fine, your version doesn't.
>
> Here are 2 logs I have just dumped: [zxystd's version [...]

oh wait... I have just toggled the switch again but it took maybe a minute then the wifi is back now.

taruyato, 2022-05-31:
I have just tried commit bdbbdfc and seems like it works correctly after waking up. No need to toggle the switch.
Here are 2 logs if you want to check:
[2.2.0 bdbbdfc Coldboot.log](https://github.com/OpenIntelWireless/itlwm/files/8801427/2.2.0.bdbbdfc.Coldboot.log)
[2.2.0 bdbbdfc After wake.log](https://github.com/OpenIntelWireless/itlwm/files/8801428/2.2.0.bdbbdfc.After.wake.log)

q394140505, 2022-06-12:
Ax210会增加隔空投送和随行、接力的支持吗？

taruyato, 2022-06-12:
> Ax210会增加隔空投送和随行、接力的支持吗？

Airdrop: 没有
Handoff: 是的

nybrum, 2022-07-28:
你好
请问AX210速度如何？
谢谢

> > Ax210会增加隔空投送和随行、接力的支持吗？
> 
> Airdrop: 没有 Handoff: 是的

closed 2023-05-29 by zxystd