OpenIntelWireless/itlwm #781: Airportitlwm_v2.1.0_Monterey.kext cause wake with unknown reason and exhaust battery life

cguo2013, 2022-06-01:
Hi there, I have successfully install Monterey 12.4 on my Dell laptop Precision 5520, I use the native intel dual band WiFi 8265 and copy Airportitlwm_v2.1.0_Monterey.kext in my OC kext folder and enabled in config.plist, and set Misc->Security->SecureBootModel=default, it works find until I found my laptop often wake up for unknown reasons and result in battery exhaust. I am pretty sure the wake up is not caused by USB or other reasons, please check below log. 

Any idea on what may cause such wake up?

cguo@cguos-MBP ~ % log show --last 15h |grep -i "wake reason"
2022-05-31 21:39:20.346317+0800 0x1ff1b7   Error       0x0                  109    0    powerd: [powerd:sleepWake] PID 158 is not entitled to set wake reason
2022-05-31 21:39:20.347574+0800 0x1fe5e6   Default     0x0                  98     0    UserEventAgent: (PerfPowerServicesEventListenerPlugin) Could not get wake reason (errno: 2)
2022-05-31 23:23:52.491634+0800 0x216872   Default     0x0                  192    0    airportd: systemWokenByWiFi: System Wake Reason not found
2022-05-31 23:23:52.504215+0800 0x216872   Default     0x0                  192    0    airportd: systemWokenByWiFi: System Wake Reason not found
2022-05-31 23:23:58.980617+0800 0x21746a   Default     0x0                  192    0    airportd: systemWokenByWiFi: System Wake Reason not found
2022-06-01 00:21:54.003658+0800 0x224e80   Default     0x0                  192    0    airportd: systemWokenByWiFi: System Wake Reason not found
2022-06-01 00:21:54.297480+0800 0x224e80   Default     0x0                  192    0    airportd: systemWokenByWiFi: System Wake Reason not found
2022-06-01 00:21:54.310375+0800 0x224e80   Default     0x0                  192    0    airportd: systemWokenByWiFi: System Wake Reason not found
2022-06-01 00:22:03.603111+0800 0x2253fa   Default     0x0                  192    0    airportd: systemWokenByWiFi: System Wake Reason not found
2022-06-01 00:23:00.078130+0800 0x2258e8   Default     0x0                  192    0    airportd: systemWokenByWiFi: System Wake Reason not found
2022-06-01 00:23:00.151237+0800 0x2258e8   Default     0x0                  192    0    airportd: systemWokenByWiFi: System Wake Reason not found
2022-06-01 00:24:12.108864+0800 0x225d86   Default     0x0                  192    0    airportd: systemWokenByWiFi: System Wake Reason not found
2022-06-01 00:24:16.241627+0800 0x225d86   Default     0x0                  192    0    airportd: systemWokenByWiFi: System Wake Reason not found
2022-06-01 00:24:21.253451+0800 0x225d88   Default     0x0                  192    0    airportd: systemWokenByWiFi: System Wake Reason not found
2022-06-01 00:25:27.333820+0800 0x2262f2   Default     0x0                  192    0    airportd: systemWokenByWiFi: System Wake Reason not found
2022-06-01 00:25:28.221537+0800 0x2262f7   Default     0x0                  192    0    airportd: systemWokenByWiFi: System Wake Reason not found
2022-06-01 00:26:38.486185+0800 0x2267d2   Default     0x0                  192    0    airportd: systemWokenByWiFi: System Wake Reason not found
2022-06-01 00:26:39.373201+0800 0x2267d0   Default     0x0                  192    0    airportd: systemWokenByWiFi: System Wake Reason not found
2022-06-01 00:27:49.616125+0800 0x226cbf   Default     0x0                  192    0    airportd: systemWokenByWiFi: System Wake Reason not found
2022-06-01 00:27:50.525617+0800 0x226cc1   Default     0x0                  192    0    airportd: systemWokenByWiFi: System Wake Reason not found
2022-06-01 03:20:37.360631+0800 0x25015c   Default     0x0                  192    0    airportd: systemWokenByWiFi: System Wake Reason not found
2022-06-01 03:20:41.075528+0800 0x25015c   Default     0x0                  192    0    airportd: systemWokenByWiFi: System Wake Reason not found
2022-06-01 03:20:46.078547+0800 0x250159   Default     0x0                  192    0    airportd: systemWokenByWiFi: System Wake Reason not found
2022-06-01 03:21:53.054752+0800 0x2507df   Default     0x0                  192    0    airportd: systemWokenByWiFi: System Wake Reason not found
2022-06-01 03:21:57.206219+0800 0x2507df   Default     0x0                  192    0    airportd: systemWokenByWiFi: System Wake Reason not found
2022-06-01 03:22:02.210606+0800 0x2507df   Default     0x0                  192    0    airportd: systemWokenByWiFi: System Wake Reason not found
2022-06-01 03:23:04.642831+0800 0x250d79   Default     0x0                  192    0    airportd: systemWokenByWiFi: System Wake Reason not found
2022-06-01 03:23:09.814105+0800 0x250cf2   Default     0x0                  192    0    airportd: systemWokenByWiFi: System Wake Reason not found
2022-06-01 03:23:56.308885+0800 0x250cf2   Default     0x0                  192    0    airportd: systemWokenByWiFi: System Wake Reason not found
2022-06-01 03:24:01.275869+0800 0x250cf2   Default     0x0                  192    0    airportd: systemWokenByWiFi: System Wake Reason not found
2022-06-01 03:24:07.268783+0800 0x2511b1   Default     0x0                  192    0    airportd: systemWokenByWiFi: System Wake Reason not found
2022-06-01 03:24:17.213560+0800 0x2511b0   Default     0x0                  192    0    airportd: systemWokenByWiFi: System Wake Reason not found
2022-06-01 03:24:37.156204+0800 0x2514aa   Default     0x0                  192    0    airportd: systemWokenByWiFi: System Wake Reason not found
2022-06-01 03:25:12.117296+0800 0x2514c1   Default     0x0                  192    0    airportd: systemWokenByWiFi: System Wake Reason not found
2022-06-01 03:26:17.080735+0800 0x251a18   Default     0x0                  192    0    airportd: systemWokenByWiFi: System Wake Reason not found
2022-06-01 03:28:22.045301+0800 0x252263   Default     0x0                  192    0    airportd: systemWokenByWiFi: System Wake Reason not found
2022-06-01 03:33:27.023002+0800 0x253303   Default     0x0                  192    0    airportd: systemWokenByWiFi: System Wake Reason not found
2022-06-01 03:43:32.003784+0800 0x25572a   Default     0x0                  192    0    airportd: systemWokenByWiFi: System Wake Reason not found
2022-06-01 03:53:37.006735+0800 0x257dd2   Default     0x0                  192    0    airportd: systemWokenByWiFi: System Wake Reason not found
2022-06-01 04:03:41.995747+0800 0x259fa4   Default     0x0                  192    0    airportd: systemWokenByWiFi: System Wake Reason not found
2022-06-01 04:13:46.985194+0800 0x25c400   Default     0x0                  192    0    airportd: systemWokenByWiFi: System Wake Reason not found
2022-06-01 04:23:51.979935+0800 0x25e81a   Default     0x0                  192    0    airportd: systemWokenByWiFi: System Wake Reason not found
2022-06-01 04:33:56.959069+0800 0x260c5b   Default     0x0                  192    0    airportd: systemWokenByWiFi: System Wake Reason not found
2022-06-01 04:44:01.974766+0800 0x2634b6   Default     0x0                  192    0    airportd: systemWokenByWiFi: System Wake Reason not found
2022-06-01 04:54:06.964168+0800 0x2658bf   Default     0x0                  192    0    airportd: systemWokenByWiFi: System Wake Reason not found
2022-06-01 05:04:11.958993+0800 0x267d3a   Default     0x0                  192    0    airportd: systemWokenByWiFi: System Wake Reason not found
2022-06-01 05:14:16.943593+0800 0x26a2a2   Default     0x0                  192    0    airportd: systemWokenByWiFi: System Wake Reason not found
2022-06-01 05:24:21.944658+0800 0x26c6f6   Default     0x0                  192    0    airportd: systemWokenByWiFi: System Wake Reason not found
2022-06-01 05:34:26.928470+0800 0x26ea4c   Default     0x0                  192    0    airportd: systemWokenByWiFi: System Wake Reason not found
2022-06-01 05:44:31.944169+0800 0x271173   Default     0x0                  192    0    airportd: systemWokenByWiFi: System Wake Reason not found
2022-06-01 05:54:36.937400+0800 0x2733f7   Default     0x0                  192    0    airportd: systemWokenByWiFi: System Wake Reason not found
2022-06-01 06:04:41.940719+0800 0x2758ab   Default     0x0                  192    0    airportd: systemWokenByWiFi: System Wake Reason not found
2022-06-01 06:14:46.927536+0800 0x277d61   Default     0x0                  192    0    airportd: systemWokenByWiFi: System Wake Reason not found
[... cut ...]
2022-06-01 09:57:06.057541+0800 0x2aaf09   Default     0x0                  192    0    airportd: systemWokenByWiFi: System Wake Reason not found
2022-06-01 09:57:06.090295+0800 0x2aaf09   Default     0x0                  192    0    airportd: systemWokenByWiFi: System Wake Reason not found
2022-06-01 09:57:06.133626+0800 0x38e      Default     0x0                  192    0    airportd: systemWokenByWiFi: System Wake Reason not found
2022-06-01 09:57:06.142862+0800 0x2a90bd   Default     0x0                  98     0    UserEventAgent: (PerfPowerServicesEventListenerPlugin) Could not get wake reason (errno: 2)
cguo@cguos-MBP ~ %

williambj1, 2022-06-01:
Do you have App Store or System updates running in the background? I can see those logs in my laptop as well but sleep works as expected. Feel free to add more comments if the issue disappears when AirportItlwm is removed.

closed 2022-06-01 by williambj1

cguo2013, 2022-06-02:
> Do you have App Store or System updates running in the background? I can see those logs in my laptop as well but sleep works as expected. Feel free to add more comments if the issue disappears when AirportItlwm is removed.

No， there is no App Store or system update running in the background, I test it for several days and all come up with same type of logs.

williambj1, 2022-06-02:
Have you tried disabling Wake for network access and power nap? Does the issue still exist if AirportItlwm is replaced with ethernet?

cguo2013, 2022-06-02:
> Have you tried disabling Wake for network access and power nap? Does the issue still exist if AirportItlwm is replaced with ethernet?

yes wake from network access and power nap are both unchecked in Monterey, WOL also unchecked in BIOS. I haven't try ethernet, but provide me an idea so I will try to turn off WiFi and see if it help on wake up problem.