OpenIntelWireless/itlwm #782: 2.2.0版本睡眠后死机

hagedr, 2022-06-04:
各位大佬辛苦了，感谢你们让我抛弃了usb网卡。不过使用过程中遇到了一点问题。
2.1.0版本有时会断网重连
2.2.0版本没问题，但不知道从2.2.0的哪个版本开始，出现了睡眠死机的问题
下面是死机后开机时提示的错误日志。我也不确定是不是网卡引起的，只是看样子有点像

panic(cpu 0 caller 0xffffff8010d1ab61): zfree_cached: double free of 0xffffff950cf1b9a0 to zone default.kalloc.32 @zalloc.c:858
Panicked task 0xffffff950d1cf670: 195 threads: pid 0: kernel_task
Backtrace (CPU 0), panicked thread: 0xffffff99d7a94000, Frame : Return Address
0xffffffd08122b680 : 0xffffff8010483e2d mach_kernel : _handle_debugger_trap + 0x41d
0xffffffd08122b6d0 : 0xffffff80105e3cb6 mach_kernel : _kdp_i386_trap + 0x116
0xffffffd08122b710 : 0xffffff80105d350d mach_kernel : _kernel_trap + 0x51d
0xffffffd08122b760 : 0xffffff8010423a60 mach_kernel : _return_from_trap + 0xe0
0xffffffd08122b780 : 0xffffff80104841fd mach_kernel : _DebuggerTrapWithState + 0xad
0xffffffd08122b8a0 : 0xffffff80104839b6 mach_kernel : _panic_trap_to_debugger + 0x2b6
0xffffffd08122b900 : 0xffffff8010d164bf mach_kernel : _panic + 0x84
0xffffffd08122b9f0 : 0xffffff8010d1ab61 mach_kernel : _panic_with_thread_context + 0x45d9
0xffffffd08122ba20 : 0xffffff80104ee98d mach_kernel : _work_interval_port_type_render_server + 0x533d
0xffffffd08122bac0 : 0xffffff8010494c1d mach_kernel : _kfree + 0x1bd
0xffffffd08122bb30 : 0xffffff8010c046ea mach_kernel : _IOFree + 0x2a
0xffffffd08122bb50 : 0xffffff80146b9ea7 com.zxystd.AirportItlwm : __Z22ieee80211_node_cleanupP12ieee80211comP14ieee80211_node + 0x57
0xffffffd08122bb70 : 0xffffff80146b953d com.zxystd.AirportItlwm : __Z19ieee80211_node_freeP12ieee80211comP14ieee80211_node + 0x1d
0xffffffd08122bb90 : 0xffffff80146bc07e com.zxystd.AirportItlwm : __Z19ieee80211_free_nodeP12ieee80211comP14ieee80211_node + 0x16e
0xffffffd08122bbc0 : 0xffffff80146b9b8b com.zxystd.AirportItlwm : __Z23ieee80211_free_allnodesP12ieee80211comi + 0x13b
0xffffffd08122bc00 : 0xffffff8014690e9a com.zxystd.AirportItlwm : __Z18ieee80211_newstateP12ieee80211com15ieee80211_statei + 0x25a
0xffffffd08122bca0 : 0xffffff80146fa92d com.zxystd.AirportItlwm : __ZN6ItlIwx8iwx_stopEP6_ifnet + 0x32d
0xffffffd08122bcf0 : 0xffffff80146e04c0 com.zxystd.AirportItlwm : __ZN6ItlIwx12iwx_activateEP9iwx_softci + 0x90
0xffffffd08122bd30 : 0xffffff80146e05ca com.zxystd.AirportItlwm : __ZN6ItlIwx7disableEP18IONetworkInterface + 0x7a
0xffffffd08122bd60 : 0xffffff8014741ba4 com.zxystd.AirportItlwm : __ZN12AirportItlwm14disableAdapterEP18IONetworkInterface + 0x54
0xffffffd08122bd90 : 0xffffff8014742c3a com.zxystd.AirportItlwm : __ZN12AirportItlwm16setPowerStateOffEv + 0x4a
0xffffffd08122bdb0 : 0xffffff8014742b5c com.zxystd.AirportItlwm : __ZL22handleSetPowerStateOffPvS_ + 0x6c
0xffffffd08122bde0 : 0xffffff8010c48a64 mach_kernel : __ZN13IOCommandGate9runActionEPFiP8OSObjectPvS2_S2_S2_ES2_S2_S2_S2_ + 0xb4
0xffffffd08122be40 : 0xffffff8014742b4e com.zxystd.AirportItlwm : __ZL22handleSetPowerStateOffPvS_ + 0x5e
0xffffffd08122be70 : 0xffffff80104d71f5 mach_kernel : _thread_call_delayed_timer + 0x505
0xffffffd08122bee0 : 0xffffff80104d82c2 mach_kernel : _thread_call_delayed_timer + 0x15d2
0xffffffd08122bfa0 : 0xffffff801042318e mach_kernel : _call_continuation + 0x2e
      Kernel Extensions in backtrace:
         com.zxystd.AirportItlwm(2.2)[6BD2E31D-AF25-3190-B7AA-657357727973]@0xffffff801468c000->0xffffff80155d3fff
            dependency: com.apple.iokit.IO80211FamilyLegacy(1200.12.2b1)[3D7E24FF-CEF0-3208-91F9-7FF6B620FCB5]@0xffffff8012210000->0xffffff8012354fff
            dependency: com.apple.iokit.IONetworkingFamily(3.4)[62888EF4-277C-35B5-B9FA-DF8008BC5D28]@0xffffff8012d6b000->0xffffff8012d81fff
            dependency: com.apple.iokit.IOPCIFamily(2.9)[A436E92C-DE10-3718-AEF4-ED2A788A466A]@0xffffff8013003000->0xffffff801302efff

Process name corresponding to current thread (0xffffff99d7a94000): kernel_task
Boot args: keepsyms=1 swd_panic=1 alcid=21 igfxonln=1 igfxrpsc=1 igfxfw=2 -igfxblr -wegnoegpu -igfxbls debug=0x100

Mac OS version:
21E258

Kernel version:
Darwin Kernel Version 21.4.0: Fri Mar 18 00:45:05 PDT 2022; root:xnu-8020.101.4~15/RELEASE_X86_64
Kernel UUID: B6F8637B-0844-355F-8C82-60FA06149384
KernelCache slide: 0x0000000010200000
KernelCache base:  0xffffff8010400000
Kernel slide:      0x0000000010210000
Kernel text base:  0xffffff8010410000
__HIB  text base: 0xffffff8010300000
System model name: MacBookPro16,2 (Mac-5F9802EFE386AA28)
System shutdown begun: NO
Panic diags file available: YES (0x0)
Hibernation exit count: 0

System uptime in nanoseconds: 2479194345188
Last Sleep:           absolute           base_tsc          base_nano
  Uptime  : 0x000002413b7f923e
  Sleep   : 0x00000235dcca41a5 0x000000007b71b664 0x0000022a7aa977e1
  Wake    : 0x00000235ec0433e3 0x000000007d0df9b6 0x00000235e8ccc44f
Compressor Info: 0% of compressed pages limit (OK) and 0% of segments limit (OK) with 0 swapfiles and OK swap space
Zone info:
  Foreign : 0xffffff8028944000 - 0xffffff8028952000
  Native  : 0xffffff8040567000 - 0xffffffa040567000
  Readonly: 0xffffff850d233000 - 0xffffff86a6bcc000
  Metadata: 0xffffffd5d6be1000 - 0xffffffd5f6d5e000
  Bitmaps : 0xffffffd5f6d5e000 - 0xffffffd5fcd5e000

williambj1, 2022-06-04:
每次睡眠都能复现吗

hagedr, 2022-06-04:
> 每次睡眠都能复现吗

不是，暂时还没发现规律，我再看看。

GSJv2-1, 2022-06-12:
This exact issue occurs when the pc is in sleep mode over night.  

panic(cpu 0 caller 0xffffff800d516abb): kfree: addr 0xffffffa57ce9c420, size 113 (offs:0) found in zone 'default.kalloc.32' with elem_size 32 @kalloc.c:2164
Panicked task 0xffffff9be44bc670: 241 threads: pid 0: kernel_task
Backtrace (CPU 0), panicked thread: 0xffffffa0b3733aa0, Frame : Return Address
0xffffffe59c6b3700 : 0xffffff800cc81c8d mach_kernel : _handle_debugger_trap + 0x41d
0xffffffe59c6b3750 : 0xffffff800cde1596 mach_kernel : _kdp_i386_trap + 0x116
0xffffffe59c6b3790 : 0xffffff800cdd0963 mach_kernel : _kernel_trap + 0x4d3
0xffffffe59c6b37e0 : 0xffffff800cc21a70 mach_kernel : _return_from_trap + 0xe0
0xffffffe59c6b3800 : 0xffffff800cc8205d mach_kernel : _DebuggerTrapWithState + 0xad
0xffffffe59c6b3920 : 0xffffff800cc81816 mach_kernel : _panic_trap_to_debugger + 0x2b6
0xffffffe59c6b3980 : 0xffffff800d515163 mach_kernel : _panic + 0x84
0xffffffe59c6b3a70 : 0xffffff800d516abb mach_kernel : _panic_with_thread_context + 0x1883
0xffffffe59c6b3ad0 : 0xffffff800cc91eba mach_kernel : _kalloc_type_var_impl + 0xda
0xffffffe59c6b3b10 : 0xffffff800d40259b mach_kernel : _IOFree + 0x1b
0xffffffe59c6b3b30 : 0xffffff8011535667 com.zxystd.AirportItlwm : __Z22ieee80211_node_cleanupP12ieee80211comP14ieee80211_node + 0x57
0xffffffe59c6b3b50 : 0xffffff8011534bdd com.zxystd.AirportItlwm : __Z19ieee80211_node_freeP12ieee80211comP14ieee80211_node + 0x1d
0xffffffe59c6b3b70 : 0xffffff801153842d com.zxystd.AirportItlwm : __Z19ieee80211_free_nodeP12ieee80211comP14ieee80211_node + 0x25d
0xffffffe59c6b3bb0 : 0xffffff801153532b com.zxystd.AirportItlwm : __Z23ieee80211_free_allnodesP12ieee80211comi + 0x13b
0xffffffe59c6b3bf0 : 0xffffff801150492a com.zxystd.AirportItlwm : __Z18ieee80211_newstateP12ieee80211com15ieee80211_statei + 0x39a
0xffffffe59c6b3ca0 : 0xffffff8011579f1d com.zxystd.AirportItlwm : __ZN6ItlIwx8iwx_stopEP6_ifnet + 0x32d
0xffffffe59c6b3cf0 : 0xffffff801155fb10 com.zxystd.AirportItlwm : __ZN6ItlIwx12iwx_activateEP9iwx_softci + 0x90
0xffffffe59c6b3d30 : 0xffffff801155fc1a com.zxystd.AirportItlwm : __ZN6ItlIwx7disableEP18IONetworkInterface + 0x7a
0xffffffe59c6b3d60 : 0xffffff80115c1174 com.zxystd.AirportItlwm : __ZN12AirportItlwm14disableAdapterEP18IONetworkInterface + 0x54
0xffffffe59c6b3d90 : 0xffffff80115c220a com.zxystd.AirportItlwm : __ZN12AirportItlwm16setPowerStateOffEv + 0x4a
0xffffffe59c6b3db0 : 0xffffff80115c212c com.zxystd.AirportItlwm : __ZL22handleSetPowerStateOffPvS_ + 0x6c
0xffffffe59c6b3de0 : 0xffffff800d446a14 mach_kernel : __ZN13IOCommandGate9runActionEPFiP8OSObjectPvS2_S2_S2_ES2_S2_S2_S2_ + 0xb4
0xffffffe59c6b3e40 : 0xffffff80115c211e com.zxystd.AirportItlwm : __ZL22handleSetPowerStateOffPvS_ + 0x5e
0xffffffe59c6b3e70 : 0xffffff800ccd43a5 mach_kernel : _thread_call_delayed_timer + 0x505
0xffffffe59c6b3ee0 : 0xffffff800ccd5472 mach_kernel : _thread_call_delayed_timer + 0x15d2
0xffffffe59c6b3fa0 : 0xffffff800cc2119e mach_kernel : _call_continuation + 0x2e
      Kernel Extensions in backtrace:
         com.zxystd.AirportItlwm(2.2)[2DF170CF-7085-3FB1-B5E6-02ECA70D28ED]@0xffffff8011500000->0xffffff8012457fff
            dependency: com.apple.iokit.IO80211FamilyLegacy(1200.12.2b1)[D0195701-5F96-3013-AEF8-0E8523F35807]@0xffffff800ea2a000->0xffffff800eb6ffff
            dependency: com.apple.iokit.IONetworkingFamily(3.4)[EDA34629-E451-3585-BD20-C07343CEAAD3]@0xffffff800f586000->0xffffff800f59cfff
            dependency: com.apple.iokit.IOPCIFamily(2.9)[654ABF9D-EB24-3B64-AB1C-EC2AC8357192]@0xffffff800f824000->0xffffff800f84ffff

Process name corresponding to current thread (0xffffffa0b3733aa0): kernel_task
Boot args: debug=0x100 keepsyms=1 -igfxblr igfxfw=2 -igfxmpc forceRenderStandby=0 swd_panic=1 -noDC9 -igfxnohdmi -igfxbls -igfxcdc -igfxdbeo chunklist-security-epoch=0 -chunklist-no-rev2-dev

Mac OS version:
21F79

Kernel version:
Darwin Kernel Version 21.5.0: Tue Apr 26 21:08:22 PDT 2022; root:xnu-8020.121.3~4/RELEASE_X86_64
Kernel UUID: 3C587984-4004-3C76-8ADF-997822977184
KernelCache slide: 0x000000000ca00000
KernelCache base:  0xffffff800cc00000
Kernel slide:      0x000000000ca10000
Kernel text base:  0xffffff800cc10000
__HIB  text base: 0xffffff800cb00000
System model name: MacBookAir9,1 (Mac-0CFF9C7C2B63DF8D)
System shutdown begun: NO
Panic diags file available: YES (0x0)
Hibernation exit count: 0

System uptime in nanoseconds: 34766931828129
Last Sleep:           absolute           base_tsc          base_nano
  Uptime  : 0x00001f9ece9f225e
  Sleep   : 0x00001f90e82900ec 0x00002f3d3d46d23b 0x00001f8110650808
  Wake    : 0x00001f90e8ef09c3 0x00002f54f7852c83 0x00001f90e85f814a
Compressor Info: 0% of compressed pages limit (OK) and 0% of segments limit (OK) with 0 swapfiles and OK swap space
Zone info:
  Zone map: 0xffffff90afb9e000 - 0xffffffb0afb9e000
  . PGZ   : 0xffffff90afb9e000 - 0xffffff90b1b9f000
  . VM    : 0xffffff90b1b9f000 - 0xffffff957e39e000
  . RO    : 0xffffff957e39e000 - 0xffffff9717b9e000
  . GEN0  : 0xffffff9717b9e000 - 0xffffff9be439e000
  . GEN1  : 0xffffff9be439e000 - 0xffffffa0b0b9e000
  . GEN2  : 0xffffffa0b0b9e000 - 0xffffffa57d39e000
  . GEN3  : 0xffffffa57d39e000 - 0xffffffaa49b9e000
  . DATA  : 0xffffffaa49b9e000 - 0xffffffb0afb9e000
  Metadata: 0xffffff80289fe000 - 0xffffff80489fe000
  Bitmaps : 0xffffff80489fe000 - 0xffffff804e9fe000

thedycui, 2023-05-31:
我也碰到这个问题了
[image]

thedycui, 2023-06-06:
@zxystd  AX211

thedycui, 2023-06-06:
@zxystd  稳定版暂时没遇到。

zzwtsy, 2023-06-09:
我在更新到 2.2.0 后出现了开机卡死或黑屏的问题。
网卡型号：AC 9260 (0x8086, 0x14)
使用的 kext: AirportItlwm
<details>
<summary>
问题信息：
</summary>
panic(cpu 0 caller 0xffffff801003900e): Kernel stack memory corruption detected @stack_protector.c:37
Panicked task 0xffffff9503474380: 18 threads: pid 225: airportd
Backtrace (CPU 0), panicked thread: 0xffffff95011caaa0, Frame : Return Address
0xfffffff58702b710 : 0xffffff801007cead mach_kernel : _handle_debugger_trap + 0x41d
0xfffffff58702b760 : 0xffffff80101df5d6 mach_kernel : _kdp_i386_trap + 0x116
0xfffffff58702b7a0 : 0xffffff80101ce943 mach_kernel : _kernel_trap + 0x4d3
0xfffffff58702b7f0 : 0xffffff801001ca70 mach_kernel : _return_from_trap + 0xe0
0xfffffff58702b810 : 0xffffff801007d27d mach_kernel : _DebuggerTrapWithState + 0xad
0xfffffff58702b930 : 0xffffff801007ca36 mach_kernel : _panic_trap_to_debugger + 0x2b6
0xfffffff58702b990 : 0xffffff8010914ed3 mach_kernel : _panic + 0x84
0xfffffff58702ba80 : 0xffffff801003900e mach_kernel : ___stack_chk_fail + 0x1e
0xfffffff58702ba90 : 0xffffff8011eb69ba com.apple.iokit.IO80211FamilyLegacy : __ZL8getStatsP17IO80211ControllerP16IO80211InterfaceP23IO80211VirtualInterfaceP21IO80211InfraInterfaceP13apple80211reqb
0xfffffff58702baf0 : 0xffff1b09ee1a 
Unaligned frame
Backtrace terminated-invalid frame pointer 0x2d00000000710546
      Kernel Extensions in backtrace:
         com.apple.iokit.IO80211FamilyLegacy(1200.12.2b1)[300DFACE-5EDA-36E4-8A96-0AE741FE179F]@0xffffff8011e3c000->0xffffff8011f81fff
            dependency: com.apple.driver.AppleMobileFileIntegrity(1.0.5)[68B4E144-4971-344D-8EFD-EFC286F00942]@0xffffff80117b1000->0xffffff80117d2fff
            dependency: com.apple.driver.corecapture(1.0.4)[530FD947-1077-358E-BABC-E7656E312541]@0xffffff801349d000->0xffffff80134b6fff
            dependency: com.apple.iokit.CoreAnalyticsFamily(1)[D9B274C7-317C-3CA7-B51B-F6F60AABBBD5]@0xffffff8011ad3000->0xffffff8011adafff
            dependency: com.apple.iokit.IONetworkingFamily(3.4)[2DFBB6A3-5EA4-3141-81B6-7AC52D5392DD]@0xffffff8012998000->0xffffff80129aefff
            dependency: com.apple.iokit.IOSkywalkFamily(1.0)[FD18C436-5C79-36AD-945A-5050821D5781]@0xffffff8012d16000->0xffffff8012d60fff
            dependency: com.apple.kec.corecrypto(12.0)[D5F1BD46-6F9D-3F71-982E-15DA8EDA4FC1]@0xffffff80134c3000->0xffffff8013543fff

Process name corresponding to current thread (0xffffff95011caaa0): airportd
Boot args: -v keepsyms=1 rtcfx_exclude=00-FF alcid=1 debug=0x100 chunklist-security-epoch=0 -chunklist-no-rev2-dev chunklist-security-epoch=0 -chunklist-no-rev2-dev

Mac OS version:
21G320

Kernel version:
Darwin Kernel Version 21.6.0: Sun Nov  6 23:31:16 PST 2022; root:xnu-8020.240.14~1/RELEASE_X86_64
Kernel UUID: F831C932-31CA-3F2C-9E4C-2304C5641587
KernelCache slide: 0x000000000fe00000
KernelCache base:  0xffffff8010000000
Kernel slide:      0x000000000fe10000
Kernel text base:  0xffffff8010010000
__HIB  text base: 0xffffff800ff00000
System model name: MacPro7,1 (Mac-27AD2F918AE68F61)
System shutdown begun: NO
Panic diags file available: YES (0x0)
Hibernation exit count: 0

System uptime in nanoseconds: 12607120510
Last Sleep:           absolute           base_tsc          base_nano
  Uptime  : 0x00000002ef7167c9
  Sleep   : 0x0000000000000000 0x0000000000000000 0x0000000000000000
  Wake    : 0x0000000000000000 0x0000001aba234efa 0x0000000000000000
Compressor Info: 0% of compressed pages limit (OK) and 0% of segments limit (OK) with 0 swapfiles and OK swap space
Zone info:
  Zone map: 0xffffff803510a000 - 0xffffffa03510a000
  . PGZ   : 0xffffff803510a000 - 0xffffff803910b000
  . VM    : 0xffffff803910b000 - 0xffffff850543e000
  . RO    : 0xffffff850543e000 - 0xffffff869eaa4000
  . GEN0  : 0xffffff869eaa4000 - 0xffffff8b6add7000
  . GEN1  : 0xffffff8b6add7000 - 0xffffff903710a000
  . GEN2  : 0xffffff903710a000 - 0xffffff950343d000
  . GEN3  : 0xffffff950343d000 - 0xffffff99cf770000
  . DATA  : 0xffffff99cf770000 - 0xffffffa03510a000
  Metadata: 0xffffffca5fd46000 - 0xffffffca7fd46000
  Bitmaps : 0xffffffca7fd46000 - 0xffffffca8bd46000
</details>

thedycui, 2023-06-16:
> 我在更新到 2.2.0 后出现了开机卡死或黑屏的问题。 网卡型号：AC 9260 (0x8086, 0x14) 使用的 kext: AirportItlwm
> 
> 问题信息：
> 
> ```
> panic(cpu 0 caller 0xffffff801003900e): Kernel stack memory corruption detected @stack_protector.c:37
> Panicked task 0xffffff9503474380: 18 threads: pid 225: airportd
> Backtrace (CPU 0), panicked thread: 0xffffff95011caaa0, Frame : Return Address
> 0xfffffff58702b710 : 0xffffff801007cead mach_kernel : _handle_debugger_trap + 0x41d
> 0xfffffff58702b760 : 0xffffff80101df5d6 mach_kernel : _kdp_i386_trap + 0x116
> 0xfffffff58702b7a0 : 0xffffff80101ce943 mach_kernel : _kernel_trap + 0x4d3
> 0xfffffff58702b7f0 : 0xffffff801001ca70 mach_kernel : _return_from_trap + 0xe0
> 0xfffffff58702b810 : 0xffffff801007d27d mach_kernel : _DebuggerTrapWithState + 0xad
> 0xfffffff58702b930 : 0xffffff801007ca36 mach_kernel : _panic_trap_to_debugger + 0x2b6
> 0xfffffff58702b990 : 0xffffff8010914ed3 mach_kernel : _panic + 0x84
> 0xfffffff58702ba80 : 0xffffff801003900e mach_kernel : ___stack_chk_fail + 0x1e
> 0xfffffff58702ba90 : 0xffffff8011eb69ba com.apple.iokit.IO80211FamilyLegacy : __ZL8getStatsP17IO80211ControllerP16IO80211InterfaceP23IO80211VirtualInterfaceP21IO80211InfraInterfaceP13apple80211reqb
> 0xfffffff58702baf0 : 0xffff1b09ee1a 
> Unaligned frame
> Backtrace terminated-invalid frame pointer 0x2d00000000710546
>       Kernel Extensions in backtrace:
>          com.apple.iokit.IO80211FamilyLegacy(1200.12.2b1)[300DFACE-5EDA-36E4-8A96-0AE741FE179F]@0xffffff8011e3c000->0xffffff8011f81fff
>             dependency: com.apple.driver.AppleMobileFileIntegrity(1.0.5)[68B4E144-4971-344D-8EFD-EFC286F00942]@0xffffff80117b1000->0xffffff80117d2fff
>             dependency: com.apple.driver.corecapture(1.0.4)[530FD947-1077-358E-BABC-E7656E312541]@0xffffff801349d000->0xffffff80134b6fff
>             dependency: com.apple.iokit.CoreAnalyticsFamily(1)[D9B274C7-317C-3CA7-B51B-F6F60AABBBD5]@0xffffff8011ad3000->0xffffff8011adafff
>             dependency: com.apple.iokit.IONetworkingFamily(3.4)[2DFBB6A3-5EA4-3141-81B6-7AC52D5392DD]@0xffffff8012998000->0xffffff80129aefff
>             dependency: com.apple.iokit.IOSkywalkFamily(1.0)[FD18C436-5C79-36AD-945A-5050821D5781]@0xffffff8012d16000->0xffffff8012d60fff
>             dependency: com.apple.kec.corecrypto(12.0)[D5F1BD46-6F9D-3F71-982E-15DA8EDA4FC1]@0xffffff80134c3000->0xffffff8013543fff
> 
> Process name corresponding to current thread (0xffffff95011caaa0): airportd
> Boot args: -v keepsyms=1 rtcfx_exclude=00-FF alcid=1 debug=0x100 chunklist-security-epoch=0 -chunklist-no-rev2-dev chunklist-security-epoch=0 -chunklist-no-rev2-dev
> 
> Mac OS version:
> 21G320
> 
> Kernel version:
> Darwin Kernel Version 21.6.0: Sun Nov  6 23:31:16 PST 2022; root:xnu-8020.240.14~1/RELEASE_X86_64
> Kernel UUID: F831C932-31CA-3F2C-9E4C-2304C5641587
> KernelCache slide: 0x000000000fe00000
> KernelCache base:  0xffffff8010000000
> Kernel slide:      0x000000000fe10000
> Kernel text base:  0xffffff8010010000
> __HIB  text base: 0xffffff800ff00000
> System model name: MacPro7,1 (Mac-27AD2F918AE68F61)
> System shutdown begun: NO
> Panic diags file available: YES (0x0)
> Hibernation exit count: 0
> 
> System uptime in nanoseconds: 12607120510
> Last Sleep:           absolute           base_tsc          base_nano
>   Uptime  : 0x00000002ef7167c9
>   Sleep   : 0x0000000000000000 0x0000000000000000 0x0000000000000000
>   Wake    : 0x0000000000000000 0x0000001aba234efa 0x0000000000000000
> Compressor Info: 0% of compressed pages limit (OK) and 0% of segments limit (OK) with 0 swapfiles and OK swap space
> Zone info:
>   Zone map: 0xffffff803510a000 - 0xffffffa03510a000
>   . PGZ   : 0xffffff803510a000 - 0xffffff803910b000
>   . VM    : 0xffffff803910b000 - 0xffffff850543e000
>   . RO    : 0xffffff850543e000 - 0xffffff869eaa4000
>   . GEN0  : 0xffffff869eaa4000 - 0xffffff8b6add7000
>   . GEN1  : 0xffffff8b6add7000 - 0xffffff903710a000
>   . GEN2  : 0xffffff903710a000 - 0xffffff950343d000
>   . GEN3  : 0xffffff950343d000 - 0xffffff99cf770000
>   . DATA  : 0xffffff99cf770000 - 0xffffffa03510a000
>   Metadata: 0xffffffca5fd46000 - 0xffffffca7fd46000
>   Bitmaps : 0xffffffca7fd46000 - 0xffffffca8bd46000
> ```

我不使用蓝牙驱动，就没出现过自动重启，昨天把蓝牙驱动也加上后，一会就出现了跟你一样的，突然自动重启了。我就把蓝牙的3个kext又给去掉了。

zxystd, 2023-06-16:
@thedycui 用的是前几天更新的蓝牙驱动吗？

zxystd, 2023-06-16:
> 我在更新到 2.2.0 后出现了开机卡死或黑屏的问题。 网卡型号：AC 9260 (0x8086, 0x14) 使用的 kext: AirportItlwm
>
> 问题信息：
>
> ```
> panic(cpu 0 caller 0xffffff801003900e): Kernel stack memory corruption detected @stack_protector.c:37 [...]

你这个应该是用错版本了，Ventura跟Monterey版本不能混用，按系统来

zxystd, 2023-06-16:
> > 
> 
> 对，我之前没用蓝牙驱动，只用AirportItlwm2.2稳定版，好几天没有突然重启了，昨天想试试蓝牙驱动可行，就又装了下，结果没过一会就突然重启了。

重启以后的日志找一下，看看。在这里找，对应的日期
[image]

另外你确定用的蓝牙版本是这个？https://github.com/OpenIntelWireless/IntelBluetoothFirmware/actions/runs/5210422067

thedycui, 2023-06-16:
那3个kext都是我下载的最新版，其中有一个用的是BlueToolFixup.kext
[image]

thedycui, 2023-06-16:
[image]

thedycui, 2023-06-16:
我刚刚又遇到一次突然重启。看来可能跟蓝牙驱动没有关系。
[Kernel-2023-06-16-235648.panic.txt](https://github.com/OpenIntelWireless/itlwm/files/11772120/Kernel-2023-06-16-235648.panic.txt)

zxystd, 2023-06-16:
@thedycui 你在boot-args里加上`keepsyms=1`吧，不然都看不到符号

zzwtsy, 2023-06-17:
> 你这个应该是用错版本了，Ventura跟Monterey版本不能混用，按系统来

非常感谢，问题已经解决了

codesnoopy5, 2023-10-27:
@zxystd ax210网卡，用最新版本2.3.0 alpha睡眠唤醒的时候也遇到这个问题了，但是不是每次必现。好像是蓝牙的问题，我上次开关蓝牙开关的时候，也突然重启了。
`panic(cpu 1 caller 0xffffff801dfdefbf): zfree_drop: double free of 0xffffff8f33d957a0 to zone kalloc.type.var5.32 @zalloc.c:816
Panicked task 0xffffff9d98452638: 225 threads: pid 0: kernel_task
Backtrace (CPU 1), panicked thread: 0xffffff9d98bafb30, Frame : Return Address
0xffffffd0f57bf6c0 : 0xffffff801d83aca1 
0xffffffd0f57bf710 : 0xffffff801d9993c0 
0xffffffd0f57bf750 : 0xffffff801d988b0c 
0xffffffd0f57bf7d0 : 0xffffff801d7d7971 
0xffffffd0f57bf7f0 : 0xffffff801d83af8d 
0xffffffd0f57bf8e0 : 0xffffff801d83a633 
0xffffffd0f57bf940 : 0xffffff801dfd957b 
0xffffffd0f57bfa30 : 0xffffff801dfdefbf 
0xffffffd0f57bfa60 : 0xffffff801d8ac15d 
0xffffffd0f57bfa90 : 0xffffff801decbbf0 
0xffffffd0f57bfab0 : 0xffffff80214fd56b 
0xffffffd0f57bfae0 : 0xffffff8021505ea3 
0xffffffd0f57bfc00 : 0xffffff802150e38c 
0xffffffd0f57bfc60 : 0xffffff8021508431 
0xffffffd0f57bfcf0 : 0xffffff80215080f2 
0xffffffd0f57bfdb0 : 0xffffff80215095ac 
0xffffffd0f57bfe40 : 0xffffff80214a000f 
0xffffffd0f57bffa0 : 0xffffff801d7d719e 
      Kernel Extensions in backtrace:
         com.zxystd.AirportItlwm(2.3)[843FB55D-FD93-3BEE-9585-B36DE5A3A3F5]@0xffffff802149d000->0xffffff8022427fff
            dependency: com.apple.driver.corecapture(1.0.4)[0DED4F91-AC29-379C-9A48-83E6009E40F0]@0xffffff8020369000->0xffffff802038bfff
            dependency: com.apple.iokit.IO80211Family(1200.13.0)[9E127BA1-2022-3BD8-A4FE-25410228FDC4]@0xffffff801f43e000->0xffffff801f67cfff
            dependency: com.apple.iokit.IONetworkingFamily(3.4)[5D912FD8-C4CD-3CF6-B214-DF5BF7AB4FA0]@0xffffff801f8c0000->0xffffff801f8d7fff
            dependency: com.apple.iokit.IOPCIFamily(2.9)[C63A508E-9546-335A-A391-9A958643634A]@0xffffff801fb21000->0xffffff801fb52fff
            dependency: com.apple.iokit.IOSkywalkFamily(1.0)[8FAC5BA7-70BB-3117-9D79-6B5B57F41E7A]@0xffffff801fc0e000->0xffffff801fc3cfff

Process name corresponding to current thread (0xffffff9d98bafb30): kernel_task
Boot args: -igfxblt -igfxbls -igfxblr revpatch=auto,sbvmm,asset darkwake=1 agdpmod=vit9696 igfxfw=2 igfxonln=1 gfxrst=1 alcverbs=1 -disablegfxfirmware -lilubetaall

Mac OS version:
23B73

Kernel version:
Darwin Kernel Version 23.1.0: Mon Oct  9 21:27:27 PDT 2023; root:xnu-10002.41.9~6/RELEASE_X86_64
Kernel UUID: 7C8FE931-57C6-38AD-9EBB-CC5572293AE6
roots installed: 0
KernelCache slide: 0x000000001d400000
KernelCache base:  0xffffff801d600000
Kernel slide:      0x000000001d4e0000
Kernel text base:  0xffffff801d6e0000
__HIB  text base: 0xffffff801d500000
System model name: MacBookPro16,1 (Mac-E1008331FDC96864)
System shutdown begun: NO
Panic diags file available: YES (0x0)
Hibernation exit count: 0

System uptime in nanoseconds: 1264643721070
Last Sleep:           absolute           base_tsc          base_nano
  Uptime  : 0x0000012672a3cbbf
  Sleep   : 0x00000124dd1d294e 0x00000000e6863e3c 0x00000123e146d716
  Wake    : 0x0000012509b16723 0x00000000e6d3f28c 0x00000124ddc05f8c
Compressor Info: 0% of compressed pages limit (OK) and 0% of segments limit (OK) with 0 swapfiles and OK swap space
Zone info:
  Zone map: 0xffffff88ca0b9000 - 0xffffffa8ca0b9000
  . PGZ   : 0xffffff88ca0b9000 - 0xffffff88ce0ba000
  . VM    : 0xffffff88ce0ba000 - 0xffffff8d9a3ed000
  . RO    : 0xffffff8d9a3ed000 - 0xffffff8f33a53000
  . GEN0  : 0xffffff8f33a53000 - 0xffffff93ffd86000
  . GEN1  : 0xffffff93ffd86000 - 0xffffff98cc0b9000
  . GEN2  : 0xffffff98cc0b9000 - 0xffffff9d983ec000
  . GEN3  : 0xffffff9d983ec000 - 0xffffffa26471f000
  . DATA  : 0xffffffa26471f000 - 0xffffffa8ca0b9000
  Metadata: 0xffffff80672ff000 - 0xffffff80872ff000
  Bitmaps : 0xffffff80872ff000 - 0xffffff808d2ff000
  Extra   : 0 - 0

[... 166 lines cut ...]
>!ACyrus	1
>!ACredentialManager	1.0
>KernelRelayHost	1
|IOUSBHost!F	1.2
>!UHostMergeProperties	1.2
>usb.!UCommon	1.0
>!AMobileFileIntegrity	1.0.5
$!AImage4	5.0.0
@kext.CoreTrust	1
|CoreAnalytics!F	1
>!ABusPower!C	1.0
>!ASEPManager	1.0.1
>IOSlaveProcessor	1
>!AACPIPlatform	6.1
>!ASMC	3.1.9
|IOPCI!F	2.9
|IOACPI!F	1.4
>watchdog	1
@kec.pthread	1
@kec.XrtHostedXnu	1
@kec.Libm	1
@kec.Compression	1.0
@kec.corecrypto	14.0

`

codesnoopy5, 2023-10-27:
查了一下这个错误，好像是一个指针被free两次

zxystd, 2024-03-16:
Please try v2.3.0-alpha version, seems has been resolved.

zxystd, 2024-03-19:
Reopen if you face the same issue again.

closed 2024-03-19 by zxystd

idootop, 2024-04-16:
> Please try v2.3.0-alpha version, seems has been resolved.

AX210 网卡，v2.3.0-alpha 版本问题依然存在，不过出现概率貌似有所下降。
<details>
<summary>👉 完整日志</summary>

```shell
panic(cpu 4 caller 0xffffff801cde14f5): zfree_drop: double free of 0xffffff8b7aece820 to zone kalloc.type.var4.32 @zalloc.c:815
Panicked task 0xffffff90473a2bf8: 201 threads: pid 0: kernel_task
Backtrace (CPU 4), panicked thread: 0xffffff8b7adfd598, Frame : Return Address
0xffffffa8792336d0 : 0xffffff801c670c7d mach_kernel : _handle_debugger_trap + 0x4ad
0xffffffa879233720 : 0xffffff801c7c4f74 mach_kernel : _kdp_i386_trap + 0x114
0xffffffa879233760 : 0xffffff801c7b4a87 mach_kernel : _kernel_trap + 0x3b7
0xffffffa8792337b0 : 0xffffff801c611971 mach_kernel : _return_from_trap + 0xc1
0xffffffa8792337d0 : 0xffffff801c670f5d mach_kernel : _DebuggerTrapWithState + 0x5d
0xffffffa8792338c0 : 0xffffff801c670607 mach_kernel : _panic_trap_to_debugger + 0x1a7
0xffffffa879233920 : 0xffffff801cddb96b mach_kernel : _panic + 0x84
0xffffffa879233a10 : 0xffffff801cde14f5 mach_kernel : ___smr_tail_invalid + 0x294f
0xffffffa879233a40 : 0xffffff801c6e1423 mach_kernel : _workload_config_available + 0x8063
0xffffffa879233a70 : 0xffffff801ccd0ec0 mach_kernel : _IOFree + 0x20
0xffffffa879233a90 : 0xffffff8020ffdeb4 com.zxystd.AirportItlwm : __ZL4freePv + 0x54
0xffffffa879233ac0 : 0xffffff802100694d com.zxystd.AirportItlwm : __ZN6ItlIwx12iwx_send_cmdEP9iwx_softcP12iwx_host_cmd + 0x87d
0xffffffa879233bd0 : 0xffffff802100f194 com.zxystd.AirportItlwm : __ZN6ItlIwx19iwx_send_cmd_statusEP9iwx_softcP12iwx_host_cmdPj + 0x64
0xffffffa879233c30 : 0xffffff8021009045 com.zxystd.AirportItlwm : __ZN6ItlIwx23iwx_send_cmd_pdu_statusEP9iwx_softcjtPKvPj + 0x75
0xffffffa879233cb0 : 0xffffff8021010d76 com.zxystd.AirportItlwm : __ZN6ItlIwx13iwx_drain_staEP9iwx_softcP8iwx_nodei + 0xb6
0xffffffa879233d30 : 0xffffff8021010c0b com.zxystd.AirportItlwm : __ZN6ItlIwx13iwx_flush_staEP9iwx_softcP8iwx_node + 0x4b
0xffffffa879233d70 : 0xffffff802101752e com.zxystd.AirportItlwm : __ZN6ItlIwx12iwx_run_stopEP9iwx_softc + 0x5e
0xffffffa879233dd0 : 0xffffff80210185ce com.zxystd.AirportItlwm : __ZN6ItlIwx17iwx_newstate_taskEPv + 0x1ce
0xffffffa879233e40 : 0xffffff8020facc37 com.zxystd.AirportItlwm : __Z12taskq_threadPv + 0x97
0xffffffa879233fa0 : 0xffffff801c61119e mach_kernel : _call_continuation + 0x2e
      Kernel Extensions in backtrace:
         com.zxystd.AirportItlwm(2.3)[97652176-6521-3FAE-9E1C-F8F39F9F24BA]@0xffffff8020fa0000->0xffffff8021f33fff
            dependency: com.apple.iokit.IO80211FamilyLegacy(1200.12.2b1)[93312C5E-DBEF-38CF-A9D3-352B3FE33A5E]@0xffffff801e2f3000->0xffffff801e439fff
            dependency: com.apple.iokit.IONetworkingFamily(3.4)[9464DB11-7BAB-372E-BE42-312E2C708040]@0xffffff801ee53000->0xffffff801ee69fff
            dependency: com.apple.iokit.IOPCIFamily(2.9)[6631E76B-0994-3BDC-AFB3-1C8A04D6812C]@0xffffff801f0b4000->0xffffff801f0e5fff

Process name corresponding to current thread (0xffffff8b7adfd598): kernel_task
Boot args: brcmfx-country=#a igfxonln=1 brcmfx-aspm debug=0x100 keepsyms=1 alcid=12 -alcbeta -lilubetaall

Mac OS version:
22G436

Kernel version:
Darwin Kernel Version 22.6.0: Tue Nov  7 21:48:06 PST 2023; root:xnu-8796.141.3.702.9~2/RELEASE_X86_64
Kernel UUID: 7921C692-C7F5-3244-B15C-A6179401ACDF
roots installed: 0
KernelCache slide: 0x000000001c200000
KernelCache base:  0xffffff801c400000
Kernel slide:      0x000000001c2dc000
Kernel text base:  0xffffff801c4dc000
__HIB  text base: 0xffffff801c300000
System model name: iMac20,1 (Mac-CFF7D910A743CAAF)
System shutdown begun: NO
Panic diags file available: YES (0x0)
Hibernation exit count: 0

System uptime in nanoseconds: 167873436169148
Last Sleep:           absolute           base_tsc          base_nano
  Uptime  : 0x000098ae1487b664
  Sleep   : 0x0000000000000000 0x0000000000000000 0x0000000000000000
  Wake    : 0x0000000000000000 0x0000000c1b0dd015 0x0000000000000000
Compressor Info: 0% of compressed pages limit (OK) and 0% of segments limit (OK) with 0 swapfiles and OK swap space
Zone info:
  Zone map: 0xffffff8045359000 - 0xffffffa045359000
  . PGZ   : 0xffffff8045359000 - 0xffffff804935a000
  . VM    : 0xffffff804935a000 - 0xffffff851568d000
  . RO    : 0xffffff851568d000 - 0xffffff86aecf3000
  . GEN0  : 0xffffff86aecf3000 - 0xffffff8b7b026000
  . GEN1  : 0xffffff8b7b026000 - 0xffffff9047359000
  . GEN2  : 0xffffff9047359000 - 0xffffff951368c000
[... 139 lines cut ...]
>!AFDEKeyStore	28.30
>!AEffaceable!S	1.0
>!ACyrus	1
>!AMobileFileIntegrity	1.0.5
$!AImage4	5.0.0
@kext.CoreTrust	1
>!ACredentialManager	1.0
|CoreAnalytics!F	1
>KernelRelayHost	1
|IOUSBHost!F	1.2
>!UHostMergeProperties	1.2
>usb.!UCommon	1.0
>!ABusPower!C	1.0
>!ASEPManager	1.0.1
>IOSlaveProcessor	1
>!AACPIPlatform	6.1
>!ASMC	3.1.9
|IOPCI!F	2.9
|IOACPI!F	1.4
>watchdog	1
@kec.pthread	1
@kec.Libm	1
@kec.corecrypto	12.0
```
</details>