OpenIntelWireless/itlwm #785: Firmware crashes when using iPhone network

usr-sse2, 2022-06-10:
**Have You Read Our Docs**
No

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: 2.2.0 from #764
- WiFi Card Model: 8265
- PCI Product ID: 0x0150
- macOS Version: 12.4

**Description**
When using iPhone network, I always get frequent disconnections with the following messages in the log:
```
[51994.733114]: itlwm: setting GTK successfully
[51994.733122]: itlwm: marking port 6e:7c:19:3d:c0:e3 valid
[51994.733130]: itlwm: ieee80211_set_link_state nstate=4, old_state=4
[51994.738832]: itlwm: setROAM_PROFILE cnt=1 flags=4
[51994.738906]: itlwm: setROAM_PROFILE cnt=1 flags=2
[51994.740142]: itlwm: setMulticastList multicast count=5 bssid=6e:7c:19:3d:c0:e3
[51994.740169]: itlwm: setMulticastList multicast count=6 bssid=6e:7c:19:3d:c0:e3
[51994.996305]: itlwm: __iwl_mvm_rs_tx_status new rate: Legacy | ANT: B Rate: 1 Mbps
[51995.102233]: itlwm: apple80211Request Unhandled IOCTL BTCOEX_PROFILE (255) set
[51995.104731]: itlwm: apple80211Request Unhandled IOCTL BTCOEX_PROFILE (255) set
[51995.154772]: itlwm: __iwl_mvm_rs_tx_status new rate: Legacy | ANT: A Rate: 1 Mbps
[51995.156247]: itlwm: apple80211Request Unhandled IOCTL BTCOEX_PROFILE (255) set
[51995.159746]: itlwm: setROAM_PROFILE cnt=1 flags=4
[51995.159833]: itlwm: setROAM_PROFILE cnt=1 flags=2
[51995.160196]: itlwm: apple80211Request Unhandled IOCTL BTCOEX_PROFILE (255) set
[51995.162709]: itlwm: apple80211Request Unhandled IOCTL BTCOEX_PROFILE (255) set
[51995.166510]: uipc_accept: peer disconnected unp_gencnt 14129uipc_accept: peer disconnected unp_gencnt 14131itlwm: setSCANCACHE_CLEAR
[51995.180964]: itlwm: apple80211Request Unhandled IOCTL BTCOEX_PROFILE (255) set
[51995.182140]: itlwm: apple80211Request Unhandled IOCTL BTCOEX_PROFILE (255) set
[51995.185207]: itlwm: taskq task_add ba_task add to queue
[51995.185457]: itlwm: itlwm: sending addba_resp to 6e:7c:19:3d:c0:e3 on channel 6 mode 11n
[51995.186383]: itlwm: apple80211Request Unhandled IOCTL BTCOEX_PROFILE (255) set
[51995.190891]: itlwm: __iwl_mvm_rs_tx_status new rate: Legacy | ANT: A Rate: 2 Mbps
[51995.209871]: uipc_accept: peer disconnected unp_gencnt 14137uipc_accept: peer disconnected unp_gencnt 14140uipc_accept: peer disconnected unp_gencnt 14145itlwm: __iwl_mvm_rs_tx_status new rate: Legacy | ANT: A Rate: 5.5 Mbps
[51995.282551]: itlwm: __iwl_mvm_rs_tx_status new rate: Legacy | ANT: A Rate: 11 Mbps
[51995.389542]: itlwm: setSCANCACHE_CLEAR
[51995.390633]: itlwm: __iwl_mvm_rs_tx_status new rate: Legacy | ANT: B Rate: 18 Mbps
[51995.397191]: itlwm: __iwl_mvm_rs_tx_status new rate: Legacy | ANT: A Rate: 18 Mbps
[51995.472495]: itlwm: iwm_bgscan
[51995.472500]: itlwm: iwm_umac_scan
[51995.494519]: itlwm: __iwl_mvm_rs_tx_status new rate: Legacy | ANT: A Rate: 24 Mbps
[51995.727414]: uipc_accept: peer disconnected unp_gencnt 14159uipc_accept: peer disconnected unp_gencnt 14163itlwm: __iwl_mvm_rs_tx_status new rate: Legacy | ANT: A Rate: 36 Mbps
[51995.753710]: itlwm: __iwl_mvm_rs_tx_status new rate: Legacy | ANT: A Rate: 48 Mbps
[51995.769800]: itlwm: __iwl_mvm_rs_tx_status new rate: Legacy | ANT: A Rate: 36 Mbps
[51995.772439]: itlwm: __iwl_mvm_rs_tx_status new rate: Legacy | ANT: A Rate: 48 Mbps
[51995.777759]: itlwm: __iwl_mvm_rs_tx_status new rate: Legacy | ANT: A Rate: 54 Mbps
[51995.822505]: itlwm: __iwl_mvm_rs_tx_status new rate: 0x48004106: HT | ANT: A BW: 20Mhz MCS: 6 NSS: 1 NGI LDPC 
[51995.852260]: uipc_accept: peer disconnected unp_gencnt 14167itlwm: __iwl_mvm_rs_tx_status new rate: 0x48004107: HT | ANT: A BW: 20Mhz MCS: 7 NSS: 1 NGI LDPC 
[51995.860024]: itlwm: __iwl_mvm_rs_tx_status new rate: 0x48004106: HT | ANT: A BW: 20Mhz MCS: 6 NSS: 1 NGI LDPC 
[51995.866346]: itlwm: __iwl_mvm_rs_tx_status new rate: 0x48004105: HT | ANT: A BW: 20Mhz MCS: 5 NSS: 1 NGI LDPC 
[51995.866976]: itlwm: __iwl_mvm_rs_tx_status new rate: 0x48004106: HT | ANT: A BW: 20Mhz MCS: 6 NSS: 1 NGI LDPC 
[51995.868380]: itlwm: __iwl_mvm_rs_tx_status new rate: 0x48008106: HT | ANT: B BW: 20Mhz MCS: 6 NSS: 1 NGI LDPC 
[51995.874375]: uipc_accept: peer disconnected unp_gencnt 14171uipc_accept: peer disconnected unp_gencnt 14174itlwm: __iwl_mvm_rs_tx_status new rate: 0x48004106: HT | ANT: A BW: 20Mhz MCS: 6 NSS: 1 NGI LDPC 
[51995.877621]: uipc_accept: peer disconnected unp_gencnt 14179itlwm: __iwl_mvm_rs_tx_status new rate: 0x4800c10c: HT | ANT: AB BW: 20Mhz MCS: 12 NSS: 2 NGI LDPC 
[51995.884794]: itlwm: __iwl_mvm_rs_tx_status new rate: 0x4800c10d: HT | ANT: AB BW: 20Mhz MCS: 13 NSS: 2 NGI LDPC 
[51995.892226]: itlwm: __iwl_mvm_rs_tx_status new rate: 0x4800c10e: HT | ANT: AB BW: 20Mhz MCS: 14 NSS: 2 NGI LDPC 
[51995.999810]: tcp_timers: tcp_output() returned 0 with retransmission timer disabled for 55336 > 62078 in state 4, reset timer to 410tcp_timers: tcp_output() returned 0 with retransmission timer disabled for 55335 > 62078 in state 4, reset timer to 412itlwm: __iwl_mvm_rs_tx_status new rate: 0x4800c10d: HT | ANT: AB BW: 20Mhz MCS: 13 NSS: 2 NGI LDPC 
[51996.030133]: itlwm: __iwl_mvm_rs_tx_status new rate: 0x4800e10d: HT | ANT: AB BW: 20Mhz MCS: 13 NSS: 2 SGI LDPC 
[51996.039887]: itlwm: __iwl_mvm_rs_tx_status new rate: 0x4800e10e: HT | ANT: AB BW: 20Mhz MCS: 14 NSS: 2 SGI LDPC 
[51996.050710]: itlwm: __iwl_mvm_rs_tx_status new rate: 0x4800e10f: HT | ANT: AB BW: 20Mhz MCS: 15 NSS: 2 SGI LDPC 
[51996.068734]: itlwm: __iwl_mvm_rs_tx_status new rate: 0x4800e10e: HT | ANT: AB BW: 20Mhz MCS: 14 NSS: 2 SGI LDPC 
[51996.239087]: itlwm: itlwm: sending addba_req to 6e:7c:19:3d:c0:e3 on channel 6 mode 11n
[51996.311140]: tcp_timers: tcp_output() returned 0 with retransmission timer disabled for 55333 > 62078 in state 4, reset timer to 758itlwm: taskq task_add ba_task add to queue
[51996.334346]: itlwm: iwm_ba_task tx_start ssn=0, tid=0 scd_queue=10
[... 422 lines cut ...]
[52280.225418]: itlwm:   tx ring 22: qid=22 cur=0   queued=0  
[52280.225420]: itlwm:   tx ring 23: qid=23 cur=0   queued=0  
[52280.225423]: itlwm:   tx ring 24: qid=24 cur=0   queued=0  
[52280.225425]: itlwm:   tx ring 25: qid=25 cur=0   queued=0  
[52280.225427]: itlwm:   tx ring 26: qid=26 cur=0   queued=0  
[52280.225429]: itlwm:   tx ring 27: qid=27 cur=0   queued=0  
[52280.225432]: itlwm:   tx ring 28: qid=28 cur=0   queued=0  
[52280.225434]: itlwm:   tx ring 29: qid=29 cur=0   queued=0  
[52280.225436]: itlwm:   tx ring 30: qid=30 cur=0   queued=0  
[52280.225438]: itlwm:   rx ring: cur=63
[52280.225440]: itlwm:   802.11 state RUN
[52280.225442]: itlwm: itlwm: fatal firmware error
[52280.225444]: itlwm: taskq task_add init_task add to queue
[52280.363936]: tcp_timers: tcp_output() returned 0 with retransmission timer disabled for 55749 > 8081 in state 4, reset timer to 14449tcp_timers: tcp_output() returned 0 with retransmission timer disabled for 55748 > 8081 in state 4, reset timer to 22318itlwm: iwm_enable_txq failed error=35
[52282.226982]: itlwm: iwm_init_task
[52282.226989]: itlwm: iwm_stop
[52282.226993]: itlwm: iwm_stop_device
[52282.277921]: itlwm: iwm_conf_msix_hw
[52282.277937]: itlwm: iwm_check_rfkill RF_KILL hw: 0
[52282.277940]: itlwm: iwm_prepare_card_hw
[52282.277977]: itlwm: itlwm: RUN -> INIT
```

**Kext Download Source**
https://github.com/OpenIntelWireless/itlwm/issues/764#issuecomment-1111041263

zxystd, 2022-06-11:
Thank you for the feedback, is it reproducible? and if it is the same for v2.2.0-alpha version which is control by RA?

usr-sse2, 2022-06-14:
> Thank you for the feedback, is it reproducible? and if it is the same for v2.2.0-alpha version which is control by RA?

Yes, it happens every time when I use the iPhone network. No, on v2.2.0-alpha it's stable.

zxystd, 2022-08-24:
[Monterey.zip](https://github.com/OpenIntelWireless/itlwm/files/9416930/Monterey.zip)
[Ventura.zip](https://github.com/OpenIntelWireless/itlwm/files/9416931/Ventura.zip)

@usr-sse2 It is the fixed version, sorry for the late reply.

closed 2023-12-11 by usr-sse2