OpenIntelWireless/itlwm #788: MacOS not recognizing ac7265

that-one-gamer1015, 2022-06-13:
**Have You Read Our Docs
Yes

**Are You Reporting A Bug**
Yes

**Environment**
Dell latitude e5450 using opencore
- Kext Version: v2.1.0
- WiFi Card Model: intel ac-7265
- PCI Product ID: PCI\VEN_8086&DEV_095A&SUBSYS_91108086
- macOS Version: montoray

**Description**
Mac os will not use or recognize my wifi card at all.

**Bug Report Archive**
Cant get heliport to create a diagnostic report.

**Kext Download Source**
Original from itlwm github

zxystd, 2022-06-15:
Can reopen once log is provided.

closed 2022-06-15 by zxystd

zxystd, 2022-06-15:
No log means not running, should check your config by yourself, and it is out of our control, we can't do anything.
https://openintelwireless.github.io/itlwm/Troubleshooting.html