OpenIntelWireless/itlwm #791: WiFi getting disconnected Automatically

pratiknborkar, 2022-06-22:
**Have You Read Our Docs**
                 Yes

**Are You Reporting A Bug**
 Yes

**Environment**

- Kext Version: v2.1.0 stable
- WiFi Card Model: AX 201 160 MHz
- PCI Product ID: PCI\VEN_8086&DEV_06F0&SUBSYS_00748086&REV_00
- macOS Version: MacOS_Monterey 12.4

**Description**
After updating to 12.4, WiFi is getting disconnected after a few min even. if I move my system then also disconnected. I have to reboot and re-connect after reboot it works for a couple of mins then the same again.
Works fine on Windows. Testest version v2.0.0 / v2.1.0 / v2.2.0-alpha

**Bug Report Archive**

[Log_2022-06-22_12-49-27.zip](https://github.com/OpenIntelWireless/itlwm/files/8955433/Log_2022-06-22_12-49-27.zip)

**Kext Download Source**
https://github.com/OpenIntelWireless/itlwm/releases/tag/v2.1.0

zxystd, 2022-06-22:
The log is useful, please collect it again after automatic disconnected, also please install DebugEnhance.kext and use v2.2.0-alpha version of itlwm instead.

pratiknborkar, 2022-06-22:
Thanks for your reply. here are the new logs with v2.2.0 before disconnect and after disconnect also I am using DebugEnhance.kext too.

[After Disconnect Log_2022-06-22_16-35-54.log](https://github.com/OpenIntelWireless/itlwm/files/8957003/After.Disconnect.Log_2022-06-22_16-35-54.log)

[Before Disconnect Log_2022-06-22_16-35-20.log](https://github.com/OpenIntelWireless/itlwm/files/8956996/Before.Disconnect.Log_2022-06-22_16-35-20.log)

zxystd, 2022-06-22:
did it also happened on 5GHz band wifi?

zxystd, 2022-06-23:
seems that interrupt was blocked by something, does the driver works fine before? which version? only happened after you upgrade to Monterey 12.4?

pratiknborkar, 2022-06-23:
Yes till last version of mac it was fine and works stable.
In big sur and Monterey 12.3 it was super stable but after update same issue even same issue in Macos 13 also

zxystd, 2022-06-23:
@pratiknborkar So I am doubting that it is not the issue of the driver, maybe you can try another similar EFI or reinstall the system again?

pratiknborkar, 2022-06-23:
Yes I have tried reinstalling with with two different EFI one with DSDT and one with SSDT DORTANIA on both case same result.

zxystd, 2022-06-23:
We have test it and it is working fine on 12.4. From the log there is something wrong with the system that can't process much frequency interrupts immediately, so the network card broken after associated to the AP. We don't know where is the problem of the system, but the Hackintosh problem is out of our control, there is nothing can be changed with the driver. You can try another source of help, and I will keep this issue open for some days, if you have any clue, please update it.

closed 2022-07-13 by zxystd