OpenIntelWireless/itlwm #792: A210 - physical modification necessary?

peddy22, 2022-07-02:
**Have You Read Our Docs**

Yes
**Are You Reporting A Bug**

Yes
**Environment**

- Kext Version: 2.1
- WiFi Card Model: AX210
- PCI Product ID: OpenPCI
- macOS Version: Monterey

**Description**
I can't see devices using Airdrop and the Bluetooth dialog hangs

**Bug Report Archive**

   81    0 0                  0xfa4000   0xfa4000   com.zxystd.AirportItlwm (2.1.0) FA02D059 <80 50 16 7 6 3 1>

**Kext Download Source**

main gh release

peddy22, 2022-07-02:
Regarding the title, I have taped pin 51 according to some instructions (it's an mPCIE card)

williambj1, 2022-07-02:
AFAIK no hardware mod is necessary, read the docs to learn more about continuity

closed 2022-07-02 by williambj1