OpenIntelWireless/itlwm #793: AC9560   wifi无法驱动

Yeswwq, 2022-07-07:
引导：oc 8.2
系统：monterey 12.4
主板：微星 z490m s01
驱动：[[AirportItlwm-Monterey-v2.2.0-DEBUG-alpha-d9f1294.zip]](https://github.com/OpenIntelWireless/itlwm/releases/download/v2.2.0-alpha/AirportItlwm-Monterey-v2.2.0-DEBUG-alpha-d9f1294.zip)
网卡：Intel(R) Wireless-AC 9560 160MHz

这块网卡被主板识别成了ax201 160mhz，蓝牙是正常的，Wi-Fi驱动也是正常载入了，但是不起作用

日志：
[    1.082917]: AirportItlwm::init IO80211Legacy_kexts-37 "IO80211Legacy_kexts-37" Apr 28 2022 03:29:49
[    3.307423]: hook_thread_userret calleduserret(490): invoking callback; 0/4 remainingIO80211Controller::createIOReporters 0xfbd1cca5ee5588a3
[    4.495847]: IO80211Controller::findAndAttachToFaultReporter no pipe found, making it
[    4.495994]: IO80211Controller::start _controller is 0xfbd1cca5ee5588a3, provider is 0xfbd1cca121584ea3
[    4.506613]: itlwm: No config found for PCI dev 06f0/0000, rev=0x354, rfid=0x0
[    4.506772]: itlwm: ieee80211_ifdetach
[    4.506774]: itlwm: ieee80211_proto_detach
[    4.506775]: itlwm: ieee80211_crypto_detach
[    4.506776]: itlwm: ieee80211_node_detach
[    4.506778]: itlwm: ifmedia_delete_instance
[    4.506779]: itlwm: attach fail
[    4.507304]: itlwm: free
[    4.507305]: itlwm: virtual void ItlHalService::free()

[Log_2022-07-07_04-48-02.log](https://github.com/OpenIntelWireless/itlwm/files/9063286/Log_2022-07-07_04-48-02.log)

zxystd, 2022-07-07:
[ 4.506613]: itlwm: No config found for PCI dev 06f0/0000, rev=0x354, rfid=0x0

你的网卡出问题了，subid是0，不知道是网卡的bug还是系统的bug，之前见过有人重启到Ubuntu然后再重启回来就好了。你可以试试彻底断电再启动，还不行的话就去Windows下或者Ubuntu下重启过来就行了。
这个问题跟这个驱动没有关系。

webkong, 2022-07-11:
可以驱动的。正常使用。建议回windows看一下是否正常。

closed 2022-07-13 by zxystd

Yeswwq, 2022-07-23:
> [ 4.506613]: itlwm: No config found for PCI dev 06f0/0000, rev=0x354, rfid=0x0 你的网卡出问题了，subid是0，不知道是网卡的bug还是系统的bug，之前见过有人重启到Ubuntu然后再重启回来就好了。你可以试试彻底断电再启动，还不行的话就去Windows下或者Ubuntu下重启过来就行了。 这个问题跟这个驱动没有关系。

从Linux重新启动后就可以了，非常感谢，目前更换了killer 1650i网卡，wifi链接正常，只是wifi在开关后列表加载有些滞后