OpenIntelWireless/itlwm #798: Intel ac7265 - Random reboots and WiFi drops

mWachu, 2022-07-12:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**
Thinkpad x250

- Kext Version: v2.1.0
- WiFi Card Model: Intel ac7265
- PCI Product ID: 
- macOS Version: 12.4 (21F79)

**Description**

System is randomly losing WiFi connection and reboots afterwards. Sometimes it can run smoothly for hours, sometimes it happens after few minutes.

Just after the WiFi outage and before the system reboot there is a lot of libnetwork.dylib related entries in system log:

 Error       0x0                  540    0    com.apple.WebKit.Networking: (libnetwork.dylib) [com.apple.network:] nw_http3_save_output_handler called with null quic_protocol, dumping backtrace:
    0   libnetwork.dylib                    0x00007ff8201577f8 __nw_create_backtrace_string + 120
    1   libnetwork.dylib                    0x00007ff81fbff4d0 nw_http3_save_output_handler + 832

I also do see some crash diagnostics records being created upon system startup:
[image]

**Bug Report Archive**

marcin.wach@MacBook-Pro ~ % kextstat | grep -i "itlwm"                                                                                   
Executing: /usr/bin/kmutil showloaded
No variant specified, falling back to release
   82    0 0                  0xfa4000   0xfa4000   com.zxystd.AirportItlwm (2.1.0) FA02D059-D593-3D8E-A031-1F99EA4D67EB <81 49 16 7 6 3 1>

[Uploading Log_2022-07-13_00-15-40.log…]()

**Kext Download Source**
Latest v2.1.0 stable release

zxystd, 2022-07-13:
Not related to this driver.

zxystd, 2022-07-13:
I think the problem may related to this thread.
https://bbs.pcbeta.com/viewthread-1938139-1-1.html

mWachu, 2022-07-13:
Oh okay, but the URL you mention does not describes the ac7265. I assume the issue is in Apple's driverKit E1000e? And there is nothing i can do except trying to disable vt-d?

Thanks in advance.

Human7900, 2022-07-13:
If you want WiFi to work without any kexts, consider buying a Broadcom card.
https://dortania.github.io/Wireless-Buyers-Guide/.