OpenIntelWireless/itlwm #799: Location services doesn't work \w AirportItlwm on macOS Ventura

kpostekk, 2022-07-14:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: 2.2.0 (build from d9f129454ee15d81e039acc2dc5e785465cd65f9)
- WiFi Card Model: Intel Wireless AC 9462
- PCI Product ID: uhhhh, 0x2A4 *(not sure)*
- macOS Version: 13.0 Beta (22A5295i)

**Description**
Location services doesn't work. They were working just fine on Monterey.

**Bug Report Archive**
[Log_2022-07-14_18-33-02.log](https://github.com/OpenIntelWireless/itlwm/files/9113602/Log_2022-07-14_18-33-02.log)

**Kext Download Source**
This repo

ameenjuz, 2022-08-15:
Also wifi do not scan the network and connection issue it is not finding network on macOS ventura 13 beta 5
Please fix this for macOS ventura 13
I compiled kext from your source with latest commit changes but not luck

xuewencao, 2022-08-24:
俺也一样

melxman, 2022-09-05:
我现在发现了一个现象：
首先我系统是13.0 Beta6版(22A5331f)
1，如果我在系统重启前关掉了wifi开关，重启后能连上之前连过的wifi(但是距离过远50m左右，也会出现偶尔发现不了这个信号的情况)。但是无法发现新的信号源
2，如果我重启前没有关闭wifi开关。重启后自动链接之前连过的wifi。且可以发现其他周围的信号源。但是无法链接(偶尔能连上)

这两种情况能自动连上的那个wifi，是我在更新到13beta6(从12直接更新的)第一次链接的，现在我也不敢忽略后重新再连了。哈哈

如果作者需要日志文件的话。我后续会上传的

usr-sse2, 2022-10-25:
Scanning is the reason, and location services is the consequence, because location services use BSSIDs of nearby networks, and if networks are not seen, then the location can't be determined, so this should be renamed to "Network scanning doesn't work on macOS Ventura"

zxystd, 2022-10-30:
@usr-sse2 Yes, seems that scanning is problematic on Ventura, I can see the airportd always log not scan but use the scan cache in the output console, and the scanning operation signal is not dispatched the to kernel space, do you have some conclusion?

usr-sse2, 2022-10-30:
I see that networks are scanned and sometimes displayed in System Profiler and System Preferences, but very rarely in the Wi-Fi menu. It seems that it has become slightly better after I changed noise to always be 0 (because on Broadcom it's either 0 or negative, but itlwm reports it as a positive value, which is incorrect), but still it has problems. I'll continue experimenting and debugging. I'm also beginning to write a Lilu plugin to add logging of ioctl arguments and results in Broadcom driver to help understanding its operation.

zxystd, 2022-10-30:
> I see that networks are scanned and sometimes displayed in System Profiler and System Preferences, but very rarely in the Wi-Fi menu. It seems that it has become slightly better after I changed [...]

I am also doubted that I post the wrong scan results to the airportd. I have also test at virtual machine that the airportd always think it already has scan cache although the wifi driver is first time loaded and never scan before, it is weird.

> I'll continue experimenting and debugging. I'm also beginning to write a Lilu plugin to add logging of ioctl arguments and results in Broadcom driver to help understanding its operation.

It is a good idea, it is suggested that save the IOCTL full result to file, because the log will not print if its length is too long.

GotierLdl, 2022-11-05:
Hello, Same issue for me.

bootlegseahorse, 2022-11-08:
Same on my Intel AX200.

FieryFlames, 2022-11-09:
Having the same issue on my Intel Dual Band Wireless AC 8260

uifor, 2022-11-10:
Same on my Intel 7265.

Boos4721, 2022-11-12:
same on AX210

usr-sse2, 2022-11-12:
> I am also doubted that I post the wrong scan results to the airportd. I have also test at virtual machine that the airportd always think it already has scan cache although the wifi driver is first [...]

Here is the mentioned logger: https://github.com/usr-sse2/AirPortDebug (I made it work only now, as I had no prior experience in making Lilu plugins).

alsdk98, 2022-11-15:
Same on AX201

uifor, 2022-11-15:
Nic model: Intel 7265
System version: macOS13.0.1

It has been found that when the computer is just started, the location icon appears, the location can work briefly, then the location will not work

anaander, 2022-11-30:
adding the boot-arg io80211.awdl=0 to opencore fixed location service for me. Still not seeing anything show up in the wifi network scan but at the very least location services are working again.

AkikoKumagara, 2022-12-01:
> adding the boot-arg io80211.awdl=0 to opencore fixed location service for me. Still not seeing anything show up in the wifi network scan but at the very least location services are working again.

Thanks for sharing! This works for me.

bootlegseahorse, 2022-12-02:
> adding the boot-arg io80211.awdl=0 to opencore fixed location service for me. Still not seeing anything show up in the wifi network scan but at the very least location services are working again.

Thank you for sharing, but my location service is still broken after adding "io80211.awdl=0" in boot-args with my Intel AX200 card.

anaander, 2022-12-04:
I think I was wrong, the boot arg was a red herring and didn't fix it. Location service would work ONCE after boot but then fail to work after that. Strangely, mapping my USB ports properly seems to have had some effect. It's the only thing I've changed today and location services are working with my AX210 card 100% of the time now. 

Specifically I set the USB port associated with the bluetooth to internal. I mean I mapped all the other ports, but that's the only one that could possibly interact with the wireless in any capacity.

[image]

FieryFlames, 2022-12-15:
> Having the same issue on my Intel Dual Band Wireless AC 8260

The latest 2.2.0 Alpha fixed this for me.

qiqco, 2022-12-21:
> adding the boot-arg io80211.awdl=0 to opencore fixed location service for me. Still not seeing anything show up in the wifi network scan but at the very least location services are working again.

我是AX200，添加这个参数后定位正常。Ventura13.1。谢谢🙏

bootlegseahorse, 2022-12-30:
FYI: There is no permanent solution yet, but "sudo pkill airportd" fixes the location/scan issue instantly. 

If you're lucky enough you might get it to work constantly (until your computer sleeps), but in most cases (90% at least) it would only work for a minute or two. 

Anyway, anytime you need location service with your Intel WiFi card, just do sudo pkill airportd. It's only a temporary fix but it will work.

williambj1, 2023-06-01:
Fixed in https://github.com/OpenIntelWireless/itlwm/releases/tag/v2.2.0

closed 2023-06-01 by williambj1