OpenIntelWireless/itlwm #805: Slow download speed on IPv6 with 802.11ac Wi-Fi

vasishath, 2022-08-25:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version:  AirportItlwm v2.2.0-alpha (commit id d404254f2f9fd46d21ca53b10e1afea59631bbdd)
- WiFi Card Model: Intel 8265
- PCI Product ID: 0x8086 0x24FD
- macOS Version: BigSur 11.1

**Description**
When connected to a 802.11ac (5GHz) network with IPv6 enabled, the download speed drops drastically. My internet plan offers 150Mbps down and up but I get only 5-7Mbps with IPv6 enabled. Wireshark shows lots of TCP retransmissions when this occurs. The upload speed is unaffected though. This affects all websites that support IPv6. On disabling IPv6, the speed becomes normal. 
My other devices get normal speed during the same time on the same website (fast.com) with IPv6.
If I switch to 2.4GHz Wi-Fi, then I get 60-70Mbps with IPv6 enabled which is normal for 2.4G. I am attaching screenshots of my speedtest.

**5GHz wifi and IPv6 disabled (normal)**:
[image]

**5GHz wifi and IPv6 enabled**:
[image]

**2.4GHz wifi and IPv6 enabled (normal)**:
[image]

**Bug Report Archive**
[Log_2022-08-25_15-02-07.log](https://github.com/OpenIntelWireless/itlwm/files/9423429/Log_2022-08-25_15-02-07.log)

**Kext Download Source**
This repo

profzei, 2022-09-22:
Hi,
same here: I can confirm the issue!

WiFi Card Model: Intel 8275
PCI Product ID: 0x8086 0x3E02
macOS Version: latest Big Sur, latest Monterey, Ventura developer beta 8

The issue was **not** present in AirportItlwm v2.2.0-alpha (commit id [`6a804ef`](https://github.com/OpenIntelWireless/itlwm/commit/6a804ef81215634a5ea3547782e8d3741042f9b1)) since I could achieve full speed on IPv6 with 802.11ac Wi-Fi while it seems to be already present in AirportItlwm v2.2.0-alpha (commit id [`34441bc`](https://github.com/OpenIntelWireless/itlwm/commit/34441bc68ad35b57d4299f5d28d9ff7879beed4f))... all subsequent releases seem to have the same issue (at least in my case)!

usr-sse2, 2022-10-25:
Oh, maybe you've found out what's wrong with my network: it has IPv6, and some things like Apple Music and Yandex Maps work terribly slow, while other thinks (which have only IPv4) work OK, and having an old router after the main one fixed the issue. I'll try disabling IPv6 and checking again.

TheCrypt0, 2022-10-26:
Can confirm this issue. Downloading macOS Ventura was going at about ~300KB/s. Once set the IPv6 mode to Link-Local Only the speed was back around ~40MB/s. 

How to temporarily disable IPv6: [Link (scroll down and skip the Windows part)](https://support.blindsidenetworks.com/hc/en-us/articles/360050168052-How-to-disable-IPV6)

williambj1, 2022-10-30:
Can this be reproduced with `itlwm.kext`? Neither mine nor @zxystd's ISP provide IPV6 so it is hard to verify and debug.

vasishath, 2022-10-30:
> Can this be reproduced with `itlwm.kext`? Neither mine nor @zxystd's ISP provide IPV6 so it is hard to verify and debug.

Yes. It is reproduced with both itlwm and AirportItlwm.

williambj1, 2022-10-30:
Thank you for the quick confirmation.

> Maybe try communicating between 2 systems in a LAN using link-local address?

I'm planning on setting up a file server on my RPI4 but I'm extremely busy these days. Unfortunately it might take me some time to start testing.

vasishath, 2022-10-30:
> Thank you for the quick confirmation.
> 
> > Maybe try communicating between 2 systems in a LAN using link-local address?
> 
> I'm planning on setting up a file server on my RPI4 but I'm extremely busy these days. Unfortunately it might take me some time to start testing.

No need for that.. you can just use`iperf3` to test speed. I did it on my system with iperf3 server running on RPi2 and here are the results.
[image]

[image]

zxystd, 2022-10-30:
Can you please help to test if the history versions also have the same problem?

vasishath, 2022-10-31:
> Can you please help to test if the history versions also have the same problem?

you can check with @profzei's comment above where they say that it worked with commit id 6a804ef. Unfortunately, I can't confirm this as I tried building the kext after checking out the repo on this commit but the kext didn't load. (Probably because my xcode is old since I am on big sur). I also tried using the stable 2.1.0 version but the kext was too big for my EFI volume.

wimdoan, 2022-11-25:
ipv6网速确实很慢，爱奇艺根本没法播放，用有线连接就没问题，把系统切到windows wifi连接也没问题，所以还是驱动哪里不合适

https://user-images.githubusercontent.com/1529662/203929669-29c8cab4-4c4b-4bc6-9f1e-87e6b20cde9b.mov

wimdoan, 2022-11-25:
> ipv6网速确实很慢，爱奇艺根本没法播放，用有线连接就没问题，把系统切到windows wifi连接也没问题，所以还是驱动哪里不合适
> 
>  2022-11-25.15.442.mov

发现此问题并非ipv6独有的，对于此域名ipv4解析也同样慢，但网线连接就没有这样的问题，不知道会是什么问题

wimdoan, 2022-11-25:
关闭11ac,这些卡的链接就不会卡了，但最高网速就只能跑100m左右了

wimdoan, 2022-11-26:
ac模式下，目前就发现爱奇艺这个cdn域名特别慢(ib302.zjcdn.com)，ipv4 v6都慢，现在只能单独让这个域名走代理服务器，才能解决网速问题了
[image]

elgab, 2024-07-27:
7260HMW + Sonoma, with itlwm_v2.3.0_stable.kext.zip  
had to block IPv6 as mentioned by TheCrypt0 to have fast speeds on 5Ghz AC.
tried both the -novht and -noht40 but did not help to make ipv6 usable; so now is working super nice on IPv4 👍