OpenIntelWireless/itlwm #813: Not seeing wifi

herotm, 2022-09-27:
Hackintosh (Monterey 12.6) is installed on a Proxmox VM

Passthrough the M2 Wifi+Bluetooth AX210 to the VM ( Tested passthrough on Windows 10, works perfectly fine so i would assume hardware is okay here )

Downloaded the Airportitlwm Monterey release(2.1.0), pasted the kext to my bootloader kext folder, used Propertree and edited the config to include the kext, rebooted.

tom@iMac-Pro ~ % kextstat | grep -i "Air*"       
Executing: /usr/bin/kmutil showloaded
No variant specified, falling back to release
Index Refs Address            Size       Wired      Name (Version) UUID <Linked Against>
   94    0 0                  0xfa4000   0xfa4000   com.zxystd.AirportItlwm (2.1.0) FA02D059-D593-3D8E-A031-1F99EA4D67EB <93 49 16 7 6 3 1>
tom@iMac-Pro ~ % 

  Software Versions:
  CoreWLAN:	16.0 (1657)
  CoreWLANKit:	16.0 (1657)
  Menu Extra:	17.0 (1728)
  System Information:	15.0 (1502)
  IO80211 Family:	12.0 (1200.12.2b2)
  Diagnostics:	11.0 (1163)
  AirPort Utility:	6.3.9 (639.16)

Sorry, total newb here let me know what other infos are missing.

williambj1, 2022-09-27:
Please follow the instructions from the documentation and post logs from dmesg.

herotm, 2022-10-06:
I do apologize, its a problem with me not enabling pcie passthrough option, it works dandy now.

Thank you

closed 2022-10-06 by herotm