OpenIntelWireless/itlwm #814: Can't see any netoworks Wireless 7265

franchioping, 2022-10-01:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes
well it very much might just be user error

**Environment**

- Kext Version: itlwm 2.1.0
- WiFi Card Model: Wireless 7265
- PCI Product ID: 8086:095a (what shows in "lspci -nn" on linux)
- macOS Version: 12.6

**Description**
No networks are shown, cant manually connect to any network either

**Bug Report Archive**
[bugreport_41128.zip](https://github.com/OpenIntelWireless/itlwm/files/9689958/bugreport_41128.zip)

**Kext Download Source**
Github Releases

franchioping, 2022-10-04:
I have since tested this with a diferent model wifi card, same result. it is most likely my problem and not a bug

closed 2022-10-23 by franchioping

williambj1, 2023-01-07:
>    1.688839]: itlwm: iwm_check_rfkill RF_KILL hw: 1

Wi-Fi is locked by RFKILL by the motherboard.