OpenIntelWireless/itlwm #817: 每次睡眠唤醒之后，wifi要等5分钟才能使用

sweliar, 2022-10-15:
各位大佬辛苦了，帮我看看这个问题。
我的驱动版本是AirportItlwm-Ventura-v2.2.0，每次睡眠后wifi都要等5分钟左右才能使用，每次都复现。
以下是log信息：

[image]

Sat Oct 15 20:20:41.003 <kernel> IO80211VirtualInterface::controllerWillChangePowerState : Bringing down link
Sat Oct 15 20:20:41.052 <kernel> en1: setting diversity to: -1
Sat Oct 15 20:20:41.052 <kernel> en1: Error configuring antenna diversity (index = -1).
Sat Oct 15 20:20:41.052 <kernel> en1: setting tx antenna: -1
Sat Oct 15 20:20:41.052 <kernel> en1: Error configuring transmit antenna (index = -1).
Sat Oct 15 20:20:46.215 <kernel> Not adding subscription due to missing boot-arg wlan.iorsubscriptions
Sat Oct 15 20:20:46.215 <kernel>  Creating all peerManager reporters
Sat Oct 15 20:21:16.217 <kernel> Not adding subscription due to missing boot-arg wlan.iorsubscriptions
Sat Oct 15 20:21:16.218 <kernel>  Creating all peerManager reporters
Sat Oct 15 20:21:16.291 Usb Host Notification hostNotificationUSBDeviceRemoved HD Camera(unseen product name) seqNum 3 Total -4
Sat Oct 15 20:21:16.291 Usb Host Notification hostNotificationUSBDeviceRemoved HD Camera(unseen product name) seqNum 3 Total -5
Sat Oct 15 20:21:16.291 Usb Host Notification hostNotificationUSBDeviceRemoved HD Camera(unseen product name) seqNum 3 Total -6
Sat Oct 15 20:21:16.307 Usb Host Notification Error Apple80211Set: Input/output error seqNum 3 Total -6 chg 0 en1
Sat Oct 15 20:21:46.219 <kernel> Not adding subscription due to missing boot-arg wlan.iorsubscriptions
Sat Oct 15 20:21:46.219 <kernel>  Creating all peerManager reporters
Sat Oct 15 20:22:16.222 <kernel> Not adding subscription due to missing boot-arg wlan.iorsubscriptions
Sat Oct 15 20:22:16.224 <kernel>  Creating all peerManager reporters
Sat Oct 15 20:22:46.231 <kernel> Not adding subscription due to missing boot-arg wlan.iorsubscriptions
Sat Oct 15 20:22:46.231 <kernel>  Creating all peerManager reporters
Sat Oct 15 20:23:16.233 <kernel> Not adding subscription due to missing boot-arg wlan.iorsubscriptions
Sat Oct 15 20:23:16.233 <kernel>  Creating all peerManager reporters
Sat Oct 15 20:23:46.240 <kernel> Not adding subscription due to missing boot-arg wlan.iorsubscriptions
Sat Oct 15 20:23:46.240 <kernel>  Creating all peerManager reporters
Sat Oct 15 20:24:21.320 <kernel> Unexpected payload found for message 9, dataLen 0
Sat Oct 15 20:24:21.320 <kernel> Error storing postMessage 9
Sat Oct 15 20:24:21.323 <kernel> postMessage::1443 APPLE80211_M_BSSID_CHANGED received
Sat Oct 15 20:24:21.323  <kernel> No InterfaceAverageCCA
Sat Oct 15 20:24:21.323 <kernel> IO80211ControllerMonitor::getChanCCA no fCCAReporterValue
Sat Oct 15 20:24:21.323 <kernel> Failed to getChanCCA e00002f0 161 0 
Sat Oct 15 20:24:21.323 <kernel> No InterfaceAverageCCA
Sat Oct 15 20:24:21.323 <kernel> IO80211ControllerMonitor::getChanCCA no fCCAReporterValue
Sat Oct 15 20:24:21.323 <kernel> Failed to getChanCCA e00002f0 161 0 
Sat Oct 15 20:26:14.182 <kernel>  IOCTL (from pid 291) not recognized: 383 out of 417
Sat Oct 15 20:26:14.182 <kernel>  IOCTL (from pid 291) not recognized: 383 out of 417

liu6s6, 2022-10-29:
期待作者的更新！

zh604131924, 2022-10-29:
同样的问题

zxystd, 2022-10-30:
Can use itlwm.kext instead for temporary solution

kingkongdog, 2022-10-30:
> 

Can wifi work if using itlwm.kext?

kingkongdog, 2022-10-30:
> Can use itlwm.kext instead for temporary solution

Can wifi work if using itlwm.kext?

sweliar, 2022-10-31:
> 可以使用 itlwm.kext 代替临时解决方案

好的，我去试试，谢谢

kingkongdog, 2022-11-03:
解决了吗？

sweliar, 2022-11-03:
> 解决了吗？

Yes，you can used to itlwm, but GPS won't work because the wifi doesn't have a drive

williambj1, 2022-11-17:
Duplicate of #283, closing this one since the other follows the issue template.

williambj1, 2022-11-17:
Duplicate of #823

closed 2022-11-17 by williambj1

taotaoaaaaaa, 2022-11-21:
同样，开机进入系统之后要菜单栏要很久才显示图标，显示出Wi-Fi图标后要很久才能连接上Wi-Fi，并且只能搜索到一个Wi-Fi。设置里面Wi-Fi是完全不会显示连接到Wi-Fi的一直是显示Wi-Fi未连接，只能通过菜单栏连接！！！