OpenIntelWireless/itlwm #821: ax201 macos13  wifi崩溃，蓝牙正常

songkea, 2022-10-30:
**Have You Read Our Docs**

**Are You Reporting A Bug**
蓝牙使用正常可以接力，但是wifi只要开机启动就会崩溃重启电脑； 事情发生在进入桌面一分钟左右，wifi打开则电脑崩溃。 
而系统崩溃报告中显示，确实是wifi问题导致崩溃，请大神帮看看是哪里的问题。
AirportItlwm.kext 2.2.0

**Environment**

- Kext Version: 
- WiFi Card Model: ax201
- PCI Product ID: 
- macOS Version: macOS 13 ventura

**Description**

panic(cpu 10 caller 0xffffff800074aee3): Kernel trap at 0xffffff8000301057, type 13=general protection, registers:
CR0: 0x000000008001003b, CR2: 0x0000700005e74fd8, CR3: 0x0000000007923000, CR4: 0x00000000003626e0
RAX: 0xaaaaaaaaaaaaaaaa, RBX: 0xffffffe2685ff610, RCX: 0x0000000000000141, RDX: 0x0000000000000141
RSP: 0xffffffe2685ff528, RBP: 0xffffffe2685ff560, RSI: 0xffffff903f2e0d08, RDI: 0xaaaaaaaaaaaaaaaa
R8:  0x0000000000000141, R9:  0x00000000ffffffff, R10: 0x0000000000000001, R11: 0x0000000000000002
R12: 0x0000000000000002, R13: 0x0000000000000030, R14: 0xffffff950e6a22f0, R15: 0xffffff950e6a22f0
RFL: 0x0000000000010286, RIP: 0xffffff8000301057, CS:  0x0000000000000008, SS:  0x0000000000000010
Fault CR2: 0x0000700005e74fd8, Error code: 0x0000000000000000, Fault CPU: 0xa, PL: 0, VF: 0

Panicked task 0xffffff9042458218: 238 threads: pid 0: kernel_task
Backtrace (CPU 10), panicked thread: 0xffffff950f317b30, Frame : Return Address
0xffffff80003111a0 : 0xffffff80005edf9d 
0xffffff80003111f0 : 0xffffff800075b786 
0xffffff8000311230 : 0xffffff800074aa10 
0xffffff8000311280 : 0xffffff8000588951 
0xffffff80003112a0 : 0xffffff80005ee27d 
0xffffff8000311390 : 0xffffff80005ed929 
0xffffff80003113f0 : 0xffffff8000de0ecb 
0xffffff80003114e0 : 0xffffff800074aee3 
0xffffff8000311660 : 0xffffff800074abd2 
0xffffff80003116b0 : 0xffffff8000588951 
0xffffff80003116d0 : 0xffffff8000301057 
0xffffffe2685ff560 : 0xffffff80045c76b5 
0xffffffe2685ff5e0 : 0xffffff800230f57d 
0xffffffe2685ffa40 : 0xffffff800230da87 
0xffffffe2685ffbf0 : 0xffffff80022db761 
0xffffffe2685ffc60 : 0xffffff800451e970 
0xffffffe2685ffce0 : 0xffffff800451661b 
0xffffffe2685ffd50 : 0xffffff8000d162f4 
0xffffffe2685ffdb0 : 0xffffff8004516536 
0xffffffe2685ffdf0 : 0xffffff800458f547 
0xffffffe2685ffe50 : 0xffffff800458fbc6 
0xffffffe2685ffed0 : 0xffffff8000d14d04 
0xffffffe2685fff20 : 0xffffff8000d135fe 
0xffffffe2685fff60 : 0xffffff8000d12c37 
0xffffffe2685fffa0 : 0xffffff800058819e 
      Kernel Extensions in backtrace:
         com.apple.iokit.IO80211FamilyLegacy(1200.12.2b1)[E84C2807-ECFC-38F9-9EFB-91740FDC2199]@0xffffff800223f000->0xffffff8002384fff
            dependency: com.apple.driver.AppleMobileFileIntegrity(1.0.5)[2E7C729D-2485-32DC-B478-D490A81E1419]@0xffffff8001b91000->0xffffff8001bbffff
            dependency: com.apple.driver.corecapture(1.0.4)[EB19F50F-F267-3463-BBCF-777D4AAFD797]@0xffffff80037ee000->0xffffff8003810fff
            dependency: com.apple.iokit.CoreAnalyticsFamily(1)[72620FD1-D89B-3033-B43E-D883078906E1]@0xffffff8001edb000->0xffffff8001ee4fff
            dependency: com.apple.iokit.IONetworkingFamily(3.4)[B4C9DF98-D9E5-3460-967C-55853B28D162]@0xffffff8002d9e000->0xffffff8002db4fff
            dependency: com.apple.iokit.IOSkywalkFamily(1.0)[91CA74BE-8DD8-38F9-9584-6C872AC07072]@0xffffff80030e8000->0xffffff8003115fff
            dependency: com.apple.kec.corecrypto(12.0)[A589451C-2D9B-3094-B16B-9FBA03EF7EAD]@0xffffff800381f000->0xffffff8003896fff
         com.zxystd.AirportItlwm(2.0)[8D72B887-C419-3630-9D5A-1538E67DFD89]@0xffffff8004516000->0xffffff80051fbfff
            dependency: com.apple.iokit.IO80211FamilyLegacy(1200.12.2b1)[E84C2807-ECFC-38F9-9EFB-91740FDC2199]@0xffffff800223f000->0xffffff8002384fff
            dependency: com.apple.iokit.IONetworkingFamily(3.4)[B4C9DF98-D9E5-3460-967C-55853B28D162]@0xffffff8002d9e000->0xffffff8002db4fff
            dependency: com.apple.iokit.IOPCIFamily(2.9)[E2CE416B-59DA-3836-9985-FDC527C5E6C7]@0xffffff8002fff000->0xffffff800302efff

Process name corresponding to current thread (0xffffff950f317b30): kernel_task
Boot args: -v -alcbeta -lilubetaall

Mac OS version:
22A380
[... 164 lines cut ...]
>!AEFINVRAM	2.1
>!AEFIRuntime	2.1
|IOHID!F	2.0.0
|IOTimeSync!F	1110.13
|IOSkywalk!F	1.0
>mDNSOffloadUserClient	1.0.1b8
|IONetworking!F	3.4
>DiskImages	493.0.0
|IO!B!F	9.0.0
|IOReport!F	47
$quarantine	4
$sandbox	300.0
@kext.!AMatch	1.0.0d1
>!ASSE	1.0
>!AKeyStore	2
>!UTDM	547
|IOUSBMass!SDriver	232
|IOSCSIBlockCommandsDevice	476
|IO!S!F	2.1
|IOSCSIArchitectureModel!F	476
>!AMobileFileIntegrity	1.0.5
$!AImage4	5.0.0
@kext.CoreTrust	1
>!AFDEKeyStore	28.30
>!AEffaceable!S	1.0

zxystd, 2022-10-30:
你的已加载的kext列表里明明白白地写着: com.zxystd.AirportItlwm 2.0.0

closed 2022-10-30 by zxystd

songkea, 2022-10-30:
感谢回复，问题确实是这样的，是我的疏忽，当我换成2.2版本后，确实正常了，wifi功能和蓝牙功能稳如老狗... 
ax201 ，接力正常；  感谢您，祝您生活愉快