OpenIntelWireless/itlwm #823: Ventura: Wifi takes 2-3 minutes to work after booting

jamesst20, 2022-10-30:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: 2.2.0 - ee567086f288951766f4259f8239c472be66679f
- WiFi Card Model: AX201 from ROG STRIX B660-I GAMING WIFI (Alder lake)
- PCI Product ID: 0x8086, 0x74
- Firmware Version: itlwm: 2.2.0_ee56708 fw: 68.01d30b0c.0
- macOS Version: 13.0

**Description**
Note: I upgraded from Monterrey today to Ventura. Prior updating I did not have any issue.

When I boot my computer, the wifi takes a lot of time to start working. Maybe 2-3 minutes. It says "Searching..." where there should be SSIDs found and if I just wait few minutes it starts working, find networks and automatically connect.

There is also something else weird that might or might not be related too: When I boot my computer, the clock on the status bar (top right) doesn't show up maybe for the first 30 minutes. No idea if it's related to some loading generated by this kext or not. All I can say is that it ain't happening on my M1 Macbook Pro

**Bug Report Archive**
1. Install DebugEnhancer
2. Reboot
3. `sudo /System/Library/PrivateFrameworks/Apple80211.framework/Versions/A/Resources/airport debug +AllUserland +AllDriver +AllVendor`
4. `sudo dmesg | grep -E "itlwm|Airport|IO80211|EAPOL" > ~/Desktop/Log_"$(date '+%Y-%m-%d_%H-%M-%S')".log`

[Log_2022-10-30_16-26-17.log](https://github.com/OpenIntelWireless/itlwm/files/9896773/Log_2022-10-30_16-26-17.log)

**Kext Download Source**
Github Release Alpha

bisaalz, 2022-11-01:
Yes I am facing exact problem

izwb003, 2022-11-01:
Same problem, but also get stuck if the wifi turns on with the system... The system will stuck in that condition. I have to turn off wifi immediately once the system booted into the desktop to prevent such a panic, and turn it back on after few seconds, then I have to wait for 2-3 minutes for it to search a WIFI...

leo10ui, 2022-11-02:
I'm with the same problem

mishurov, 2022-11-02:
Same sh*t.

SySpetsnaz, 2022-11-03:
hello excuse me but i am new to hackintosh i have the same Intel ax201 but i dont know what to download and where to put the file.
Can someone help me please ?

thank you in advance

Core-i99, 2022-11-03:
> hello excuse me but i am new to hackintosh i have the same Intel ax201 but i dont know what to download and where to put the file.
> Can someone help me please ?
> 
> thank you in advance

This is not a support forum

SySpetsnaz, 2022-11-03:
> > hello excuse me but i am new to hackintosh i have the same Intel ax201 but i dont know what to download and where to put the file.
> > Can someone help me please ?
> > thank you in advance
> 
> This is not a support forum

sorry for this do you know where i can ask for help ?

williambj1, 2022-11-04:
We are aware of this issue but unfortunately it's near the end of the year and maintainers / testers are extremely busy. Sorry for the inconvenience. A possible workaround is to use `itlwm.kext` for now.

GotierLdl, 2022-11-05:
Hello, same issue on Intel Dual Band Wireless AC 9560 (DELL Latitude 5500)

bilijp153, 2022-11-06:
我的 intel Wireless AC 9461 也是有一个问题

ameenjuz, 2022-11-08:
i have also same issue ax201 dell latitude 5410

dominic-03, 2022-11-10:
Same issue on AX210 (standalone WiFi card, Aorus branded).

Xryphon, 2022-11-11:
On another side note, I can see the Wi - Fi networks but the kext does not start "working", per se, around a minute after boot. I guess this is just a side effect of Ventura?

james-em, 2022-11-11:
> On another side note, I can see the Wi - Fi networks but the kext does not start "working", per se, around a minute after boot. I guess this is just a side effect of Ventura?

Yes like it was said. Works fine itlwm+heliport though

mmtou, 2022-11-16:
Same issue on AX200

joaodematejr, 2022-11-18:
Same issue

ahmedorko, 2022-11-18:
Its on Alpha stage. We have to wait until final release

taotaoaaaaaa, 2022-11-21:
同样，开机进入系统之后要菜单栏要很久才显示图标，显示出Wi-Fi图标后要很久才能连接上Wi-Fi，并且只能搜索到一个Wi-Fi。设置里面Wi-Fi是完全不会显示连接到Wi-Fi的一直是显示Wi-Fi未连接，只能通过菜单栏连接！！！

meteoro, 2022-11-26:
Same situation, waiting for stable version. Thanks.

taotaoaaaaaa, 2022-11-26:
OK，Thanks for your reply.

 

970644796
***@***.***

TomHeaven, 2022-11-30:
In my case, it takes about 30 seconds to connect to WiFi after login. I'm using an Intel Wireless 9560 AC card.

TomHeaven, 2022-12-01:
A possible workaround：Loo at this [repo](https://github.com/tlefko/Matebook-X-Pro-2018-Ventura/releases/tag/venturamajor), which claims to solve the WiFi lantency issue on Ventura.
The author loads Catalina version of AirportItlwm.kext first in Opencore, and load the Ventura version latter.
I'll try that method to give further feedbacks.

alsdk98, 2022-12-02:
same on intel wifi 6 ax201 tongfang gk7mr0r

aneeshlingala, 2022-12-22:
+1 on macOS Ventura 13.1

It is a little bit faster for me when connecting manually with your network name and password.

verygenericname, 2022-12-27:
same problem on a Lenovo Yoga 260

qipuaw664, 2022-12-30:
Any update？Ventura with AX200 has this  issue.

andrebadini, 2023-01-05:
Same issue with notebook Dell G15 5510 - Intel® Core™ i5-10500H - Killer Wi-Fi 6 AX1650 (2x2).

kevinlieser, 2023-01-12:
Same here. WiFi Switches are not synchronized in menu bar and system settings. Takes a few minutes to connect to the WiFi then only 30mbit fast.

AX210

DC4114, 2023-01-23:
I'm facing this issue along with wifi not connecting after sleep on my Intel Wireless-AC 9560 160MHz Wi-Fi.

jeanmartin13, 2023-01-27:
Same issue on Ventura 13.1. Wi-Fi  (0x8086, 0x74), Asus Tuf Gaming Z790 WIFI D4 MoBo, itlwm: 2.2.0_e0f745e fw: 68.01d30b0c.0.

It also looks like it perturbs the mouse cursor behavior the time the WiFi is fully connected (stucked stuttering cursor) at least at first boot.

yzx1208, 2023-02-04:
the same problem.

rboldini, 2023-02-04:
Same here with 7265, after sleep it takes some minutes to find networks. As additional observation, some times happened with this issue running, that disabling Bluetooth, from menu bar toggle, also turn off my Wi-Fi. 

-------
Probably useless.
-------
Instant Hotspot is displayed instantly, but as known doesn't work.
Infinity clipboard, instead, stop working after connection if this issue happened and device was not rebooted.

Unfortunately no log, I'm sorry.

jclsn, 2023-02-24:
For me it just crashes and MacOS reboots unfortunately...

kevinlieser, 2023-04-02:
So, I updated to 13.3 and tried again with 2.2.0-alpha.

The bad thing is that I switched the WiFi Repeater yesterday to a lower model since I need the better model in another room. The good thing is I am not stuck at 30 MBit, I got 120 MBit – around the same as on iPhone (90 MBit) (even a little bit faster). I will let it active, but on activation it took 1 minute to find the first WiFi networks. But connects fast and acts fast.

adolfomoro, 2023-04-09:
Aorus X570 Pro Wifi with AX200 Embedded no list networks and can't connect. Solution?

zxystd, 2023-05-26:
Try the latest alpha version.

buzzcoding, 2023-05-26:
Sending this from MacOS Ventura 13.3 with Intel Wi-Fi 6 AX200 and the latest alpha version of AirportItlwm. The issues are fixed. Previously, when I have tested the older commits, it would not show any networks when searching, but now it works with no issues.

seanleavy, 2023-06-01:
Looking good on 13.4 as well!

williambj1, 2023-06-02:
https://github.com/OpenIntelWireless/itlwm/releases/tag/v2.2.0

closed 2023-06-02 by williambj1

meastwoods, 2023-06-20:
I have the same problem on 11.7.7. Possibly started happening last week after my Macbook refused to shut down and I was forced to hold down the power switch.