OpenIntelWireless/itlwm #827: AC 3165 can't find networks

DreamLin-233, 2022-11-13:
**Have You Read Our Docs**

Yes
**Are You Reporting A Bug**

Yes
**Environment**

- Kext Version: Airportitlwm v2.1.0
- WiFi Card Model: Intel(R) Dual Band Wireless AC 3165
- PCI Product ID: (不会查)
- macOS Version: 12.6

**Description**
kext&device working.can turn on wifi but list is empty.
(驱动 设备都在 但是找不到网络)

**Bug Report Archive**
[bugreport_60249.zip](https://github.com/OpenIntelWireless/itlwm/files/9997345/bugreport_60249.zip) in HeliPort.
install DebugEnhancer.kext,try to Enable airportd logs,but failed.（kext is working）
"Unable to retrieve vendor debug files...ignoring error.
Unable to set vendor debug flags...ignoring error."

**Kext Download Source**
[AirportItlwm_v2.1.0_stable_Monterey.kext.zip](https://github.com/OpenIntelWireless/itlwm/releases/download/v2.1.0/AirportItlwm_v2.1.0_stable_Monterey.kext.zip)

closed 2022-11-14 by zxystd

williambj1, 2022-11-16:
Device is locked by RFKILL, unrelated user configuration error.

DreamLin-233, 2022-11-16:
> Device is locked by RFKILL, unrelated user configuration error.

so what can i do now...?