OpenIntelWireless/itlwm #830: Bug on network scaning on mac os ventura

johnpapad24, 2022-11-28:
Bug report:

**Environment**

- Kext Version: AirportItlwm 2.20-alpha (Ventura) / AirportItlwm 2.1.0 (montrey but used on ventura)
- WiFi Card Model: Intel Wireless-AC 8260
- PCI Product ID: vendor id: 0x8086 
                               device id: 0x2f3 
                               subsystem vendor id: 0x8086
                               subsystem id: 0x1130
                               reversion id: 0x003a
- macOS Version: 13.0.1

**Description**
AirportItlwm on mac os ventura loads and detects the card but does not find any networks after scanning (and also does not connect by giving wifi ssid manually).

The simple Itlwm is working ok

**Kext Download Source**
github/OpenIntelWireless

trongtinh1212, 2022-11-29:
same issue with AX201 , downgrade to Monterey and monterey kext seem fine

trongtinh1212, 2022-12-01:
you should try this, add this line to boot-arg  ""io80211.awdl=0"

andreszerocross, 2022-12-02:
> you should try this, add this line to boot-arg ""io80211.awdl=0" without the quotes

Where did you get this kind of boot argument?? it's didn't mention in this manual or read.me files

trongtinh1212, 2022-12-02:
> > you should try this, add this line to boot-arg ""io80211.awdl=0" without the quotes
> 
> Where did you get this kind of boot argument?? it's didn't mention in this manual or read.me files

https://gitter.im/OpenIntelWireless/itlwm?at=63876ab32cb65702404f1d31

on gitter

bootlegseahorse, 2022-12-02:
> you should try this, add this line to boot-arg ""io80211.awdl=0" without the quotes

Thanks for sharing. Still problematic with my AX200. No location service nor network scan.

johnpapad24, 2022-12-08:
io80211.awdl=0 and latest airportitlwm 2.2.0-alpha (fadf42c) unfortunatelly didn't fix the problem

AuthenTiiq, 2022-12-08:
Same for me, the last Airportitlwm 2.2.0-alpha doesn't work.

I can't find any WiFi network.
The WiFi menu says "No Internet Connection..." and when I click on it, a window says "Your Mac successfully joined the Wi-Fi network, but cannot reach Internet."

Except that I haven't joined any network, since I can't see them...

expo1337, 2022-12-18:
Same on Intel AC 8265.

bootlegseahorse, 2022-12-31:
Try sudo pkill airportd. It fixes the location/scan issue instantly. If you're lucky enough you might get it to work constantly (until your computer sleeps/restarts), but in most cases (90% at least) it would only work temporarily for a minute or two.

AuthenTiiq, 2023-01-01:
it finally works for me with the new build from yesterday

mw3200, 2023-01-16:
My ax201 can work with AirportItlwm2.20-alpha just a little bit, but the wifi scanning is very slow, and usually have no location service.

mw3200, 2023-01-19:
> you should try this, add this line to boot-arg ""io80211.awdl=0" without the quotes

Unfortunately it can't solve the problem on intel ax201

AuthenTiiq, 2023-01-19:
> My ax201 can work with AirportItlwm2.20-alpha just a little bit, but the wifi scanning is very slow, and usually have no location service.

Same problem for me with AX210, I downgrade to Monterey :/
With Ventura Airportitlwm random kernel panic and Wireless scanning is semi-broken in Settings apps

wwoltz, 2023-02-02:
Having similar issues running Ventura on a Lenovo Thinkpad T460s. I enable the WiFi and it won't display any networks. If I watch the display closely, I can see the display flicker where it says "No Networks", as if something is happening during the scanning process for network SSIDs. It will continue to flicker like this and doesn't seem to stop. I tried running itlwm-v2.2.0 with Heliport (disabled AirportItlwm-Ventura-v2.2.0 and was able to see and select a network and establish a Wifi connection running a hotspot on my phone. When trying to join a corporate guest network that displays a web page for authentication, I couldn't get in the network because the web page doesn't load and display. The attached video shows the flicker issue if you watch closely to about the halfway or 2 second mark. Watching the "No Networks" text, you will see it trying to update with something but immediately goes back to the same text.

https://user-images.githubusercontent.com/43864151/216388680-704d6d32-5724-494d-a923-c686393940ab.mp4

elfametesar, 2023-02-10:
> Having similar issues running Ventura on a Lenovo Thinkpad T460s. I enable the WiFi and it won't display any networks. If I watch the display closely, I can see the display flicker where it says "No Networks", as if something is happening during the scanning process for network SSIDs. It will continue to flicker like this and doesn't seem to stop. I tried running itlwm-v2.2.0 with Heliport (disabled AirportItlwm-Ventura-v2.2.0) and was able to see and select a network and establish a Wifi connection running a hotspot on my phone. When trying to join a corporate guest network that displays a web page for authentication, I couldn't get in the network because the web page doesn't load and display. The attached video shows the flicker issue if you watch closely to about the halfway or 2 second mark. Watching the "No Networks" text, you will see it trying to update with something but immediately goes back to the same text.
> 
>  20230202_102425.mp4

same but it doesn't work for me on itlwm 2.2.0 either. my card is wireless-n 7260

rforberger, 2023-02-13:
Having the same problem. Boot-arg io80211.awdl=0 doesn't help. Ventura with AirportItlwm 2.2.0-alpha.

Hashim1999164, 2023-02-14:
I don't know why? But I have same issue, My card is Intel 4965AGB 450MBPs, Behaving like everyone's elses.
I manage to connect with wifi network by following steps:
In wifi settings, add other wifi network, ok, and it will show error with option Run Diagnostics.
The diagnostics will ask for questions and will start it's process, just after it starts, you have a very limited time (like maximum 5-10 seconds) to check wifi networks list and only in that moment you can connect to wifi device, and then it will remain connected. but if you disconnect, it wont connect again and you'll have to repeat process but won't need to insert network key. i hope fix is out

pcking99, 2023-02-18:
Having same issues on Ventura. No WIFI found and the boot-arg didn't work.

Williamtung1, 2023-02-21:
i am using Intel AX211 on macOS Ventura. Only itlwm works and Airportitlwm doesn't. Having the same issues above that my devices can't scan any Wi-Fi

weiwenhuaming, 2023-02-22:
i am using Intel AX211 on macOS Ventura 13.2.1 with the beta version for Ventura(2.2.0). Only itlwm works and Airportitlwm doesn't. Let me add Boot-arg io80211.awdl=0, i  got boot failed.

Then i set SecureBootModel = j160 or j137, it  boot failed again.

Then i reset boot-args and SecureBootModel ,it  boot failed again and again ！！

Now i will reinstall Ventura. so sad~

t0niX, 2023-02-27:
Any update on this? Is somebody investigating or working on it?

Williamtung1, 2023-02-27:
FYI, it is 3 weeks since their latest commit. I think they will eventually work it out though. Probably ongoing

Sunshow, 2023-02-27:
Same, but if you ever stored a SSID, it could connected to it after long long waiting after first reboot, and if waking from deep sleep the wireless connection resume immediately, I live on this everyday.

weiwenhuaming, 2023-02-28:
> 
I live on this everyday too, my Ethernet kernrl panic with SSDT-I225V.aml and AppleIntelI210Ethernet.kext, i disable it.I live on this everyday too., my Ethernet kernrl panic with SSDT-I225V.aml and AppleIntelI210Ethernet.kext, i disable it.

akashcraft, 2023-02-28:
I have AX201. Nothing works for me on Ventura (Airportitlwm nor itlwm, not even alpha). Boot Arguments no luck. Currently, I have downgraded to Monterey and am using v2.1.0 stable just like many of you. Let us wait a bit.
To the devs: thanks for your hard work in advance!

Williamtung1, 2023-02-28:
> I have AX201. Nothing works for me on Ventura (Airportitlwm nor itlwm, not even alpha). Boot Arguments no luck. Currently, I have downgraded to Monterey and am using v2.1.0 stable just like many of you. Let us wait a bit. To the devs: thanks for your hard work in advance!

I think you will need to have your USB ports properly mapped and apple secure boot on as well

zxystd, 2023-03-01:
If you can't see the network list, you can manually click "other"->"join" and type the network ssid/password then it will connect.

t0niX, 2023-03-01:
> If you can't see the network list, you can manually click "other"->"join" and type the network ssid/password then it will connect.

Thanks a lot for your reply!

This is just a hotfix as location services (e.g. my position inside Apple Maps) don’t work on my end, too. Are there any plans/possibilities to fix these issues in the near future?

zxystd, 2023-03-01:
> Thanks a lot for your reply!
> 
> This is just a hotfix as location services (e.g. my position inside Apple Maps) don’t work on my end, too. Are there any plans/possibilities to fix these issues in the near future?

It is probablly hard to say, I am busy with life, and I don't know why the Ventura Control center not willing to accept wifi list, maybe I need a Broadcom card and hook the native WiFi driver to collect some information to compare with. But I don't have these conditions in hand now, my dev machine is E3v2+B75 which can not install Ventura as well.

wwoltz, 2023-03-01:
> If you can't see the network list, you can manually click "other"->"join" and type the network ssid/password then it will connect.

Manually adding would be great, the problem is that it won't let you manually add a network.  When trying to manually add a network, it pops up a network not found dialog.

alikaylan, 2023-04-05:
You don't need to revert the OS, or use heliport. Use the following commands to bypass the network selection window and do it with terminal:

/System/Library/PrivateFrameworks/Apple80211.framework/Versions/Current/Resources/airport -s
then
networksetup -setairportnetwork en0 <SSID_OF_NETWORK> <PASSWORD>

need to make sure en0 is correct for you. In my setup, it is en1. I find it in the network utility copied over from an older version of the OSX. If you don't have it, find it online.

Worked for me.

from: https://sharmank.medium.com/how-to-connect-to-wifi-using-mac-os-command-line-3a76c2e6669c

tmlibutan, 2023-05-05:
> 

Getting a Segmentation Fault: 11 with this. Trying other workarounds but doubtful and might just set up Monterey..

bdonvr, 2023-05-07:
> You don't need to revert the OS, or use heliport. Use the following commands to bypass the network selection window and do it with terminal:
> 
> /System/Library/PrivateFrameworks/Apple80211.framework/Versions/Current/Resources/airport -s then networksetup -setairportnetwork en0 <SSID_OF_NETWORK>
> 
> need to make sure en0 is correct for you. In my setup, it is en1. I find it in the network utility copied over from an older version of the OSX. If you don't have it, find it online.
> 
> Worked for me.
> 
> from: https://sharmank.medium.com/how-to-connect-to-wifi-using-mac-os-command-line-3a76c2e6669c

Like others I'm experiencing the exact same issue.

This command does show a scan of SSIDs, which is promising. However it gives `Could not find network my-ssid-here` when I attempt the second command.

hypercreeper, 2023-05-17:
I found a temporary fix:
Open System Settings and go to the Wi-Fi page and click on advanced and turn on the "show legacy networks and options" this should show the available networks on the Wi-Fi menu in the status bar, but not in system settings

Lavmint, 2023-05-19:
> Please don't let us down. It seems that if you won't fix the kext, no one will.
bruh, sounds like a lot of pressure. Anyway:
```
06: PCI 14.3: 0282 WLAN controller                              
  [Created at pci.386]
  Unique ID: Dhtl.E+Mbxg5ceyE
  SysFS ID: /devices/pci0000:00/0000:00:14.3
  SysFS BusID: 0000:00:14.3
  Hardware Class: network
  Model: "Intel Cannon Point-LP CNVi [Wireless-AC]"
  Vendor: pci 0x8086 "Intel Corporation"
  Device: pci 0x9df0 "Cannon Point-LP CNVi [Wireless-AC]"
  SubVendor: pci 0x8086 "Intel Corporation"
  SubDevice: pci 0x0034 
  Revision: 0x11
  Driver: "iwlwifi"
  Driver Modules: "iwlwifi"
  Device File: wlp0s20f3
  Features: WLAN
  Memory Range: 0xf1198000-0xf119bfff (rw,non-prefetchable)
  IRQ: 16 (no events)
  HW Address: 7c:b2:7d:fa:6b:f3
  Permanent HW Address: 7c:b2:7d:fa:6b:f3
  Link detected: yes
  WLAN channels: 1 2 3 4 5 6 7 8 9 10 11 12 13 36 40 44 48 52 56 60 64 100 104 108 112 116 120 124 128 132 136 140
  WLAN frequencies: 2.412 2.417 2.422 2.427 2.432 2.437 2.442 2.447 2.452 2.457 2.462 2.467 2.472 5.18 5.2 5.22 5.24 5.26 5.28 5.3 5.32 5.5 5.52 5.54 5.56 5.58 5.6 5.62 5.64 5.66 5.68 5.7
  WLAN encryption modes: WEP40 WEP104 TKIP CCMP
  WLAN authentication modes: open sharedkey wpa-psk wpa-eap
  Module Alias: "pci:v00008086d00009DF0sv00008086sd00000034bc02sc80i00"
  Driver Info #0:
    Driver Status: iwlwifi is active
    Driver Activation Cmd: "modprobe iwlwifi"
  Config Status: cfg=new, avail=yes, need=no, active=unknown
  ```

zxystd, 2023-05-26:
https://github.com/OpenIntelWireless/itlwm/issues/823#issuecomment-1564129817

Williamtung1, 2023-05-30:
For my AX211, with the version you updated yesterday, connecting to a known Wi-Fi works flawlessly and instantly on startup. However, I still can't discover any network in Settings. I am on Ventura 13.4. Thanks

alikaylan, 2023-05-30:
on AC 7265NGW and AX211 (also), everything seem to work. Settings show networks as well. Venture 13.4.

closed 2023-06-02 by zxystd

naka7a3, 2024-12-04:
Hello! I am happy to announce that i fixed this issue in my Acer Aspire Hackintosh machine, running Ventura 13.7.1 OS version with changed wi-fi card to intelAX210. So, how i did that...the answer is.. PURE luck! 

 After spending like a couple of days, sitting and watching this Wi-Fi signal searching, It indicates to me that it's working. So what I tried to do is to reinstall one by one each of the .kext files - airportitlwm, itlwm + Heliport; nothings happens.. and i was like, Ok, I'm going to give it one more chance before switching back to Windows.. 
 After spending hours trying to fix the problem (again) I was in the sofa, half awake, trying to move my airportitlwm 2.2 kext to the kexts folder in Efi - I did that succesfully. The next **_step was to open the .config list in EFI, so I can manually remove the airportitlwm running driver and place the " new 2.2 airportitlwm" kext._** **_So i started to drag it with the trackpad (pretty slowly tho, I'm a PC user) and accidentally dropped the .kext below the "IntelBlueToothFirmware, IntelBTPatcher and BlueToolFixup" and I was like "f**k it, not gonna waste more time, just reinstalling it anyways"..

After the restart I started to recieve messages and.. the "magic" happens! My wifi (and like a bonus bluetooth) started working!_** 

That was my "lucky story" guys, hope someone else is going to have that success rate like me! 

System product key: iMac18,1, using the latest opencore configurator, laptop machine name: Acer Aspire E 15 ( E5-575G-59P2)

[image]

 Have a wonderful day! 
Cheers!