OpenIntelWireless/itlwm #831: Issue with Intel Wi-Fi 6 AX201 2x2 "Build-in soldered into motherboard"

oronfel2911, 2022-11-28:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: AirportItlwm.kext 
- WiFi Card Model: "Build-in soldered in motherboard" Intel Wi-Fi 6 AX201 2x2 (Gig+) + Bluetooth 5.1  
- PCI Product ID: Device ID: 0x1911 - Subsystem Vendor ID: 0x8086 - Subsystem ID: 0x097e - revision ID: 0x0000
- macOS Version: Monterey 12.3.1

**Description**
I have Dell Precision 5550 Running Macos Monterey 12.3.1  with Intel Wi-Fi 6 AX201 2x2 (Gig+) + Bluetooth 5.1
I'm using AirportItlwm.kext - IntelBluetoothFirmware.kext - BlueToolFixup.kext with the wifi is s unstable Keeps disconnecting. 
The wifi icon shows on the top bar, i used to work for few hours then keep loading as its conecting then nothing happened, not working. 

Build-in soldered WIFI Card
https://imgur.com/ms8ckw1

My Dell Has no option to disable the "Fast Boot"
https://imgur.com/wzhf54D

**Bug Report Archive**
Read this guide if you are using AirportItlwm http://docs.oiw.workers.dev/itlwm/Troubleshooting.html#manually-for-airportitlwm

Report 
No variant specified, falling back to release
   85    0 0                  0xcf0000   0xcf0000   com.zxystd.AirportItlwm (2.0.0) 8D72B887-C419-3630-9D5A-1538E67DFD89 <84 49 16 7 6 3 1>

**Kext Download Source**
https://github.com/OpenIntelWireless/itlwm

oronfel2911, 2022-12-02:
Thank you @zxystd for trying top help me 

here is my log

 
————————————————————————————————————————————————————————————————————————————————————————————————————————
Timestamp    Name                          Duration    Result    Description                            
————————————————————————————————————————————————————————————————————————————————————————————————————————
Ping LAN                                   0.007       Yes       23:46:46.493        Could ping router (192.168.1.254 / 3.15ms / BE / 64 bytes / i=0.100000s / W=1.600000s t=1.600000s / c=1)
Ping LAN                                   0.007       Yes       23:46:46.500        Could ping router (192.168.1.254 / 2.98ms / BK / 65 bytes / i=0.100000s / W=1.600000s t=1.600000s / c=1)
Ping LAN                                   0.006       Yes       23:46:46.507        Could ping router (192.168.1.254 / 2.95ms / VI / 66 bytes / i=0.100000s / W=1.600000s t=1.600000s / c=1)
Ping LAN                                   0.007       Yes       23:46:46.514        Could ping router (192.168.1.254 / 2.85ms / VO / 67 bytes / i=0.100000s / W=1.600000s t=1.600000s / c=1)
Ping WAN                                   1.003       No        23:46:46.521        Could not ping DNS server (17.254.0.22 / 0.00ms / BE / 68 bytes / i=0.100000s / W=1.600000s / t=1.600000s / c=1)
Ping WAN                                   0.006       Yes       23:46:47.525        Could ping DNS server (192.168.1.254 / 3.03ms / BE / 69 bytes / i=0.100000s / W=1.600000s / t=1.600000s / c=1)
Ping LAN                                   0.518       Yes       23:46:47.532        Could ping router (192.168.1.254 / 2.92ms / BE / 70 bytes / i=0.100000s / W=0.200000s t=1.600000s / c=6)
Ping LAN                                   0.516       Yes       23:46:48.050        Could ping router (192.168.1.254 / 2.95ms / BK / 71 bytes / i=0.100000s / W=0.200000s t=1.600000s / c=6)
Ping LAN                                   0.515       Yes       23:46:48.567        Could ping router (192.168.1.254 / 2.58ms / VI / 72 bytes / i=0.100000s / W=0.200000s t=1.600000s / c=6)
Ping LAN                                   0.509       Yes       23:46:49.082        Could ping router (192.168.1.254 / 2.99ms / VO / 73 bytes / i=0.100000s / W=0.200000s t=1.600000s / c=6)
Ping6 AWDL                                 0.001       No        23:46:49.592        Could not ping AWDL ((null) / 0.00ms / BE / 64 bytes / w=0.000000s / c=0)
Resolve DNS                                0.012       Yes       23:46:49.594        Could resolve 'captive.apple.com'
Reach Apple                                0.000       Yes       23:46:49.606        Could reach 'www.apple.com'
Curl Apple                                 0.050       Yes       23:46:49.607        Could curl 'https://captive.apple.com'
Double NAT                                 0.057       No        23:46:49.657        Did not detect double NAT
————————————————————————————————————————————————————————————————————————————————————————————————————————
Timestamp    Name                          Duration    Result    Description                            
————————————————————————————————————————————————————————————————————————————————————————————————————————
Wi-Fi Link                                 0.001       Yes       23:46:49.728        Wi-Fi is associated to 'ATTkfv9QDS'
IPv4 Assigned                              0.001       Yes       23:46:49.729        Wi-Fi has IPv4 assigned (router=192.168.1.254 ip=192.168.1.133)
IPv6 Assigned                              0.001       Yes       23:46:49.730        Wi-Fi has IPv6 assigned (router=fe80::2e7e:81ff:fe93:d490 ip=2600:1700:2990:ab0:1420:5ecb:3566:8416)
Primary IPv4 is Wi-Fi                      0.003       Yes       23:46:49.732        en0 (Wi-Fi / 5EF824F4-0A67-451C-A0E9-D8D0B4B9723B)
Primary IPv6 is Wi-Fi                      0.003       Yes       23:46:49.735        en0 (Wi-Fi / 5EF824F4-0A67-451C-A0E9-D8D0B4B9723B)
Custom IPv4                                0.001       No        23:46:49.738        Wi-Fi is using 'DHCP' for IPv4
Custom IPv6                                0.001       No        23:46:49.739        Wi-Fi is using 'Automatic' for IPv6
Custom DNS                                 0.001       No        23:46:49.740        Did not detect custom DNS settings
Custom Web Proxy                           0.001       No        23:46:49.741        Did not detect custom web proxy settings
BT Paired Count                            0.043       Yes       23:46:49.742        Paired Bluetooth device count (1)
BT Connected Count                         0.000       No        23:46:49.785        Connected Bluetooth device count (0)
BT Coex Active                             0.000       No        23:46:49.786        Bluetooth coexistence is not active
AWDL Enabled                               0.000       Yes       23:46:49.786        AWDL is active
AWDL Real Time Mode                        0.000       No        23:46:49.787        AWDL real time mode is not active
Poor Wi-Fi Signal                          0.001       No        23:46:49.787        Current network has good signal (RSSI=-50dBm, SNR=19dB)
Uncommon Beacon Interval                   0.003       No        23:46:49.789        Beacon interval is 100ms
Uncommon DTIM Interval                     0.001       Yes       23:46:49.792        DTIM interval is 0ms
[... cut ...]
No Scan Results                            0.000       Yes       23:46:49.849        No Wi-Fi networks are visible
Conflicting PHY Mode 11ac                  0.001       No        23:46:49.850        0 PNL networks have conflicting PHY modes (802.11ac).
Conflicting PHY Mode 11n                   0.001       No        23:46:49.851        0 PNL networks have conflicting PHY modes (802.11n).
Conflicting Wi-Fi Sec
urity                 0.000       No        23:46:49.852        Did not find conflicting security types

here is my tmp folder if you would like to look at it
 https://ufile.io/70x99urf

oronfel2911, 2022-12-05:
Here is the log when I'm using itlwm.kwxt

[    1.638641]: itlwm: iwm_match devId=0x06C8
[    1.638660]: itlwm: free
[    1.638711]: itlwm: iwm_match devId=0x1901
[    1.638728]: itlwm: free
[    1.638751]: itlwm: iwm_match devId=0x06F9
[    1.638810]: itlwm: free
[    1.638847]: itlwm: iwm_match devId=0x06EF
[    1.638860]: itlwm: free
[    1.639206]: itlwm: iwm_match devId=0x06E0
[    1.639225]: itlwm: free
[    1.639264]: itlwm: iwm_match devId=0x06A4
[    1.639279]: itlwm: free
[    1.639343]: itlwm: iwm_match devId=0x1903
[    1.639349]: itlwm: free
[    1.639436]: itlwm: iwm_match devId=0x9BC4
[    1.639445]: itlwm: free
[    1.639473]: itlwm: iwm_match devId=0x1911
[    1.639479]: itlwm: free
[    1.639729]: itlwm: iwm_match devId=0x06B8
[    1.639745]: itlwm: free
[    1.639769]: itlwm: iwm_match devId=0x06A3
[    1.639770]: itlwm: iwm_match devId=0x06A8
[    1.639778]: itlwm: free
[    1.639792]: itlwm: free
[    1.639863]: itlwm: iwm_match devId=0x06BE
[    1.639871]: itlwm: free
[    1.639883]: itlwm: iwm_match devId=0x06E9
[    1.639893]: itlwm: free
[    1.640015]: itlwm: iwm_match devId=0x068E
[    1.640022]: itlwm: free
[    1.640140]: itlwm: iwm_match devId=0x06B0
[    1.640159]: itlwm: free
[    1.640175]: itlwm: iwm_match devId=0x06FC
[    1.640181]: itlwm: free
[    1.640219]: itlwm: iwm_match devId=0x06E8
[    1.640234]: itlwm: free
[    1.640647]: itlwm: iwm_match devId=0x06ED
[    1.640664]: itlwm: free
[    1.641174]: itlwm: iwm_match devId=0x15EA
[    1.641179]: itlwm: free
[    1.651752]: itlwm: iwm_match devId=0x15EA
[    1.651764]: itlwm: free
[    1.652039]: itlwm: iwm_match devId=0x15EA
[    1.652056]: itlwm: free
[    1.652226]: itlwm: iwm_match devId=0x15EA
[    1.652243]: itlwm: free
[    1.652350]: itlwm: iwm_match devId=0x15EA
[    1.652364]: itlwm: free
[    1.653880]: itlwm: iwm_match devId=0x15EB
[    1.653885]: itlwm: free
[    1.654822]: itlwm: iwm_match devId=0x15EC
[    1.654827]: itlwm: free
[    1.734690]: itlwm: Found device cfg: Intel(R) Wi-Fi 6 AX201 160MHz fw: iwlwifi-QuZ-a0-hr-b0-48.ucode index: 36:192
[    1.734696]: itlwm: Found PCI dev 06f0/4070, rev=0x354, rfid=0x10a100 fw: iwlwifi-QuZ-a0-hr-b0-48.ucode
[    1.761833]: itlwm: taskq taskq_create_thread lock
[    1.761835]: itlwm: taskq taskq_create_thread lock
[    1.761839]: itlwm: taskq taskq_create_thread unlock
[    1.761841]: itlwm: taskq taskq_create_thread unlock
[    1.761846]: itlwm: ieee80211_channel_init
[    1.761847]: itlwm: taskq taskq_create_thread lock
[    1.761849]: itlwm: taskq taskq_create_thread unlock
[    1.761850]: itlwm: taskq taskq_create_thread lock
[    1.761852]: itlwm: taskq taskq_create_thread unlock
[    1.761853]: itlwm: ieee80211_set_link_state nstate=2, old_state=-1
[    1.761855]: itlwm: ieee80211_set_link_state LINK_STATE_IS_DOWN
[    1.761856]: itlwm: taskq taskq_thread schedule task
[    1.761858]: itlwm: taskq taskq_thread schedule task
[    1.761861]: itlwm: ieee80211_media_init
[... 523 lines cut ...]
[   59.928201]: itlwm: iwx_cmd_done: command 0x1d done
[   59.928219]: itlwm: iwx_ba_task tx queue alloc succeed qid=4 err=0
[   63.555894]: itlwm: setMulticastList multicast count=7 bssid=26:7e:81:93:d4:93
[   63.555903]: itlwm: iwx_send_cmd: Sending command (1.d0), 56 bytes at [26]:0 ver: 0
[   63.555912]: itlwm: setMulticastList multicast count=8 bssid=26:7e:81:93:d4:93
[   63.555918]: itlwm: iwx_send_cmd: Sending command (1.d0), 60 bytes at [27]:0 ver: 0
[   63.555966]: itlwm: iwx_cmd_done: command 0xd0 done
[   63.556062]: itlwm: iwx_cmd_done: command 0xd0 done
[   63.556074]: itlwm: setMulticastList multicast count=9 bssid=26:7e:81:93:d4:93
[   63.556085]: itlwm: iwx_send_cmd: Sending command (1.d0), 68 bytes at [28]:0 ver: 0
[   63.556151]: itlwm: iwx_cmd_done: command 0xd0 done
[   63.865214]: itlwm: setMulticastList multicast count=10 bssid=26:7e:81:93:d4:93
[   63.865241]: itlwm: iwx_send_cmd: Sending command (1.d0), 72 bytes at [29]:0 ver: 0
[   63.865300]: itlwm: iwx_cmd_done: command 0xd0 done
[   65.047430]: Sandbox apply: mdworker_shared[577] <bytes>Sandbox apply: mdworker_shared[571] <bytes>Sandbox apply: mdworker_shared[578] <bytes>Sandbox apply: mdworker_shared[574] <bytes>Sandbox apply: mdworker_shared[572] <bytes>Sandbox apply: mdworker_shared[576] <bytes>Sandbox apply: mdworker_shared[573] <bytes>Sandbox apply: mdworker_shared[575] <bytes>Sandbox apply: mdworker_shared[579] <bytes>itlwm: iwx_rs_update new rate: 0x4c00f017: VHT | ANT: AB BW: 80Mhz MCS: 7 NSS: 2 SGI LDPC 
[   65.952562]: itlwm: setMulticastList multicast count=11 bssid=26:7e:81:93:d4:93
[   65.952575]: itlwm: iwx_send_cmd: Sending command (1.d0), 80 bytes at [30]:0 ver: 0
[   65.952661]: itlwm: iwx_cmd_done: command 0xd0 done
[   66.956098]: uipc_accept: peer disconnected unp_gencnt 354itlwm: iwx_add_task iwx_ba_task
[   67.070657]: itlwm: taskq task_add iwx_ba_task add to queue
[   67.070673]: itlwm: iwx_sta_rx_agg start=1 tid=5 ssn=1 winsize=64
[   67.070676]: itlwm: iwx_send_cmd_pdu_status
[   67.070685]: itlwm: iwx_send_cmd: Sending command (1.18), 48 bytes at [31]:0 ver: 0
[   67.070789]: itlwm: iwx_cmd_done: command 0x18 done
[   67.070797]: itlwm: : sending addba_resp to 26:7e:81:93:d4:93 on channel 149 mode 11ac

oronfel2911, 2022-12-08:
Yes Sir, same behavior

williambj1, 2023-01-07:
CC @zxystd

zxystd, 2023-02-09:
Have you checked the antennas?

closed 2023-06-02 by zxystd