OpenIntelWireless/itlwm #835: Slow Network Speed Intel 8260

amirgi73, 2022-12-07:
**Have You Read Our Docs**
YES

**Are You Reporting A Bug**
YES

**Environment**

- Kext Version: 2.2 alpha fedf42c
- WiFi Card Model: Intel 8260
- PCI Product ID: 0x8086 0x24F3
- macOS Version: Ventura 13.0.1

**Description**
Slow Network Speed compared With ethernet from the same Modem. I'm using a 2.5 GHz network as my Modem doesn't support 5GHz. 
Speed Test using Wifi: https://www.speedtest.net/result/14044564420
Speed Test using Ethernet: https://www.speedtest.net/result/14044550925
I've tried using Airportitlwm kext with no luck! (Wifi icon appears but it won't find any newtworks to connect to, and shows scanning...)

**Bug Report Archive**
[bugreport_29499.zip](https://github.com/OpenIntelWireless/itlwm/files/10180942/bugreport_29499.zip)

**Kext Download Source**
From this Github page.

johnpapad24, 2022-12-08:
I had an old router that worked pretty well with other pc's but in Mac OS (using itlwm/airportitlwm) the speed was about half for some reason.

If you have another router try it and check the speed.

usr-sse2, 2022-12-08:
@amirgi73 @johnpapad24 Have you tried disabling IPv6? (See https://github.com/OpenIntelWireless/itlwm/issues/805)

amirgi73, 2022-12-08:
> If you have another router try it and check the speed.

It's the same with another Router! My primary is TL-MR6400 with OpenWrt and the secound one is a Huawei TD-LTE with stock firmware...

amirgi73, 2022-12-09:
> @amirgi73 @johnpapad24 Have you tried disabling IPv6? (See #805)

disabling IPv6 won't help. (setting ipv6 to local link only in Mac os settings and/or disabling ipv6 completly in my reouter)

johnpapad24, 2022-12-10:
@usr-sse2 testing was done using Mac OS Catalina back then.

williambj1, 2023-01-07:
Your Wi-Fi is running in 802.11n 2.4GHz mode, so that might be the expected result. Does it differ a lot when compared with other OSes?

amirgi73, 2023-01-16:
> Your Wi-Fi is running in 802.11n 2.4GHz mode, so that might be the expected result. Does it differ a lot when compared with other OSes?

Yes it does differ alot! The speed 1/3 or even 1/4 compared with linux or windows
Also I have tried the latest Stable release (2.1) but it's the same

pryid, 2023-01-16:
-novht: Disables 802.11AC support
-noht40: Disables 40MHz when using 2.4GHz (Use this option if the network with this config causes instabilities. BRCM cards disable 2.4GHz HT40 by default)

ntvsp, 2023-02-17:
same isssue

cyrushaze, 2024-03-15:
I have the exact same problem , in windows/ Linux the speed is 150 mbs WiFi 600-700 mbs Ethernet , in Ventura 16-17 mbps WiFi max 100 mbps Ethernet

zxystd, 2024-03-16:
@cyrushaze try this version to see if it can improve wifi performance?
[Ventura.zip](https://github.com/OpenIntelWireless/itlwm/files/14622180/Ventura.zip)

cyrushaze, 2024-03-17:
I try it , an are result is the same, some times great speed an some times dog shite ,its driving me bananas 
[image]

felixguilherme, 2024-08-07:
I thought I had problems with the kext, tried -novht, -noht40, and disabling ipv6. No luck. Found out it was a DNS problem, changed it to 8.8.8.8 and now it works fine. If you're experiencing network slowdowns, check your DNS first.

husnainalam888, 2024-09-14:
Issue with AirportTwlm Kext: Slow Internet Speeds

Hi,

I've been using the AirportTwlm kext on my Intel-based Hackintosh for the WiFi card, but I've noticed that the internet speeds are significantly slower compared to what I expect. After troubleshooting, I switched to using itwlm with HeliPort, and the performance has improved dramatically—it works perfectly for me now.

I would recommend others facing similar speed issues to try this combination. Perhaps there could be an improvement or alternative setup with AirportTwlm that addresses this speed issue?

Thanks for all the hard work on this project!

Best,
Husnain Alam

ChainedTears, 2024-11-04:
Okay I know this is an old issue but I'm using a 8260 too - I've tried enabling the ``-novht --noht40`` flags, disabling ipv6, and trying itlwm + helioport instead of airportitlwm; but my speeds are still stuck at less than 0.5mbps making everything basically unusable. I've tried v2.3, v2.2, v2.1 all with the same results. I even tried setting my WiFi to wpa2 instead of wpa3 with no results. I'm using a HP EliteDesk 800 G2 on MacOS Monterey.

If anyone finds out how to fix this issue another way please let me know. Thanks for the hard work that went into this project!

vanillaogsky, 2024-11-17:
> Okay I know this is an old issue but I'm using a 8260 too - I've tried enabling the `-novht --noht40` flags, disabling ipv6, and trying itlwm + helioport instead of airportitlwm; but my speeds are [...]

you can fix with apple airport network card