OpenIntelWireless/itlwm #836: "Turn on wifi" doesn't work after using ax210 in hackintosh ventura, Windows 11

perr0r, 2022-12-14:
**Have You Read Our Docs**

No

**Are You Reporting A Bug**

No

**Environment**

- Kext Version: itlwm v2.2.0 DEBUG alpha
- WiFi Card Model: Intel AX210
- PCI Product ID: AX210
- macOS Version: Ventura (13.0)

**Description**

My wifi card ax210 doesn't work in any OS.
In Windows device manager, my wifi card activate correctly. But when I try to turn on wifi, wifi button become disabled.

**Bug Report Archive**

**Kext Download Source**

Boos4721, 2022-12-14:
what motherboard u using?

perr0r, 2022-12-14:
> what motherboard u using?

I don't know certain name but my laptop is DELL inspiron 3576

superbogie, 2022-12-15:
Disable WiFi in System Settings. If you don't have Heliport.app, DL it from the OpenIntelWireless/Heliport repo, place it in the Application folder, and then run it. You should now have a new "WiFi" icon in your Menu Bar that will have a list of available WiFi networks..

ShroomKing, 2022-12-16:
> My wifi card ax210 doesn't work in any OS.

Very likely broken wifi module

I put an Intel AX210 in my Dell Latitude 7480 and it's working perfectly in Linux, Windows and MacOS

closed 2023-01-07 by williambj1