OpenIntelWireless/itlwm #838: WiFi 6E not available in macOS Ventura while using compatible device (AX210)

superbogie, 2022-12-15:
PCIe card with AX210 chipset has fully functional WiFi 6E under Windows 11 but has no option/setting for such under macOS Ventura when using itwlm.kext and Heliport.app (although 160MHz channel width IS enabled. Also, Airportitwlm.kext has yet to receive a Ventura compatible update so is currently not an option in this regard). If this is a bug, please fix. Ifit is a feature, please elaborate.

williambj1, 2022-12-16:
> PCIe card with AX210 chipset has fully functional WiFi 6E under Windows 11 but has no option/setting for such under macOS Ventura when using itlwm.kext and Heliport.app

Currently not implemented, feel free to file pull requests and improve this project.

> Also, Airportitlwm.kext has yet to receive a Ventura-compatible update so is currently not an option in this regard).

Not sure what your expectations for "a Ventura-compatible update" is, there is a functional version provided in the release page with known limitations documented in other issues.

I will close this issue since this is not a bug report. Hopefully this answers all your questions.

closed 2022-12-16 by williambj1

superbogie, 2022-12-17:
1. Thanks for replying but I don't have any software/coding experience so I'm not able to contribute by making a pull request. I'm merely puzzled as to why 802.11ax has been intentionally disabled in PCIe WiFi adapters with the AX210 chipset - especially now that this feature is available and sought-after in ever more devices (well, at least until 802.11be-capable devices reach critical mass sometime next year) In a perfect (coding) world, this would just be a toggle/bit to turn on but that is probably far from the truth (shrug).
2. I'm also at a loss as to why you suggest I use a functional version of Airportitlwm.kext for Ventura when there is none. The only stable release versions available are for Big Sur, Catalina, High Sierra, Mojave, and Monterey. Itlwm.kext is the only current option for Ventura as trying to use any of the Airportitlwm versions previously mentioned will immediately crash OpenCore at boot.