OpenIntelWireless/itlwm #840: Wifi issue Intel 3165, Intel 7260, Intel CNVi WIFI

Demontager, 2022-12-22:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: v2.1.0 stable
- WiFi Card Model: Intel 3165, Intel 7260, Intel CNVi WIFI
- PCI Product ID: 0x8086, 0x34

**Description**
After upgrading to Ventura (OCR 0.8.7) I found minor problem which affected all three wifi devices above. Moreover they are installed in different motherboards (2xminipc, 1xdesktop-pc)
 All three wifi devices causing immediate OSX reboot after i do connecting to WIFI AP, also i don't see wifi ESSID list in menubar, i need to open Wifi settings to reveal them.
p.s. I had no wifi issues on Monterey. No using any wifi tweaks in config.plist

**Bug Report Archive**
I can't gather report because when i connect to any access point system reboots.
[image]
  

**Kext Download Source**
Orinal github vanila source

williambj1, 2022-12-22:
The Monterey version is not compatible for Ventura.

closed 2022-12-22 by williambj1

Demontager, 2022-12-22:
Okay understood. I also tried kext from  AirportItlwm-Ventura-v2.2.0-DEBUG-alpha-acc7ff9.zip (OCR 0.8.7 Release)  it does not cause a reboot, but can't connect to any wifi AP.

trongtinh1212, 2022-12-23:
> Okay understood. I also tried kext from AirportItlwm-Ventura-v2.2.0-DEBUG-alpha-acc7ff9.zip (OCR 0.8.7 Release) it does not cause a reboot, but can't connect to any wifi AP.

change to itlw and HeliPort , i think it will fix in the next update