OpenIntelWireless/itlwm #841: macOS 13 (Ventura) support for AirportItlwm

Laith-GLaDOS, 2022-12-27:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
No

**Environment**

- Kext Version: v2.1.0-stable
- WiFi Card Model: Intel (I don't know what model but AirportItlwm works fine for my WiFi card in Catalina)
- PCI Product ID: I don't know
- macOS Version: 13 (Ventura)

**Description**
As far as I know, v2.1.0-stable does not have support for Ventura, but it works with Monterey
v2.2.0-alpha works with Ventura but I have tried it and it was unstable
I read a comment on an issue regarding this version of macOS and support for it from AirportItlwm that said to use Monterey version of AirportItlwm and to disable RSN so it won't kernel panic, but, I don't know where that option is, and I've searched for it everywhere in OCAT (OpenCore Auxiliary Tool, a config.plist configurator)
Here is the comment: https://github.com/OpenIntelWireless/itlwm/issues/784#issuecomment-1152903476
Any help on getting v2.1.0-stable to work on Ventura is greatly appreciated

**Bug Report Archive**

**Kext Download Source**
https://github.com/OpenIntelWireless/itlwm/releases/download/v2.1.0/AirportItlwm_v2.1.0_stable_Monterey.kext.zip

williambj1, 2022-12-27:
Duplicate of #823

williambj1, 2022-12-27:
Disabling Apple RSN will not fix your "not working", "unstable" issue. If you disagree with this statement and you don't believe this is a duplicate please provide more information, thanks.

closed 2022-12-27 by williambj1

Laith-GLaDOS, 2022-12-28:
Okay, thanks anyways.

Vandoka1972, 2023-04-25:
Consegui - agora ta show no Ventura 13.3.1 coloca ip manualmente ai so corre pro abraço