OpenIntelWireless/itlwm #848: [AirportItlwm] Crash after enter wifi password

vacnex, 2023-01-12:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: (AirportItlwm: 2.1.0)
- WiFi Card Model: Intel AX201 Wi-Fi 
- PCI Product ID: 0x8086
- macOS Version: 13.1

**Description**
My hack got immediately crash after enter wifi password
**Bug Report Archive**
```
panic(cpu 12 caller 0xffffff801e748953): Kernel trap at 0xffffff801e301057, type 14=page fault, registers:
CR0: 0x0000000080010033, CR2: 0x0000000000000000, CR3: 0x00000008878bd03d, CR4: 0x00000000003626e0
RAX: 0x0000000000000000, RBX: 0xffffff9065c2d800, RCX: 0x0000000000000109, RDX: 0x0000000000000109
RSP: 0xffffffa0e71a75e8, RBP: 0xffffffa0e71a7620, RSI: 0xffffff906772dd88, RDI: 0x0000000000000000
R8:  0x0000000000000109, R9:  0x0000000000000001, R10: 0xffffff9065c3f000, R11: 0xffffff8b98e26c80
R12: 0xffffff9065c3f000, R13: 0x0000000000000000, R14: 0xffffffa0e71a7b38, R15: 0x0000000000000000
RFL: 0x0000000000010286, RIP: 0xffffff801e301057, CS:  0x0000000000000008, SS:  0x0000000000000010
Fault CR2: 0x0000000000000000, Error code: 0x0000000000000002, Fault CPU: 0xc, PL: 0, VF: 0

Panicked task 0xffffff9065612620: 23 threads: pid 231: airportd
Backtrace (CPU 12), panicked thread: 0xffffff9066bd6b30, Frame : Return Address
0xffffffa0e71a6fc0 : 0xffffff801e5eb31d 
0xffffffa0e71a7010 : 0xffffff801e759236 
0xffffffa0e71a7050 : 0xffffff801e748480 
0xffffffa0e71a70a0 : 0xffffff801e585951 
0xffffffa0e71a70c0 : 0xffffff801e5eb5fd 
0xffffffa0e71a71b0 : 0xffffff801e5eaca9 
0xffffffa0e71a7210 : 0xffffff801ede08cb 
0xffffffa0e71a7300 : 0xffffff801e748953 
0xffffffa0e71a7480 : 0xffffff801e748642 
0xffffffa0e71a74d0 : 0xffffff801e585951 
0xffffffa0e71a74f0 : 0xffffff801e301057 
0xffffffa0e71a7620 : 0xffffff802287a765 
0xffffffa0e71a76a0 : 0xffffff80202bc3c0 
0xffffffa0e71a7af0 : 0xffffff80202d3cfe 
0xffffffa0e71a7b20 : 0xffffff80202d7817 
0xffffffa0e71a7b80 : 0xffffff8020da9ef2 
0xffffffa0e71a7bb0 : 0xffffff801ed15d54 
0xffffffa0e71a7c10 : 0xffffff8020da9f6a 
0xffffffa0e71a7c70 : 0xffffff80202d76d5 
0xffffffa0e71a7ca0 : 0xffffff801e851c87 
0xffffffa0e71a7d00 : 0xffffff801e848b4e 
0xffffffa0e71a7db0 : 0xffffff801e84f0af 
0xffffffa0e71a7de0 : 0xffffff801eb51435 
0xffffffa0e71a7e40 : 0xffffff801eb4c11e 
0xffffffa0e71a7f40 : 0xffffff801ec5588b 
0xffffffa0e71a7fa0 : 0xffffff801e585d96 
      Kernel Extensions in backtrace:
         com.apple.iokit.IONetworkingFamily(3.4)[196DFD8C-1D79-39E6-AEFA-FCF645C51977]@0xffffff8020d9f000->0xffffff8020db5fff
         com.apple.iokit.IO80211FamilyLegacy(1200.12.2b1)[63565270-42B3-31A8-80FB-16DA0EBBFA4C]@0xffffff8020240000->0xffffff8020386fff
            dependency: com.apple.driver.AppleMobileFileIntegrity(1.0.5)[4E741445-0025-3DB8-8292-C7533F3C0917]@0xffffff801fb92000->0xffffff801fbc0fff
            dependency: com.apple.driver.corecapture(1.0.4)[3A3001CC-D289-35D6-B101-7B16A7F8AC9E]@0xffffff80217f0000->0xffffff8021812fff
            dependency: com.apple.iokit.CoreAnalyticsFamily(1)[D0FF0186-6F90-3AF1-9F7E-A2BD383CCA20]@0xffffff801fedc000->0xffffff801fee5fff
            dependency: com.apple.iokit.IONetworkingFamily(3.4)[196DFD8C-1D79-39E6-AEFA-FCF645C51977]@0xffffff8020d9f000->0xffffff8020db5fff
            dependency: com.apple.iokit.IOSkywalkFamily(1.0)[F96DD1E4-1285-31A2-8AE8-4C6884D59E11]@0xffffff80210e9000->0xffffff8021116fff
            dependency: com.apple.kec.corecrypto(12.0)[7A2A4035-FDBB-38A2-88B7-7AF62A0BD48B]@0xffffff8021821000->0xffffff8021899fff
         com.zxystd.AirportItlwm(2.1)[FA02D059-D593-3D8E-A031-1F99EA4D67EB]@0xffffff80227c3000->0xffffff802375cfff
            dependency: com.apple.iokit.IO80211FamilyLegacy(1200.12.2b1)[63565270-42B3-31A8-80FB-16DA0EBBFA4C]@0xffffff8020240000->0xffffff8020386fff
            dependency: com.apple.iokit.IONetworkingFamily(3.4)[196DFD8C-1D79-39E6-AEFA-FCF645C51977]@0xffffff8020d9f000->0xffffff8020db5fff
            dependency: com.apple.iokit.IOPCIFamily(2.9)[752CD69F-309E-31FF-881C-2A22ACB6C0B6]@0xffffff8021000000->0xffffff802102ffff

Process name corresponding to current thread (0xffffff9066bd6b30): airportd
Boot args: -wegnoigpu agdpmod=pikera -ctrsmt 

Mac OS version:
22C65

Kernel version:
Darwin Kernel Version 22.2.0: Fri Nov 11 02:08:47 PST 2022; root:xnu-8792.61.2~4/RELEASE_X86_64
Kernel UUID: 6B3374E9-7FD7-354D-886F-E995FAAC2CA3
roots installed: 0
KernelCache slide: 0x000000001e200000
KernelCache base:  0xffffff801e400000
Kernel slide:      0x000000001e2dc000
Kernel text base:  0xffffff801e4dc000
__HIB  text base: 0xffffff801e300000
System model name: MacPro7,1 (Mac-27AD2F918AE68F61)
System shutdown begun: NO
Panic diags file available: YES (0x0)
Hibernation exit count: 0

System uptime in nanoseconds: 476716977927
Last Sleep:           absolute           base_tsc          base_nano
  Uptime  : 0x0000006efe8bd145
  Sleep   : 0x0000000000000000 0x0000000000000000 0x0000000000000000
  Wake    : 0x0000000000000000 0x0000000c93f9aba9 0x0000000000000000
Compressor Info: 0% of compressed pages limit (OK) and 0% of segments limit (OK) with 0 swapfiles and OK swap space
Zone info:
  Zone map: 0xffffff8064007000 - 0xffffffa064007000
  . PGZ   : 0xffffff8064007000 - 0xffffff8068008000
  . VM    : 0xffffff8068008000 - 0xffffff853433b000
  . RO    : 0xffffff853433b000 - 0xffffff86cd9a1000
  . GEN0  : 0xffffff86cd9a1000 - 0xffffff8b99cd4000
  . GEN1  : 0xffffff8b99cd4000 - 0xffffff9066007000
  . GEN2  : 0xffffff9066007000 - 0xffffff953233a000
  . GEN3  : 0xffffff953233a000 - 0xffffff99fe66d000
  . DATA  : 0xffffff99fe66d000 - 0xffffffa064007000
  Metadata: 0xffffffab5d8f2000 - 0xffffffab7d8f2000
  Bitmaps : 0xffffffab7d8f2000 - 0xffffffab898f2000
```

**Kext Download Source**
(https://github.com/OpenIntelWireless/itlwm/releases/download/v2.1.0/itlwm_v2.1.0_stable.kext.zip)

williambj1, 2023-01-12:
You shouldn't be using the Monterey version on Ventura.

closed 2023-01-12 by williambj1

RobTS, 2023-04-22:
Any info on a 2.2.0 stable release? This is why some people might choose 2.1.0 Monterey kexts and fail...