OpenIntelWireless/itlwm #850: Killer(R) Wi-Fi 6 AX1650s 160MHz unstable 5GHz performance

blazer221, 2023-01-20:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: Airportitwlm 2.2.0 (latest, but also present on 2.1.0 Stable for Monterey).
- WiFi Card Model: Killer(R) Wi-Fi 6 AX1650s 160MHz Wireless Network Adapter (201D2W)
- PCI Product ID: 0x1651 (at least the System Information reports that)
- macOS Version: macOS 13.1

**Description**
I'm having constant problems with my laptop and the abovementioned WiFi card since day 1. Each time i try to connect to a 5GHz connection i'm having issues, most of the times i cannot connect at all, sometimes i'm connecting to such networks, but very unstable and usually get disconnected shortly. This is an issue with both Airportitlwm and itlwm kexts (with the second one actually i have more success, but again it's very unstable). I'm having this issue since Monterey and now on Ventura. The strange is that i have almost the same configuration, but with AX1650w and it works like a charm, only this one is getting me sick. I've tried many different configurations on my home router also, but it's not only an issue with it, it's with any 5GHz connection (ac/ax, it doesn't matter, have tried all combiations). If i'm connecting to a 2.4GHz connection it works flawlessly and fast. I have checked any documentation, searched in the issues, but didn't find anything useful that can help me. I'm also attaching a log with Airportitlwm used currently. I would be happy if there's a solution for all of this. 

**Bug Report Archive**

[itlwm_logs - 20012023.log](https://github.com/OpenIntelWireless/itlwm/files/10469987/itlwm_logs.-.20012023.log)

**Kext Download Source**
Latest alpha from Github - [AirportItlwm-Ventura-v2.2.0-DEBUG-alpha-e0f745e.zip](https://github.com/OpenIntelWireless/itlwm/releases/download/v2.2.0-alpha/AirportItlwm-Ventura-v2.2.0-DEBUG-alpha-e0f745e.zip)

ameenjuz, 2023-05-29:
I have same issue on 5GHz on Wi-Fi 6 AX201
Each time i try to connect to a 5GHz connection i'm having issues, most of the times i cannot connect at all, sometimes i'm connecting to such networks, but very unstable and usually get disconnected shortly

blazer221, 2023-06-26:
> 

I'm also with XPS9500, i've tried almost anything that was possible, but even the latest version, which tells there's support added for 1650s nothing has changed much.

SiddharthManthan, 2023-11-24:
I have the same card and it works well on sonoma.