OpenIntelWireless/itlwm #851: Password rejected - Ventura - Intel Dual Band Wireless-AC 9560

Enjoy-Combi, 2023-01-31:
Hello,

It was working some weeks ago and a day password keeps rejected. I put the wireless key but after a while, Ventura asks again the key.
I was thinking the card had an issue but it is working on windows 10.
I removed all network configurations as per troubleshooting section, same issue.
I was thinking I had a file corrupted then I reinstall Ventura, same issue.

I attach the log file. Let me know if more data are needed, I am lost regarding this issue.
Many thanks
[Log_2023-01-28_20-46-54.log](https://github.com/OpenIntelWireless/itlwm/files/10542899/Log_2023-01-28_20-46-54.log)

thymidin, 2023-02-01:
It happens on WPA3-only networks maybe it is your case ?

Enjoy-Combi, 2023-02-01:
The wireless encryption is WPA/WPA (PSK) TKUP & AES. 
Do you recommend me to change it ? It was same settings & working in the past. And a day it has stopped working without action on my side on that settings. What I am not sure, if the internet provider has updated the firmware of the router. I will investigate this

zxystd, 2023-02-09:
Try to follow the document to reset the network interface.

Enjoy-Combi, 2023-02-09:
Hello,

This is the first I did, before a full system reinstallation. No
improvement.

Enjoy-Combi, 2023-02-09:
Hello,

This is the first I did, before a full system reinstallation. No improvement.
Many thanks

1alessandro1, 2023-05-07:
Hello, with WPA3 SAE networks I have the same issue, by reverting back to WPA2 compatibility everything works.

9560NGW, Vendor 8086, PCI ID 34F0

zxystd, 2023-06-11:
it is most likely a router setting problem, itlwm doesn't support all of the encryption combination, WPA2 personal, WPA/WPA2 MIXED are confirmed to work.

closed 2023-06-11 by zxystd