OpenIntelWireless/itlwm #854: ax210从win重启进入mac提示itlwm未运行

MrWater233, 2023-02-09:
**Have You Read Our Docs**
YES

**Are You Reporting A Bug**
YES

**Environment**

- Kext Version: airportitlwm V2.2.0-alpha
- WiFi Card Model: AX210
- PCI Product ID: 
- macOS Version: 13.2

**Description**

使用itlwm：从windows重启或者关机后再进入mac系统，HeliPort会提示itlwm未运行
使用airportitlwm：从windows重启或关机后再进入mac系统，wifi会打不开
**需要在mac系统中再重启一次才能正常使用wifi**

**Bug Report Archive**

使用airportitlwm.kext并使用HeliPort生成了一份错误报告，辛苦大佬看一下
[bugreport_59330.zip](https://github.com/OpenIntelWireless/itlwm/files/10698795/bugreport_59330.zip)

**Kext Download Source**

https://github.com/OpenIntelWireless/itlwm/releases/download/v2.2.0-alpha/AirportItlwm-Ventura-v2.2.0-DEBUG-alpha-ddd2768.zip

MrWater233, 2023-02-09:
windows和mac在一块硬盘中，windows和bios都关闭了快速启动

zxystd, 2023-02-15:
不知道什么问题，但是看起来不是这个驱动的问题。

closed 2023-02-15 by zxystd