OpenIntelWireless/itlwm #861: 小米笔记本PRO TM1701  KBL ， OC，AirportItlwm_Ventura.kext 加载慢 (Intel Wireless 8265)

xjh169, 2023-03-14:
Ventura 开机加载 Wi-Fi 需要一分钟以上才能连接上网络，导致状态栏其他图标延迟显示。
如果禁用AirportItlwm_Ventura.kext ，开机进入桌面就直接看到 状态栏 的所有图标 那就不用等待加载了
测试使用Big_Sur、Catalina、Monterey版本 Wi-Fi加载速度没有异常。可能是Ventura的问题
希望AirportItlwm_Ventura.kext 加载速度可以得到优化！
[image]
[image]

w43322, 2023-04-02:
是因为debug build的原因吧，性能肯定赶不上release build

biless, 2023-04-12:
似乎不是 我这边编译了一个 release build速度上感觉没有快很多

xialiu666i, 2023-04-23:
一样

zxystd, 2023-05-26:
https://github.com/OpenIntelWireless/itlwm/issues/823#issuecomment-1564129817

panmili, 2023-05-30:
> Ventura 开机加载 Wi-Fi 需要一分钟以上才能连接上网络，导致状态栏其他图标延迟显示。 如果禁用AirportItlwm_Ventura.kext ，开机进入桌面就直接看到 状态栏 的所有图标 那就不用等待加载了 测试使用Big_Sur、Catalina、Monterey版本 Wi-Fi加载速度没有异常。可能是Ventura的问题 希望AirportItlwm_Ventura.kext 加载速度可以得到优化！ [image] [image]

最新版本的已经可以了

closed 2023-06-02 by zxystd