OpenIntelWireless/itlwm #866: When I connect to the Wifi network the notebook restarts

rodrigoselau, 2023-03-30:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: 2.1.0
- WiFi Card Model: Intel AC 9560 NGW
- PCI Product ID: G86C0007S810
- macOS Version: Ventura

**Description**
When I connect to the Wifi network the notebook restarts

**Bug Report Archive**
`panic(cpu 4 caller 0xffffff800c5485f3): Kernel trap at 0xffffff800c101057, type 13=general protection, registers:
CR0: 0x000000008001003b, CR2: 0x0000700007f692f8, CR3: 0x0000000024405000, CR4: 0x00000000003626e0
RAX: 0xaaaaaaaaaaaaaaaa, RBX: 0xffffffebf6b9b610, RCX: 0x00000000000000d4, RDX: 0x00000000000000d4
RSP: 0xffffffebf6b9b528, RBP: 0xffffffebf6b9b560, RSI: 0xffffff91761fb508, RDI: 0xaaaaaaaaaaaaaaaa
R8:  0x00000000000000d4, R9:  0x00000000ffffffff, R10: 0x0000000000000001, R11: 0x0000000000000002
R12: 0x0000000000000002, R13: 0x0000000000000030, R14: 0xffffff9b0de07a20, R15: 0xffffff9b0de07a20
RFL: 0x0000000000010286, RIP: 0xffffff800c101057, CS:  0x0000000000000008, SS:  0x0000000000000010
Fault CR2: 0x0000700007f692f8, Error code: 0x0000000000000000, Fault CPU: 0x4, PL: 0, VF: 0

Panicked task 0xffffff917574f208: 184 threads: pid 0: kernel_task
Backtrace (CPU 4), panicked thread: 0xffffff87dc935b30, Frame : Return Address
0xffffff800c10e1a0 : 0xffffff800c3eb38d mach_kernel : _handle_debugger_trap + 0x4ad
0xffffff800c10e1f0 : 0xffffff800c558ed6 mach_kernel : _kdp_i386_trap + 0x116
0xffffff800c10e230 : 0xffffff800c548120 mach_kernel : _kernel_trap + 0x3e0
0xffffff800c10e280 : 0xffffff800c385951 mach_kernel : _return_from_trap + 0xc1
0xffffff800c10e2a0 : 0xffffff800c3eb66d mach_kernel : _DebuggerTrapWithState + 0x5d
0xffffff800c10e390 : 0xffffff800c3ead19 mach_kernel : _panic_trap_to_debugger + 0x1a9
0xffffff800c10e3f0 : 0xffffff800cbe072b mach_kernel : _panic + 0x84
0xffffff800c10e4e0 : 0xffffff800c5485f3 mach_kernel : _sync_iss_to_iks + 0x2c3
0xffffff800c10e660 : 0xffffff800c5482e2 mach_kernel : _kernel_trap + 0x5a2
0xffffff800c10e6b0 : 0xffffff800c385951 mach_kernel : _return_from_trap + 0xc1
0xffffff800c10e6d0 : 0xffffff800c101057 
0xffffffebf6b9b560 : 0xffffff801072c765 com.zxystd.AirportItlwm : __ZN12AirportItlwm17apple80211RequestEjiP16IO80211InterfacePv + 0x835
0xffffffebf6b9b5e0 : 0xffffff800e16911d com.apple.iokit.IO80211FamilyLegacy : __ZN13RSNSupplicant14storeFullRSNIEEP16RSNAuthenticator + 0x77
0xffffffebf6b9ba40 : 0xffffff800e167627 com.apple.iokit.IO80211FamilyLegacy : __ZN13RSNSupplicant11process1of4EP16RSNAuthenticatorP6__mbufP7wpa_keyt + 0x1c5
0xffffffebf6b9bbf0 : 0xffffff800e135261 com.apple.iokit.IO80211FamilyLegacy : __ZN16IO80211Interface11inputPacketEP6__mbufjjPv + 0x369
0xffffffebf6b9bc60 : 0xffffff801067e3b0 com.zxystd.AirportItlwm : __ZN21AirportItlwmInterface11inputPacketEP6__mbufjjPv + 0x110
0xffffffebf6b9bce0 : 0xffffff8010675f7b com.zxystd.AirportItlwm : __ZL9_if_inputP8OSObjectPvS1_S1_S1_ + 0xdb
0xffffffebf6b9bd50 : 0xffffff800cb15aa4 mach_kernel : __ZN13IOCommandGate9runActionEPFiP8OSObjectPvS2_S2_S2_ES2_S2_S2_S2_ + 0xb4
0xffffffebf6b9bdb0 : 0xffffff8010675e96 com.zxystd.AirportItlwm : __Z8if_inputP6_ifnetP9mbuf_list + 0x66
0xffffffebf6b9bdf0 : 0xffffff80106f377a com.zxystd.AirportItlwm : __ZN6ItlIwx14iwx_notif_intrEP9iwx_softc + 0x15a
0xffffffebf6b9be50 : 0xffffff80106f3e12 com.zxystd.AirportItlwm : __ZN6ItlIwx8iwx_intrEP8OSObjectP22IOInterruptEventSourcei + 0x632
0xffffffebf6b9bed0 : 0xffffff800cb144b4 mach_kernel : __ZN22IOInterruptEventSource12checkForWorkEv + 0x114
0xffffffebf6b9bf20 : 0xffffff800cb12dae mach_kernel : __ZN10IOWorkLoop15runEventSourcesEv + 0x12e
0xffffffebf6b9bf60 : 0xffffff800cb123e7 mach_kernel : __ZN10IOWorkLoop10threadMainEv + 0x37
0xffffffebf6b9bfa0 : 0xffffff800c38519e mach_kernel : _call_continuation + 0x2e
      Kernel Extensions in backtrace:
         com.apple.iokit.IO80211FamilyLegacy(1200.12.2b1)[8CB6AB86-0DC8-324C-A9F7-B46C452522DD]@0xffffff800e098000->0xffffff800e1defff
            dependency: com.apple.driver.AppleMobileFileIntegrity(1.0.5)[DB38BB35-D38C-3110-9B53-0D2927F702CF]@0xffffff800d9c5000->0xffffff800d9f3fff
            dependency: com.apple.driver.corecapture(1.0.4)[1EFC00FC-8337-39AF-B061-EDA1CB53C6E7]@0xffffff800f64a000->0xffffff800f66cfff
            dependency: com.apple.iokit.CoreAnalyticsFamily(1)[BF4BE411-8D61-3123-AEA0-88671FB17B81]@0xffffff800dd26000->0xffffff800dd2ffff
            dependency: com.apple.iokit.IONetworkingFamily(3.4)[ACD4449A-4F90-31DB-82AD-2F2B83D8F6E3]@0xffffff800ebf8000->0xffffff800ec0efff
            dependency: com.apple.iokit.IOSkywalkFamily(1.0)[337AFA35-6288-3EA9-BC6E-AA8E075EECF6]@0xffffff800ef42000->0xffffff800ef6ffff
            dependency: com.apple.kec.corecrypto(12.0)[CADEE2DB-DF62-3630-A9F9-BC8B75D43579]@0xffffff800f67b000->0xffffff800f6f3fff
         com.zxystd.AirportItlwm(2.1)[FA02D059-D593-3D8E-A031-1F99EA4D67EB]@0xffffff8010675000->0xffffff801160efff
            dependency: com.apple.iokit.IO80211FamilyLegacy(1200.12.2b1)[8CB6AB86-0DC8-324C-A9F7-B46C452522DD]@0xffffff800e098000->0xffffff800e1defff
            dependency: com.apple.iokit.IONetworkingFamily(3.4)[ACD4449A-4F90-31DB-82AD-2F2B83D8F6E3]@0xffffff800ebf8000->0xffffff800ec0efff
            dependency: com.apple.iokit.IOPCIFamily(2.9)[802B9CA4-B851-396E-8DA4-B30E751EC48F]@0xffffff800ee59000->0xffffff800ee88fff

Process name corresponding to current thread (0xffffff87dc935b30): kernel_task
Boot args: -v igfxonln=1 igfxagdc=0 keepsyms=1 debug=0x100

Mac OS version:
22D68

Kernel version:
Darwin Kernel Version 22.3.0: Mon Jan 30 20:42:11 PST 2023; root:xnu-8792.81.3~2/RELEASE_X86_64
Kernel UUID: 10E5D254-4A37-3A2A-B560-E6956A093ADE
roots installed: 0
KernelCache slide: 0x000000000c000000
KernelCache base:  0xffffff800c200000
Kernel slide:      0x000000000c0dc000
Kernel text base:  0xffffff800c2dc000
__HIB  text base: 0xffffff800c100000
System model name: MacBookPro16,1 (Mac-E1008331FDC96864)
System shutdown begun: NO
Panic diags file available: YES (0x0)
Hibernation exit count: 0

System uptime in nanoseconds: 1123325237046
Last Sleep:           absolute           base_tsc          base_nano
  Uptime  : 0x000001058b66dac7
  Sleep   : 0x0000000000000000 0x0000000000000000 0x0000000000000000
  Wake    : 0x0000000000000000 0x00000003d1d4999e 0x0000000000000000
Compressor Info: 0% of compressed pages limit (OK) and 0% of segments limit (OK) with 0 swapfiles and OK swap space
Zone info:
  Zone map: 0xffffff817502b000 - 0xffffffa17502b000
  . PGZ   : 0xffffff817502b000 - 0xffffff817602c000
  . VM    : 0xffffff817602c000 - 0xffffff8642a92000
  . RO    : 0xffffff8642a92000 - 0xffffff87dc35e000
  . GEN0  : 0xffffff87dc35e000 - 0xffffff8ca8dc4000
  . GEN1  : 0xffffff8ca8dc4000 - 0xffffff917582a000
  . GEN2  : 0xffffff917582a000 - 0xffffff9642290000
  . GEN3  : 0xffffff9642290000 - 0xffffff9b0ecf7000
  . DATA  : 0xffffff9b0ecf7000 - 0xffffffa17502b000
  Metadata: 0xffffffa17d03b000 - 0xffffffa19d03b000
  Bitmaps : 0xffffffa19d03b000 - 0xffffffa1a003b000

`

**Kext Download Source**

alxvngn, 2023-04-02:
I have this same issue on an Asus X515JA i3-1005G1

devidlis123, 2023-04-04:
Please use version 2.2.0 of the driver. Support for Mac OS 13. x in version 2.2.0

closed 2023-05-16 by zxystd