OpenIntelWireless/itlwm #867: Naming suggestion for the versions of KEXTs

RTHPJM, 2023-04-05:
Hello everyone,

Recent releases of AirportItlwm (and itlwm) have versions for major  Mac OS Releases. For example "Ventura", "Monterey", etc...

I would like to make use of OpenCore's MinKernel and MaxKernel values to load the correct KEXT depending upon the version of Mac OS that is booting. However, all the versions of itlwm and AirportItlwm have the same name for the KEXT.

I would like to propose that each version of the KEXT be built with a unique name. This way the KEXT files can be placed in the OpenCore Kexts folder, then the OpenCore config.plist can then be used to select the required version.

Perhaps utilise the Mac OS releases number (or name)...
e.g. 
AirportItlwm12.kext (for Monterey)
AirportItlwm13.kext (for Ventura)

isfong, 2023-04-07:
Ventura's kext isn't perfect, and a fix isn't in sight, but that's an aside, of course.

zxystd, 2023-05-16:
Thank you but current naming is also acceptable.

closed 2023-05-16 by zxystd